kbialek/deye-inverter-mqtt #2: Modbus frame is not valid or empty

fatexs, 2022-10-18:
I'm using a Deye SUN-10K-SG04LP3-EU
with Datalogger Serial 23804xxxxx
Firmware version
LSW3_15_FFFF_1.0.84

I'm getting this error;
022-10-18 20:36:42,724 - DeyeDaemon - INFO - Reading start
2022-10-18 20:36:42,748 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d0200000000000000000000000000000103003c001485c9f615
2022-10-18 20:36:49,750 - DeyeConnector - DEBUG - Response frame: a53b0010150007482ee38d020121d0060091010000403e486301032800ffffff160a12162420ffffffffffffffffffffffffffffffffffff0001ffff0001ffff000003e81fa45115
2022-10-18 20:36:49,751 - DeyeModbus - ERROR - Modbus frame is not valid or empty
2022-10-18 20:36:49,758 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d02000000000000000000000000000001030050001044171315
2022-10-18 20:36:49,872 - DeyeConnector - DEBUG - Response frame: a5330010150008482ee38d020122d0060091010000403e486301032000010000ffffffffffff0001ffffffffffffffffffff0000ffff0011ffffffff3a005715
2022-10-18 20:36:49,874 - DeyeModbus - ERROR - Modbus frame is not valid or empty
2022-10-18 20:36:49,878 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d0200000000000000000000000000000103006d0004d5d47215
2022-10-18 20:36:49,953 - DeyeConnector - DEBUG - Response frame: a51b0010150009482ee38d020122d0060091010000403e486301030800c8ffff00000000dc009315
2022-10-18 20:36:49,954 - DeyeModbus - ERROR - Modbus frame is not valid or empty
2022-10-18 20:36:49,957 - DeyeDaemon - INFO - Reading completed

Any idea whats wrong here?

Thanks!

kbialek, 2022-10-19:
Hey! @fatexs 

Well, my modbus frame validation was too simplistic, and a bit incorrect. I've improved it and implemented a proper CRC check. If you are willing to test it us this docker image https://github.com/kbialek/deye-inverter-mqtt/pkgs/container/deye-inverter-mqtt/46475923?tag=2022.10.1-fix-modbus-crc

fatexs, 2022-10-19:
Wow! You are a legend!
it works now!

Some Values seem odd to me... but could be that  the inverter is in standby because its dark outside ;)

2022-10-19 20:40:01,705 - DeyeDaemon - INFO - Reading start
2022-10-19 20:40:01,709 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d0200000000000000000000000000000103003c001485c9f615
2022-10-19 20:40:04,072 - DeyeConnector - DEBUG - Response frame: a53b0010150015482ee38d02012a210800ef010000743f486301032800ffffff160a13162731ffffffffffffffffffffffffffffffffffff0001ffff0001ffff000003e88ef42215
2022-10-19 20:40:04,077 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d02000000000000000000000000000001030050001044171315
2022-10-19 20:40:09,372 - DeyeConnector - DEBUG - Response frame: a5330010150016482ee38d02012f210800f5010000743f486301032000010000ffffffffffff0001ffffffffffffffffffff0000ffff0011ffffffff3a005e15
2022-10-19 20:40:09,376 - DeyeConnector - DEBUG - Request frame: a5170010450000482ee38d0200000000000000000000000000000103006d0004d5d47215
2022-10-19 20:40:14,545 - DeyeConnector - DEBUG - Response frame: a51b0010150017482ee38d020134210800fa010000743f48630103080005ffff00000000c0cc9115
2022-10-19 20:40:14,545 - DeyeDaemon - DEBUG - Production today: 25.5
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase1 Voltage: 6553.5
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase1 Current: 0.1
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase1 Power: 655.4
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase2 Voltage: 0.1
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase2 Current: 6553.5
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase2 Power: 655.4
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase3 Voltage: 6553.5
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase3 Current: 0.0
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Phase3 Power: 0.0
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - AC Freq: 10.0
2022-10-19 20:40:14,546 - DeyeDaemon - DEBUG - Uptime: 5642.0
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV1 Voltage: 0.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV1 Current: 6553.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV1 Power: 3276.8
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV2 Voltage: 0.0
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV2 Current: 0.0
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - PV2 Power: 0.0
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - DC Total Power: 6553.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - Operating Power: 0.1
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - AC Apparent Power: 6553.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - AC Active Power: 6553.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - AC Reactive Power: 6553.5
2022-10-19 20:40:14,547 - DeyeDaemon - DEBUG - Radiator temperature: 6453.5
2022-10-19 20:40:14,548 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2022-10-19 20:40:14,551 - DeyeDaemon - INFO - Reading completed

closed 2022-10-20 by kbialek