kbialek/deye-inverter-mqtt #5: Sensor adaption for Deye SUN-12K-SG04LP3-EU 3-phase inverter

thing4711, 2022-12-06:
Hello,

I have a 3-phase inverter from Deye and tried to modify your scripts to read out what I think are the relevant modbus registers for it. Unfortunately I have absolutely no idea of modbus, I don't know docker and I simply don't get it done :-/.
I have attached how I modified your deye_sensors.py in the docker image:

[deye_sensors.py.txt](https://github.com/kbialek/deye-inverter-mqtt/files/10169856/deye_sensors.py.txt)

When running the modified docker image, it still says
```
2022-12-06 19:28:20,260 - DeyeDaemon - INFO - Reading start
2022-12-06 19:28:20,603 - DeyeDaemon - INFO - Reading completed
```
and it also produces no error messages but it doesn't publish anything to MQTT.
Hence I think it's also not successfully reading anything.
I have no idea how to debug it further.

Are you able to help maybe ?

Thank you in advance and best regards.

thing4711, 2022-12-06:
And maybe to add,  because I think this is part of the issue: When writing 0x256 as address into the deye_sensors.py, I mean the decimal address 598.

`grid_outside_voltage_L1_sensor = SingleRegisterSensor("Grid outside voltage L1", 0x256, 0.1, mqtt_topic_suffix='grid/outside/voltage/L1')`

When running
```
root@broker ~# docker run --env-file deye-inverter-mqtt/config.env ghcr.io/kbialek/deye-inverter-mqtt r 598
int: 2373, l: 69, h: 9
```
it gives the right output of `237,3V` also. So in general, the inverter is reachable, answering, etc. but I guess, my dec/hex conversion is wrong or something alike.

kbialek, 2022-12-07:
hi @thing4711 

First of all, I'm impressed by the list of registers you want to read from the device. Do you really need to read them all?

Now, why are they not read? Simply because the bridge is optimized to query only a subset of registers from the logger. Why? Because each read request takes time and quite often has to be retried due to a timeout.

Can this be changed? Absolutely yes. Here is how.
1. Define feature specific register grups, e.g. battery, grid_outside, grid_inside, offgrid, etc.
2. Add a configuration option to specify the groups to be read, by default only the current, minimal set of registers will be read
3. Enhance the reading logic to query only configured set of register groups

thing4711, 2022-12-08:
Hi Krzysztof,

thank you for the quick response. That said, it makes absolutely sense.
And sorry for posting it as an issue while it is more a feature request, I guess.

Wrt the number of registers: I have tried to include everything that I could identify by the relevant documentation from Deye as this is my first setup of the inverter and I would want to log/monitor as much as possible to really understand what it is doing in which situation. After this phase of understanding, I guess, only a few of those parameters would be relevant for daily operations.

I forgot in the list of things I don't know, that I also don't know python ;-).
Could you maybe point me to the right spot where to extend the registers that are read for later evaluation ?
I have the feeling it is somewhere inside deye_modbus.py where all those hex values are somewhere in the functions, but I have to admit, I don't get it :-/.

Don't know, if it helps going forward, but these are the groups of subsequent register numbers that are currently of interest:
- 130
- 514-535
- 540-541
- 555-558
- 587-591
- 590-591
- 598-655
- 661-667
- 672-683

kbialek, 2022-12-08:
Hi @thing4711 

If you are willing to contribute then please open a documentation PR and add new registers in the table (`README.md` file). Feel free to also add a column and define the groups.

Once this is ready I will implement it in the upcoming weeks.

archi, 2023-03-08:
Can we please reopen this? I was watching this since I've the Deye 12K sitting here, but not running _yet_, so can't contribute in a meaningful way, _yet_. However, register lists can be taken from these projects & I'd happily test them once my inverter reads other values than 0 everything ;-)

https://github.com/bagges/deye-esp32-bridge/blob/main/deye-esp32-bridge.yaml Built by a friend of mine, and seems to work well

https://github.com/kellerza/sunsynk/blob/main/sunsynk/definitions3ph.py Someone else & I did gather some registers, but this one is not really tested (and at least the inverter power is read wrong, unless they supplied me with a suprisingly compact 26MW inverter 😆)

kbialek, 2023-03-09:
Okay. Meanwhile I did some changes that should make it easier to add new metrics. I'm looking forward to your tests @archi

archi, 2023-04-10:
not stale, just waiting for the electrician ;)

kbialek, 2023-04-10:
Actually sg04lp3 is supported since the latest release https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.03.4
So I'm closing this issue as for now. Open another one if you will have a problem with this tool in the future.

closed 2023-04-10 by kbialek