kbialek/deye-inverter-mqtt #7: Manifest for Raspberry

Antagor, 2022-12-16:
Hello,
would it be possible to add the repositories for Raspberry aswell in the docker?
I run the docker on my pc, working great! Good job! thanks.
try to start it on Raspberry -> docker: no matching manifest for linux/arm64/v8 in the manifest list entries.
my pc consumes more power than my sun600 can produce in winter.
thank you for your support.
Antagor

kbialek, 2022-12-16:
Sure https://github.com/kbialek/deye-inverter-mqtt/pull/8

kbialek, 2022-12-16:
Done

closed 2022-12-16 by kbialek

Antagor, 2022-12-19:
Hello Krzyzstof,

many thanks for your support.
i started it at the same evenying on my Raspberry and it was working 2
days without an error.

Actually - as long as it is running - i'm fine with it - but i'd like to
have it as an instanz direct in my IoBroker.
Would you support this conversion to IoBroker or would it be possible to
use parts of your code for reading and transforming the data?
Actually i have no knowledge in programming for IoBroker - but most
languages or implementations are comparable....

Are you interseted in my device for compatibility in your list? Not alle
registers are used - but all i need (;-)
You can contact me direct via ***@***.***

many thanks for your work.
Have a good christmas time and a happy new year!
Marc

kbialek, 2022-12-19:
hi @Antagor 

> Actually - as long as it is running - i'm fine with it - but i'd like to have it as an instanz direct in my IoBroker. Would you support this conversion to IoBroker

No, I'm not interested in participation in the conversion to iobroker

> would it be possible to use parts of your code for reading and transforming the data? 

Sure, as long as the license is respected

> Are you interseted in my device for compatibility in your list?

Yes, absolutely

Best regards

Antagor, 2023-01-04:
> Sorry, i made a mistake.
> You have Reg 56 in your list, but as one reg value. please check if it
> is correct for you with one or two regs.

Hello Krzyzstof,

i created a fork with my changes.
Can you check them, transfer to your project if they are OK for you and
delete my fork?
or can i use my fork directly in my docker? i never have used it before
... sorry for my stupid questions.
please the the function Double... as i do not know it i can add a value
in the register adresse or if it must be defined before.

thanks for your support

Marc (Antagor)

-------- Weitergeleitete Nachricht --------
Betreff: 	Re: [kbialek/deye-inverter-mqtt] Manifest for Raspberry (Issue
#7)
Datum: 	Mon, 2 Jan 2023 10:15:43 +0100

kbialek, 2023-01-05:
Hi Marc,

In order to make the things easier, please open a PullRequest from your code into my project. Also please allow me to push into your PR (see https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/allowing-changes-to-a-pull-request-branch-created-from-a-fork) 

Thanks!

Antagor, 2023-01-05:
I tried to do it like you requested. Please have a look and let me know if I should do anything. Marc--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

kbialek, 2023-01-05:
Hi Marc,

I've made few minor changes in your PR so the code runs with your updates. I've also provided a code review.

Antagor, 2023-01-05:
Thanks. I'm at work and can't do it... If your temp values are ok - I'm fine with it. Maybe there is a difference between micro and string inverter. Names are your part as it is your project! Of course you can decide. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

kbialek, 2023-01-05:
Sure, take your time

kbialek, 2023-01-06:
@Antagor I've made a new release https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.01.1 that also includes your changes. I've added a concept of metric groups that will help to support both string and micro inverters. Could you please test it? Please read the updated docs and also update your config accordingly (you will need to add a new env variable `DEYE_METRIC_GROUPS=micro`). Thanks!