kbialek/deye-inverter-mqtt #14: Only some values are send over mqtt

IeP4nieF, 2023-01-09:
Hi,

I'm using deye-inverter-mqtt with a [Deye SUN600G3](https://www.deyeinverter.com/product/microinverter-1/sun600-800-1000g3eu230-single-phase-4-mppt-microinverter-rapid-shutdown.html). From the DEBUG output I can see, that the connection to the inverter and the connection to the MQTT-Broker works fine. Not all values makes sense, but this is not the problem at the moment.

My problem (and I think this is a bug): Only some values are sent via MQTT to the broker, as I can see in "MQTT Explorer" (verified with wireshark). For example at the moment (deye-inverter-mqtt runs for 30min now) only "day_energy" (66 times), "total_energy" (8 times), ac: "voltage" (6 times) and current (2 times) were sent. The most important "power" was never sent.

Looking at the python code I can't see, why this happens. Once again in DEBUG output all values are printed (every 20s).

Is there something I can do for further debugging? Any ideas?

Best regards

kbialek, 2023-01-09:
Hi @IeP4nieF 

Just to be sure, have you read the release notes and updated your config accordingly? https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.01.1

IeP4nieF, 2023-01-09:
Thanks for the fast answer!

No, I didn't read the release notes, because I installed yesterday. ;-) So there is no need to update the config. In `config.env` there is one line:

    DEYE_METRIC_GROUPS=micro

BTW: At this moment (after about 1h) the first all values was transfered the first time. Like I said not all values makes sense for example `DC Total Power` is `0` , while `PV1 Power` is `4.7`. But this is another problem and not to discuss here in this bugreport.

kbialek, 2023-01-09:
I see you have a microinverter, so `DEYE_METRIC_GROUPS=micro` should suit your needs. If not then you can collect the debug logs and send them to me.

IeP4nieF, 2023-01-09:
Are there sensitive informations in "Request frame" or "Response frame"? If not, I'll post them here.

kbialek, 2023-01-09:
The logger SN is included in the 1st half of the request frame. I'd not call it a sensitive information, but you can obfuscate it if you wish.
An MQTT dump would be also helpful, you can use https://mosquitto.org/man/mosquitto_sub-1.html to grab it

IeP4nieF, 2023-01-09:
ok, here the DEBUG log (without SN and POWER being 0 because the sun is gone):
```
2023-01-09 15:10:08,947 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103003c001485c99115
2023-01-09 15:10:15,467 - DeyeConnector - DEBUG - Response frame: a53b0010150091********0201ab1c0100c90500000000000001032800120000000000120000000800080000000000080000000800000924000000000000000000001388702cdf15
2023-01-09 15:10:15,471 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103005000104417ae15
2023-01-09 15:10:15,560 - DeyeConnector - DEBUG - Response frame: a5330010150092********0201ab1c0100c905000000000000010320000000000000000000000000003c000000000000079400000000000000000000d3d6a815
2023-01-09 15:10:15,566 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103006d0008d5d10e15
2023-01-09 15:10:15,652 - DeyeConnector - DEBUG - Response frame: a5230010150093********0201ab1c0100c905000000000000010310014500000145000000000000000000001934e215
2023-01-09 15:10:15,653 - DeyeDaemon - DEBUG - Production today: 1.8
2023-01-09 15:10:15,653 - DeyeDaemon - DEBUG - Production Total: 1.8
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - Phase1 Voltage: 234.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - Phase1 Current: 0.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - Phase1 Power: 0.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - AC Freq: 50.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - Uptime: 0.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - PV1 Voltage: 32.5
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - PV1 Current: 0.0
2023-01-09 15:10:15,654 - DeyeDaemon - DEBUG - PV1 Power: 0.0
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV1 Production today: 0.8
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV1 Total: 0.8
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV2 Voltage: 32.5
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV2 Current: 0.0
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV2 Power: 0.0
2023-01-09 15:10:15,655 - DeyeDaemon - DEBUG - PV2 Production today: 0.8
2023-01-09 15:10:15,656 - DeyeDaemon - DEBUG - PV2 Total: 0.8
2023-01-09 15:10:15,656 - DeyeDaemon - DEBUG - PV3 Voltage: 0.0
2023-01-09 15:10:15,656 - DeyeDaemon - DEBUG - PV3 Current: 0.0
2023-01-09 15:10:15,656 - DeyeDaemon - DEBUG - PV3 Power: 0.0
2023-01-09 15:10:15,656 - DeyeDaemon - DEBUG - PV3 Production today: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV3 Total: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV4 Voltage: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV4 Current: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV4 Power: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV4 Production today: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - PV4 Total: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - DC Total Power: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - Operating Power: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - AC Apparent Power: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - AC Active Power: 6.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - AC Reactive Power: 0.0
2023-01-09 15:10:15,657 - DeyeDaemon - DEBUG - Radiator temperature: 94.0
2023-01-09 15:10:15,658 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2023-01-09 15:10:15,659 - DeyeDaemon - INFO - Reading completed
```
The inverter turned of, because it is dark now. So today I could not send output from `mosquitto_sub`. Instead here comes the dumps from wireshark. Most times only `day_energy` was sent:
```
0000   02 42 64 ec 1c 85 02 42 ac 11 00 02 08 00 45 00   .Bd....B......E.
0010   00 4a 3d 11 40 00 40 06 60 7c ac 11 00 02 c0 a8   .J=.@.@.`|......
0020   30 65 83 63 07 5b 96 b4 9a 3f 6c 00 de ae 80 18   0e.c.[...?l.....
0030   01 f6 9d 5d 00 00 01 01 08 0a ae 9b 77 70 5c c2   ...]........wp\.
0040   14 87 30 14 00 0f 64 65 79 65 2f 64 61 79 5f 65   ..0...deye/day_e
0050   6e 65 72 67 79 31 2e 38                           nergy1.8
```
As I wrote before, all values was transfered once. Here is the dump of this MQTT-message:
```
0000   02 42 64 ec 1c 85 02 42 ac 11 00 02 08 00 45 00   .Bd....B......E.
0010   02 21 78 0e 40 00 40 06 23 a8 ac 11 00 02 c0 a8   .!x.@.@.#.......
0020   30 65 91 3d 07 5b cd db 1e 4d 4b b7 4a 79 80 19   0e.=.[...MK.Jy..
0030   01 f6 9f 34 00 00 01 01 08 0a ae 7f ae 0c 5c a6   ...4..........\.
0040   4b 22 30 1d 00 18 64 65 79 65 2f 64 63 2f 70 76   K"0...deye/dc/pv
0050   32 2f 74 6f 74 61 6c 5f 65 6e 65 72 67 79 30 2e   2/total_energy0.
0060   38 30 18 00 13 64 65 79 65 2f 64 63 2f 70 76 33   80...deye/dc/pv3
0070   2f 76 6f 6c 74 61 67 65 30 2e 30 30 18 00 13 64   /voltage0.00...d
0080   65 79 65 2f 64 63 2f 70 76 33 2f 63 75 72 72 65   eye/dc/pv3/curre
0090   6e 74 30 2e 30 30 16 00 11 64 65 79 65 2f 64 63   nt0.00...deye/dc
00a0   2f 70 76 33 2f 70 6f 77 65 72 30 2e 30 30 1b 00   /pv3/power0.00..
00b0   16 64 65 79 65 2f 64 63 2f 70 76 33 2f 64 61 79   .deye/dc/pv3/day
00c0   5f 65 6e 65 72 67 79 30 2e 30 30 1d 00 18 64 65   _energy0.00...de
00d0   79 65 2f 64 63 2f 70 76 33 2f 74 6f 74 61 6c 5f   ye/dc/pv3/total_
00e0   65 6e 65 72 67 79 30 2e 30 30 18 00 13 64 65 79   energy0.00...dey
00f0   65 2f 64 63 2f 70 76 34 2f 76 6f 6c 74 61 67 65   e/dc/pv4/voltage
0100   30 2e 30 30 18 00 13 64 65 79 65 2f 64 63 2f 70   0.00...deye/dc/p
0110   76 34 2f 63 75 72 72 65 6e 74 30 2e 30 30 16 00   v4/current0.00..
[... cut ...]
01d0   63 74 69 76 65 5f 70 6f 77 65 72 39 2e 30 30 1b   ctive_power9.00.
01e0   00 16 64 65 79 65 2f 61 63 2f 72 65 61 63 74 69   ..deye/ac/reacti
01f0   76 65 5f 70 6f 77 65 72 30 2e 30 30 18 00 12 64   ve_power0.00...d
0200   65 79 65 2f 72 61 64 69 61 74 6f 72 5f 74 65 6d   eye/radiator_tem
0210   70 38 32 2e 30 30 16 00 0e 64 65 79 65 2f 69 67   p82.00...deye/ig
0220   62 74 5f 74 65 6d 70 2d 31 30 30 2e 30 e0 00      bt_temp-100.0..
```
Hope that helps. Otherwise I'll send a dump from `mosquitto_sub` tomorrow.

kbialek, 2023-01-09:
Thanks, please try to collect more data tomorrow when there is light outside.

IeP4nieF, 2023-01-10:
Here is again the DEBUG output:
```
$ docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt
2023-01-10 13:49:24,024 - DeyeDaemon - INFO - Reading start
2023-01-10 13:49:24,053 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103003c001485c99115
2023-01-10 13:49:34,065 - DeyeConnector - WARNING - Connection timeout
2023-01-10 13:49:35,547 - DeyeConnector - DEBUG - Response frame: a53b001015002b********0201b7780100dd0300000000000001032800150000000000150000000a000900000000000a0000000900000924000000000002000000001392e691e615
2023-01-10 13:49:35,553 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103005000104417ae15
2023-01-10 13:49:38,830 - DeyeConnector - DEBUG - Response frame: a533001015002c********0201ba780100e10300000000000001032000000000000000000000000001d6000000000000079e00000000000000000000e3d07215
2023-01-10 13:49:38,840 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103006d0008d5d10e15
2023-01-10 13:49:42,307 - DeyeConnector - DEBUG - Response frame: a523001015002d********0201be780100e40300000000000001031001320007012800070000000000000000c6c82315
2023-01-10 13:49:42,309 - DeyeDaemon - DEBUG - Production today: 2.1
2023-01-10 13:49:42,310 - DeyeDaemon - DEBUG - Production Total: 2.1
2023-01-10 13:49:42,311 - DeyeDaemon - DEBUG - Phase1 Voltage: 234.0
2023-01-10 13:49:42,312 - DeyeDaemon - DEBUG - Phase1 Current: 0.2
2023-01-10 13:49:42,312 - DeyeDaemon - DEBUG - Phase1 Power: 46.8
2023-01-10 13:49:42,312 - DeyeDaemon - DEBUG - AC Freq: 50.1
2023-01-10 13:49:42,314 - DeyeDaemon - DEBUG - Uptime: 0.0
2023-01-10 13:49:42,315 - DeyeDaemon - DEBUG - PV1 Voltage: 30.6
2023-01-10 13:49:42,316 - DeyeDaemon - DEBUG - PV1 Current: 0.7
2023-01-10 13:49:42,317 - DeyeDaemon - DEBUG - PV1 Power: 21.4
2023-01-10 13:49:42,317 - DeyeDaemon - DEBUG - PV1 Production today: 1.0
2023-01-10 13:49:42,317 - DeyeDaemon - DEBUG - PV1 Total: 1.0
2023-01-10 13:49:42,318 - DeyeDaemon - DEBUG - PV2 Voltage: 29.6
2023-01-10 13:49:42,319 - DeyeDaemon - DEBUG - PV2 Current: 0.7
2023-01-10 13:49:42,320 - DeyeDaemon - DEBUG - PV2 Power: 20.7
2023-01-10 13:49:42,320 - DeyeDaemon - DEBUG - PV2 Production today: 0.9
2023-01-10 13:49:42,320 - DeyeDaemon - DEBUG - PV2 Total: 0.9
2023-01-10 13:49:42,321 - DeyeDaemon - DEBUG - PV3 Voltage: 0.0
2023-01-10 13:49:42,322 - DeyeDaemon - DEBUG - PV3 Current: 0.0
2023-01-10 13:49:42,323 - DeyeDaemon - DEBUG - PV3 Power: 0.0
2023-01-10 13:49:42,323 - DeyeDaemon - DEBUG - PV3 Production today: 0.0
2023-01-10 13:49:42,323 - DeyeDaemon - DEBUG - PV3 Total: 0.0
2023-01-10 13:49:42,326 - DeyeDaemon - DEBUG - PV4 Voltage: 0.0
2023-01-10 13:49:42,326 - DeyeDaemon - DEBUG - PV4 Current: 0.0
2023-01-10 13:49:42,327 - DeyeDaemon - DEBUG - PV4 Power: 0.0
2023-01-10 13:49:42,328 - DeyeDaemon - DEBUG - PV4 Production today: 0.0
2023-01-10 13:49:42,328 - DeyeDaemon - DEBUG - PV4 Total: 0.0
2023-01-10 13:49:42,328 - DeyeDaemon - DEBUG - DC Total Power: 0.0
2023-01-10 13:49:42,329 - DeyeDaemon - DEBUG - Operating Power: 0.0
2023-01-10 13:49:42,330 - DeyeDaemon - DEBUG - AC Apparent Power: 0.0
2023-01-10 13:49:42,331 - DeyeDaemon - DEBUG - AC Active Power: 47.0
2023-01-10 13:49:42,331 - DeyeDaemon - DEBUG - AC Reactive Power: 0.0
2023-01-10 13:49:42,333 - DeyeDaemon - DEBUG - Radiator temperature: 95.0
2023-01-10 13:49:42,333 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2023-01-10 13:49:42,342 - DeyeDaemon - INFO - Reading completed
```
(BTW: As you can see, the "DC Total Power" is wrong, which must be 21.4+20.7. But again, this is not the main problem.)

And this is what I get from `mosquitto_sub` as requested (it is not much more information, then yesterday):
```
$ mosquitto_sub -h <IP-of-mqtt-brocker> -t "deye/#" -v
deye/day_energy 2.1
```
This time only `day_energy` was transfered. Sometimes more topics are transfered. I can not figure out, at what time which topic is transfered. Yesterday it took nearly 60 "rounds" until some DC-topics where transfered.

I hope that this helps to debug.

kbialek, 2023-01-10:
Hi @IeP4nieF 

> (BTW: As you can see, the "DC Total Power" is wrong, which must be 21.4+20.7. But again, this is not the main problem.)

I'm not sure if register 0x52 is used by micro inverters. @Antagor Maybe you can say more about this? Does it work for you? The spec does not mention microinverter at this registry.

> And this is what I get from `mosquitto_sub` as requested (it is not much more information, then yesterday):
>
> ```
> $ mosquitto_sub -h <IP-of-mqtt-brocker> -t "deye/#" -v
> deye/day_energy 2.1
> ```
>
> This [...]

Let's ensure that MQTT connectivity is not a problem here. Please try this beta docker image [2023.01.2-beta](https://github.com/kbialek/deye-inverter-mqtt/pkgs/container/deye-inverter-mqtt/62774954?tag=2023.01.2-beta)
I've improved MQTT publishing in it. Previously the daemon didn't care about published messages. Now it will block for up to 3s until the message is published. Otherwise it will report an error. So let's see.

Antagor, 2023-01-10:
Hello,
DC Total Power is not used (=0) at micro. As there are enough Power values i don't need it.

MQTT old version:
i get always error messages in log file of IOBroker. 
I will try the new beta - but i can't test before friday.

I have the same problem that not always all values were transfered. i expected it would be a timeout of deye.

I have another topic i will open an issue when i have enough information:
it seams that deye can't calculate daily values without connection to china server. in the morning all daily and total values are 0 when WIFI is established. A few minutes later daily and total apear with values of thew day before.
it can be that deye need a "new" NTP Server or China Server set daily to zero via modbus.
Are you all connected to China?

As my deye is sleeping i will do aswell...
See you
Antagor

Antagor, 2023-01-10:
Wouldn't it be easier to establish the MQTT connection and than just send the data. Leave the connectioon open for next transfer. 
mqtt.info.connection is always empy but has new timestamp because connection closed.

I have some more fields which are empty but i didn't mind that they are there. I expect micro do not deliver as much data than the string & hybrid.
For me zero: 
ac.apparent_power
ac.reactive_power
dc.total_power
IGBT Temp
operating_power
uptime

IeP4nieF, 2023-01-11:
> Let's ensure that MQTT connectivity is not a problem here. Please try this beta docker image [...]

I tested with the beta docker image ([2023.01.2-beta]): The behavior is the same. Because of this I don't upload new debug logs. They look like my debug logs above. Again most times only day_energy is transfered:

```
$ mosquitto_sub -h <IP-of-mqtt-brocker> -t "deye/#" -v
deye/day_energy 2.2
deye/day_energy 2.2
deye/day_energy 2.2
[...]
```
Let me know, if I should test something else.

IeP4nieF, 2023-01-11:
> Hello, DC Total Power is not used (=0) at micro. As there are enough Power values i don't need it.

Yes, there are enough Power values! No need for DC Total Power.

> MQTT old version: i get always error messages in log file of IOBroker.

Here I don't have error messages in the logs. I use mosquitto 2.0.11-1 (Debian bullseye).

> I have the same problem that not always all values were transfered. i expected it would be a timeout of deye.

I don't think, that this is a problem with timeout of deye, because the values are always in the DEBUG logs of deye-inverter-mqtt. I think this ensures, that the values are read from deye without timeout.

> I have another topic i will open an issue when i have enough information: it seams that deye can't calculate daily values without connection to china server. in the morning all daily and total [...]

Same problem here. I'm also (of course! ;-) ) not connected to China. daily and total also apear with values of the day before. It would be very nice to find a solution for this. But this is off topic here. 🤓

IeP4nieF, 2023-01-11:
Yes, I tested the beta image: "I tested with the beta docker image ([2023.01.2-beta]): The behavior is the sa..." (Two posts above.)

IeP4nieF, 2023-01-11:
> Many off topics are popping up in here stuck_out_tongue Have you tested the beta image @IeP4nieF ?

I ran: `docker pull ghcr.io/kbialek/deye-inverter-mqtt:2023.01.2-beta` and after that `docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt`.

IeP4nieF, 2023-01-11:
Sorry, I'm new to docker: Is there a possibility for me to change the pyhton-files inside the docker image? If this would be possible, I can go on with debugging.

kbialek, 2023-01-11:
To test the beta image run `docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt:2023.01.2-beta`

IeP4nieF, 2023-01-11:
Oh, ok! Thank you and sorry. I'm new to docker. But now I have to wait until tomorrow. No sunlight here anymore.

Antagor, 2023-01-11:
Hello,
i checked my data with new beta-docker. today (all 20 sec.) i have over 350 values for pv1 voltage, but only 11 !!! for active_energy.
more i can't actually.
it's dark outside

IeP4nieF, 2023-01-12:
Now I tested with the beta image:
```
$ docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt:2023.01.2-beta
2023-01-12 13:53:29,940 - DeyeDaemon - INFO - Reading start
2023-01-12 13:53:29,982 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103003c001485c99115
2023-01-12 13:53:36,277 - DeyeConnector - DEBUG - Response frame: a53b0010150001********020112290200450100000000000001032800170000000000170000000b000a00000000000b0000000a0000095600000000000000000000138817d7d415
2023-01-12 13:53:36,319 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103005000104417ae15
2023-01-12 13:53:39,379 - DeyeConnector - DEBUG - Response frame: a5330010150002********020115290200480100000000000001032000000000000000000000000000be0000000000000960000000000000000000007c548215
2023-01-12 13:53:39,404 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103006d0008d5d10e15
2023-01-12 13:53:42,832 - DeyeConnector - DEBUG - Response frame: a5230010150003********0201192902004c01000000000000010310012b0003016d000200000000000000006a1b9815
2023-01-12 13:53:42,834 - DeyeDaemon - DEBUG - Production today: 2.3
2023-01-12 13:53:42,834 - DeyeDaemon - DEBUG - Production Total: 2.3
2023-01-12 13:53:42,834 - DeyeDaemon - DEBUG - Phase1 Voltage: 239.0
2023-01-12 13:53:42,834 - DeyeDaemon - DEBUG - Phase1 Current: 0.0
2023-01-12 13:53:42,835 - DeyeDaemon - DEBUG - Phase1 Power: 0.0
2023-01-12 13:53:42,835 - DeyeDaemon - DEBUG - AC Freq: 50.0
2023-01-12 13:53:42,835 - DeyeDaemon - DEBUG - Uptime: 0.0
2023-01-12 13:53:42,835 - DeyeDaemon - DEBUG - PV1 Voltage: 29.9
2023-01-12 13:53:42,835 - DeyeDaemon - DEBUG - PV1 Current: 0.3
2023-01-12 13:53:42,836 - DeyeDaemon - DEBUG - PV1 Power: 9.0
2023-01-12 13:53:42,836 - DeyeDaemon - DEBUG - PV1 Production today: 1.1
2023-01-12 13:53:42,836 - DeyeDaemon - DEBUG - PV1 Total: 1.1
2023-01-12 13:53:42,836 - DeyeDaemon - DEBUG - PV2 Voltage: 36.5
2023-01-12 13:53:42,837 - DeyeDaemon - DEBUG - PV2 Current: 0.2
2023-01-12 13:53:42,837 - DeyeDaemon - DEBUG - PV2 Power: 7.3
2023-01-12 13:53:42,838 - DeyeDaemon - DEBUG - PV2 Production today: 1.0
2023-01-12 13:53:42,839 - DeyeDaemon - DEBUG - PV2 Total: 1.0
2023-01-12 13:53:42,839 - DeyeDaemon - DEBUG - PV3 Voltage: 0.0
2023-01-12 13:53:42,840 - DeyeDaemon - DEBUG - PV3 Current: 0.0
2023-01-12 13:53:42,840 - DeyeDaemon - DEBUG - PV3 Power: 0.0
2023-01-12 13:53:42,841 - DeyeDaemon - DEBUG - PV3 Production today: 0.0
2023-01-12 13:53:42,841 - DeyeDaemon - DEBUG - PV3 Total: 0.0
2023-01-12 13:53:42,841 - DeyeDaemon - DEBUG - PV4 Voltage: 0.0
2023-01-12 13:53:42,842 - DeyeDaemon - DEBUG - PV4 Current: 0.0
2023-01-12 13:53:42,842 - DeyeDaemon - DEBUG - PV4 Power: 0.0
2023-01-12 13:53:42,843 - DeyeDaemon - DEBUG - PV4 Production today: 0.0
2023-01-12 13:53:42,843 - DeyeDaemon - DEBUG - PV4 Total: 0.0
2023-01-12 13:53:42,843 - DeyeDaemon - DEBUG - DC Total Power: 0.0
2023-01-12 13:53:42,844 - DeyeDaemon - DEBUG - Operating Power: 0.0
2023-01-12 13:53:42,845 - DeyeDaemon - DEBUG - AC Apparent Power: 0.0
2023-01-12 13:53:42,845 - DeyeDaemon - DEBUG - AC Active Power: 19.0
2023-01-12 13:53:42,845 - DeyeDaemon - DEBUG - AC Reactive Power: 0.0
2023-01-12 13:53:42,846 - DeyeDaemon - DEBUG - Radiator temperature: 140.0
2023-01-12 13:53:42,847 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2023-01-12 13:53:42,849 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/day_energy', value: '2.3'
2023-01-12 13:53:42,851 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '2.3'
2023-01-12 13:53:42,852 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/voltage', value: '239.0'
2023-01-12 13:53:42,852 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/current', value: '0.0'
2023-01-12 13:53:42,853 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/power', value: '0.0'
2023-01-12 13:53:42,854 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/freq', value: '50.0'
2023-01-12 13:53:42,855 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/uptime', value: '0.0'
2023-01-12 13:53:42,856 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/voltage', value: '29.9'
2023-01-12 13:53:42,857 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/current', value: '0.3'
2023-01-12 13:53:42,858 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/power', value: '9.0'
2023-01-12 13:53:42,858 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/day_energy', value: '1.1'
2023-01-12 13:53:42,859 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/total_energy', value: '1.1'
2023-01-12 13:53:42,859 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/voltage', value: '36.5'
2023-01-12 13:53:42,860 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/current', value: '0.2'
2023-01-12 13:53:42,861 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/power', value: '7.3'
2023-01-12 13:53:42,861 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/day_energy', value: '1.0'
2023-01-12 13:53:42,862 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/total_energy', value: '1.0'
2023-01-12 13:53:42,863 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv3/voltage', value: '0.0'
[... cut ...]
2023-01-12 13:53:42,875 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/reactive_power', value: '0.0'
2023-01-12 13:53:42,876 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/radiator_temp', value: '140.0'
2023-01-12 13:53:42,877 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/igbt_temp', value: '-100.0'
2023-01-12 13:53:42,878 - DeyeDaemon - INFO - Reading completed
```
Again only `day_energy` is transfered via MQTT:
```
$ mosquitto_sub -h <IP-of-mqtt-brocker> -t "deye/#" -v
deye/day_energy 2.2
```

Antagor, 2023-01-12:
OK. Beta-2 is running, but it is dark outside.
i hope i can help tomorrow.
But you should think about open the mqtt connection and leave it open - as long as the sun is shining.
bye Antagor

kbialek, 2023-01-12:
> But you should think about open the mqtt connection and leave it open - as long as the sun is shining.

@Antagor Why would it help in your opinion?
When the connection is long living, you have to keep it alive. It can be interrupted, and then you have to recover from that state.
Since the daemon does not subscribe to any mqtt topic, it makes no sense to keep the connection opened permanently.
Currently, a new connection is opened on each bulk publish, means all metrics are published in a single connection, and then it's closed.
This way the daemon is not prone to connectivity errors caused by mqtt broker restarts and networking infrastructure stability issues.

Antagor, 2023-01-13:
You are right. it is more stable for the connection. my tasmota which collect all energy values is always online.
In Beta-2 you wait between all values transfered - now i can see "deye-inverter" is connected via mqtt.

But i have no problems in mqtt. it feels like all values appear in my IOBroker.
My problem is the data request. I don't know why but the first request is timed out VERY often. I don't expect an error because it is working sometimes.
2023-01-13 10:22:13,708 - DeyeDaemon - INFO - Reading start
2023-01-13 10:22:13,715 - DeyeConnector - DEBUG - Request frame: a51700104500009d8648f70200000000000000000000000000000103003c001485c97215
2023-01-13 10:22:23,717 - DeyeConnector - WARNING - Connection timeout
2023-01-13 10:22:33,728 - DeyeConnector - WARNING - Connection timeout
2023-01-13 10:22:43,723 - DeyeConnector - DEBUG - Response frame:
2023-01-13 10:22:43,724 - DeyeModbus - ERROR - No response frame
2023-01-13 10:22:43,724 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-13 10:22:43,729 - DeyeConnector - DEBUG - Request frame: a51700104500009d8648f702000000000000000000000000000001030050001044178f15
2023-01-13 10:22:53,740 - DeyeConnector - WARNING - Connection timeout
2023-01-13 10:23:00,456 - DeyeConnector - DEBUG - Response frame: a53300101500059d8648f7020132e40d003b010000000000000103200000000000000000000000000122000000000000083400000000000000000000563b3515
2023-01-13 10:23:00,461 - DeyeConnector - DEBUG - Request frame: a51700104500009d8648f70200000000000000000000000000000103006d0008d5d1ef15
2023-01-13 10:23:00,537 - DeyeConnector - DEBUG - Response frame: a52300101500069d8648f7020133e40d003b01000000000000010310010c000501180005000000000000000099faea15
2023-01-13 10:23:00,538 - DeyeDaemon - DEBUG - PV1 Voltage: 26.8
2023-01-13 10:23:00,538 - DeyeDaemon - DEBUG - PV1 Current: 0.5
2023-01-13 10:23:00,538 - DeyeDaemon - DEBUG - PV1 Power: 13.4
2023-01-13 10:23:00,538 - DeyeDaemon - DEBUG - PV2 Voltage: 28.0
2023-01-13 10:23:00,538 - DeyeDaemon - DEBUG - PV2 Current: 0.5
2023-01-13 10:23:00,539 - DeyeDaemon - DEBUG - PV2 Power: 14.0
2023-01-13 10:23:00,539 - DeyeDaemon - DEBUG - PV3 Voltage: 0.0
2023-01-13 10:23:00,539 - DeyeDaemon - DEBUG - PV3 Current: 0.0
2023-01-13 10:23:00,539 - DeyeDaemon - DEBUG - PV3 Power: 0.0
2023-01-13 10:23:00,540 - DeyeDaemon - DEBUG - PV4 Voltage: 0.0
2023-01-13 10:23:00,540 - DeyeDaemon - DEBUG - PV4 Current: 0.0
2023-01-13 10:23:00,540 - DeyeDaemon - DEBUG - PV4 Power: 0.0
2023-01-13 10:23:00,541 - DeyeDaemon - DEBUG - DC Total Power: 0.0
2023-01-13 10:23:00,541 - DeyeDaemon - DEBUG - Operating Power: 0.0
2023-01-13 10:23:00,541 - DeyeDaemon - DEBUG - AC Apparent Power: 0.0
2023-01-13 10:23:00,542 - DeyeDaemon - DEBUG - AC Active Power: 29.0
2023-01-13 10:23:00,542 - DeyeDaemon - DEBUG - AC Reactive Power: 0.0
2023-01-13 10:23:00,542 - DeyeDaemon - DEBUG - Radiator temperature: 110.0
2023-01-13 10:23:00,543 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2023-01-13 10:23:00,544 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/voltage', value: '26.8'
2023-01-13 10:23:03,547 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/current', value: '0.5'
2023-01-13 10:23:06,550 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/power', value: '13.4'
2023-01-13 10:23:09,554 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/voltage', value: '28.0'
2023-01-13 10:23:12,557 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/current', value: '0.5'
2023-01-13 10:23:15,560 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/power', value: '14.0'
2023-01-13 10:23:18,562 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv3/voltage', value: '0.0'
2023-01-13 10:23:21,564 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv3/current', value: '0.0'
2023-01-13 10:23:24,568 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv3/power', value: '0.0'
2023-01-13 10:23:27,571 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/voltage', value: '0.0'
2023-01-13 10:23:30,574 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/current', value: '0.0'
2023-01-13 10:23:33,577 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv4/power', value: '0.0'
2023-01-13 10:23:36,580 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/total_power', value: '0.0'
2023-01-13 10:23:39,585 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/operating_power', value: '0.0'
2023-01-13 10:23:42,588 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/apparent_power', value: '0.0'
2023-01-13 10:23:45,591 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/active_power', value: '29.0'
2023-01-13 10:23:48,593 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/reactive_power', value: '0.0'
2023-01-13 10:23:51,596 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/radiator_temp', value: '110.0'
2023-01-13 10:23:54,600 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/igbt_temp', value: '-100.0'
2023-01-13 10:23:57,604 - DeyeDaemon - INFO - Reading completed

kbialek, 2023-01-13:
I've educated myself a little bit more about using paho mqtt client library :smile: It's turned out that I've overlooked one important piece - the client event loop. It has to be active when sending messages with qos > 0. It could also be an explanation why some messages were not published to the broker. After fixing it, my integration test does not hang for 3s on each publish anymore. I've built a new beta image `2023.01.2-beta-3`. @IeP4nieF @Antagor Please test it if you can.

Antagor, 2023-01-13:
I will run it tomorrow--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

IeP4nieF, 2023-01-14:
> [...] I've built a new beta image `2023.01.2-beta-3`. @IeP4nieF @Antagor Please test it if you can.

Bingo! Now all values are transfered:
```
$ docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt:2023.01.2-beta-3
2023-01-14 07:51:46,749 - DeyeDaemon - INFO - Reading start
2023-01-14 07:51:46,772 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103003c001485c99115
2023-01-14 07:51:48,270 - DeyeConnector - DEBUG - Response frame: a53b0010150001********020198810200a90200000000000001032800190000000000190000000c000b00000000000c0000000b00000942000000000000000000001388cee9d415
2023-01-14 07:51:48,311 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103005000104417ae15
2023-01-14 07:51:51,563 - DeyeConnector - DEBUG - Response frame: a5330010150002********02019c810200ac0200000000000001032000000000000000000000000000500000000000000654000000000000000000005f4f2715
2023-01-14 07:51:51,596 - DeyeConnector - DEBUG - Request frame: a5170010450000********0200000000000000000000000000000103006d0008d5d10e15
2023-01-14 07:51:54,695 - DeyeConnector - DEBUG - Response frame: a5230010150003********02019f810200af0200000000000001031001520001012f000100000000000000009ef2cb15
2023-01-14 07:51:54,696 - DeyeDaemon - DEBUG - Production today: 2.5
2023-01-14 07:51:54,697 - DeyeDaemon - DEBUG - Production Total: 2.5
2023-01-14 07:51:54,697 - DeyeDaemon - DEBUG - Phase1 Voltage: 237.0
2023-01-14 07:51:54,698 - DeyeDaemon - DEBUG - Phase1 Current: 0.0
2023-01-14 07:51:54,698 - DeyeDaemon - DEBUG - Phase1 Power: 0.0
2023-01-14 07:51:54,699 - DeyeDaemon - DEBUG - AC Freq: 50.0
2023-01-14 07:51:54,700 - DeyeDaemon - DEBUG - Uptime: 0.0
2023-01-14 07:51:54,701 - DeyeDaemon - DEBUG - PV1 Voltage: 33.8
2023-01-14 07:51:54,701 - DeyeDaemon - DEBUG - PV1 Current: 0.1
2023-01-14 07:51:54,702 - DeyeDaemon - DEBUG - PV1 Power: 3.4
2023-01-14 07:51:54,702 - DeyeDaemon - DEBUG - PV1 Production today: 1.2
2023-01-14 07:51:54,703 - DeyeDaemon - DEBUG - PV1 Total: 1.2
2023-01-14 07:51:54,703 - DeyeDaemon - DEBUG - PV2 Voltage: 30.3
2023-01-14 07:51:54,704 - DeyeDaemon - DEBUG - PV2 Current: 0.1
2023-01-14 07:51:54,704 - DeyeDaemon - DEBUG - PV2 Power: 3.0
2023-01-14 07:51:54,704 - DeyeDaemon - DEBUG - PV2 Production today: 1.1
2023-01-14 07:51:54,705 - DeyeDaemon - DEBUG - PV2 Total: 1.1
2023-01-14 07:51:54,705 - DeyeDaemon - DEBUG - PV3 Voltage: 0.0
2023-01-14 07:51:54,705 - DeyeDaemon - DEBUG - PV3 Current: 0.0
2023-01-14 07:51:54,706 - DeyeDaemon - DEBUG - PV3 Power: 0.0
2023-01-14 07:51:54,706 - DeyeDaemon - DEBUG - PV3 Production today: 0.0
2023-01-14 07:51:54,707 - DeyeDaemon - DEBUG - PV3 Total: 0.0
2023-01-14 07:51:54,707 - DeyeDaemon - DEBUG - PV4 Voltage: 0.0
2023-01-14 07:51:54,707 - DeyeDaemon - DEBUG - PV4 Current: 0.0
2023-01-14 07:51:54,707 - DeyeDaemon - DEBUG - PV4 Power: 0.0
2023-01-14 07:51:54,708 - DeyeDaemon - DEBUG - PV4 Production today: 0.0
2023-01-14 07:51:54,708 - DeyeDaemon - DEBUG - PV4 Total: 0.0
2023-01-14 07:51:54,709 - DeyeDaemon - DEBUG - DC Total Power: 0.0
2023-01-14 07:51:54,709 - DeyeDaemon - DEBUG - Operating Power: 0.0
2023-01-14 07:51:54,709 - DeyeDaemon - DEBUG - AC Apparent Power: 0.0
2023-01-14 07:51:54,709 - DeyeDaemon - DEBUG - AC Active Power: 8.0
2023-01-14 07:51:54,710 - DeyeDaemon - DEBUG - AC Reactive Power: 0.0
2023-01-14 07:51:54,710 - DeyeDaemon - DEBUG - Radiator temperature: 62.0
2023-01-14 07:51:54,710 - DeyeDaemon - DEBUG - IGBT temperature: -100.0
2023-01-14 07:51:54,711 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'deye_inverter'
2023-01-14 07:51:54,714 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/day_energy', value: '2.5'
2023-01-14 07:51:54,714 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m1), 'b'deye/day_energy'', ... (3 bytes)
2023-01-14 07:51:54,715 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-01-14 07:51:54,716 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2023-01-14 07:51:54,717 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/total_energy', value: '2.5'
2023-01-14 07:51:54,719 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'deye/total_energy'', ... (3 bytes)
2023-01-14 07:51:54,722 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
2023-01-14 07:51:54,723 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/voltage', value: '237.0'
2023-01-14 07:51:54,724 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m3), 'b'deye/ac/l1/voltage'', ... (5 bytes)
2023-01-14 07:51:54,726 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 3)
2023-01-14 07:51:54,727 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/current', value: '0.0'
2023-01-14 07:51:54,728 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m4), 'b'deye/ac/l1/current'', ... (3 bytes)
2023-01-14 07:51:54,730 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 4)
2023-01-14 07:51:54,730 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/l1/power', value: '0.0'
2023-01-14 07:51:54,731 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m5), 'b'deye/ac/l1/power'', ... (3 bytes)
2023-01-14 07:51:54,732 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 5)
2023-01-14 07:51:54,733 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/ac/freq', value: '50.0'
[... cut ...]
deye/dc/pv2/total_energy 1.1
deye/dc/pv3/voltage 0.0
deye/dc/pv3/current 0.0
deye/dc/pv3/power 0.0
deye/dc/pv3/day_energy 0.0
deye/dc/pv3/total_energy 0.0
deye/dc/pv4/voltage 0.0
deye/dc/pv4/current 0.0
deye/dc/pv4/power 0.0
deye/dc/pv4/day_energy 0.0
deye/dc/pv4/total_energy 0.0
deye/dc/total_power 0.0
deye/operating_power 0.0
deye/ac/apparent_power 0.0
deye/ac/active_power 8.0
deye/ac/reactive_power 0.0
deye/radiator_temp 62.0
deye/igbt_temp -100.0
```
@kbialek: Thanks a lot! Now go on to the next release. ;-)

closed 2023-01-14 by kbialek