kbialek/deye-inverter-mqtt #16: Timeouts on Deye SUN600G3-EU-230

ahans, 2023-01-14:
First up: awesome project, works out of the box with my Deye SUN600G3-EU-230. Really great work! However, I get occasional timeouts and errors. I was wondering if this is expected? When using the inverter's web interface, there are also times when nothing is displayed and after a number of refreshs things are fine again. So what I see in the log could be related to that. Any idea what's going on?

Log looks like this:
```
2023-01-14 13:06:55,561 - DeyeDaemon - INFO - Reading start
2023-01-14 13:07:05,635 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:07:15,646 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:07:25,657 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:07:35,663 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:07:38,774 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:08:38,831 - DeyeDaemon - INFO - Reading start
2023-01-14 13:08:50,931 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:09:50,988 - DeyeDaemon - INFO - Reading start
2023-01-14 13:09:59,429 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:10:59,489 - DeyeDaemon - INFO - Reading start
2023-01-14 13:11:09,512 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:11:19,519 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:11:29,530 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:11:39,540 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:11:49,551 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:11:49,551 - DeyeModbus - ERROR - No response frame
2023-01-14 13:11:49,552 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-14 13:11:59,571 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:12:02,529 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:13:02,579 - DeyeDaemon - INFO - Reading start
2023-01-14 13:13:02,942 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:14:03,003 - DeyeDaemon - INFO - Reading start
2023-01-14 13:14:06,933 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:15:06,994 - DeyeDaemon - INFO - Reading start
2023-01-14 13:15:07,303 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:16:07,312 - DeyeDaemon - INFO - Reading start
2023-01-14 13:16:16,174 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:17:16,227 - DeyeDaemon - INFO - Reading start
2023-01-14 13:17:26,259 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:17:36,269 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:17:46,280 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:17:56,291 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:18:06,302 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:18:06,302 - DeyeModbus - ERROR - No response frame
2023-01-14 13:18:06,302 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-14 13:18:14,029 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:19:14,088 - DeyeDaemon - INFO - Reading start
2023-01-14 13:19:25,387 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:19:35,398 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:19:45,408 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:19:55,419 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:20:05,419 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:20:05,420 - DeyeModbus - ERROR - No response frame
2023-01-14 13:20:05,420 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-14 13:20:05,427 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:21:05,487 - DeyeDaemon - INFO - Reading start
2023-01-14 13:21:18,465 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:22:18,526 - DeyeDaemon - INFO - Reading start
2023-01-14 13:22:22,353 - DeyeDaemon - INFO - Reading completed
2023-01-14 13:23:22,392 - DeyeDaemon - INFO - Reading start
2023-01-14 13:23:32,413 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:23:42,424 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:23:52,435 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:24:02,443 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:24:12,454 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:24:12,454 - DeyeModbus - ERROR - No response frame
2023-01-14 13:24:12,454 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-14 13:24:22,471 - DeyeConnector - WARNING - Connection timeout
2023-01-14 13:24:25,435 - DeyeDaemon - INFO - Reading completed
```

Antagor, 2023-01-14:
Hello, 
It looks the same on mine.... 
I expect it is a problem of the deye. 
@kbialek: maybe you can try to "warm"  it up 😀 just try to change the order for the three requests. Or wait before the first request. 
Thanks

kbialek, 2023-01-14:
@ahans I can confirm, that these timeouts appear from time to time. I suppose that the logger is "sometimes" busy with doing something else, and ignores modbus requests. Probably it talks to the inverter at that time. I think that is something we have to accept. The daemon is already prepared for that and retries each modbus request at most 5 times when there are timeouts. Perhaps I could rework logging a little bit to only report a warning when all 5 requests fail. That should reduce the number of warnings in the log, but will not solve the underlying problem of course.

closed 2023-01-16 by kbialek

kbialek, 2023-01-16:
> What was the problem?

I've deleted the arm64 arch docker image by mistake

> If it is ok for you I would advise your tool in some blocks as a good alternative to China cloud

Sure, feel free to do it.