kbialek/deye-inverter-mqtt #19: Wrong temperature conversation

geeks-r-us, 2023-01-20:
Hi,

just updated to newest docker container and recognized it shows radiator temp of 157°C.

App tells me it's 15.70°C.

So I assume there is a need of division by 10.

I'm running a mirco. Is it the same for string? 
Or do we need here a type dependent conversion?

kbialek, 2023-01-20:
Hi @geeks-r-us 

It's very likely caused by the change implemented in this PR https://github.com/kbialek/deye-inverter-mqtt/pull/18/
@Antagor has suggested that the conversion shall be different for micro inverters. Now it turns out that it's not entirely true.

Antagor, 2023-01-20:
Hello,
@geeks-r-us : Have you updated your enviroment for the inverter-type ?
Which inverter do you have?
Can it be a problem of the firmware? that deye has changed the conversation of the temp?
Wich temp you had before our change? It couldn't be right as we have added an additional divisor for micro.
bye Antagor

geeks-r-us, 2023-01-21:
Hi
@Antagor  yes I set it to micro.

I have the Deye SUN1300G3  it's on:
* communication protocol: 0.2.0.1
* DC Master: 0.0.0.4
* DC Slave: 0.2.0.1
* AC: 0.1.0.0

After a look in my influx bucket it seem temperature was not right before the update but haven't had it on my dashboard...

Antagor, 2023-01-22:
@geeks-r-us 
Hey 
Can you send a debug log. It makes no sense to me: we added an additional divisivor for micro, because I had the same problem. So the new version should divide by 100 for micro. 
Do you have daily and total values for the panels? Otherwise you have an old image. 
Antagor