kbialek/deye-inverter-mqtt #20: Question: deye micro current limiting

gianricod, 2023-01-21:
Hello thanks a lot for this work

I know that deye micro can be configured to limit the power. 
https://www.youtube.com/watch?v=j9x_cyishwY&t=332s  

I need that function to create a sort of virtual limiter to avoid feeding too much to the grid in some cases. 

It can be done in solarman if you have device control permission (i a waiting the chinese new year to finish to hit the distibutor and deye and obtain that). 

I wonder if you know the modbus register that must be written  to achieve the same via your tool.

Many thanks
gianrico

kbialek, 2023-01-21:
Hi @gianricod 

You can try this:
1. Get active power regulation factor:
`docker run --rm --env-file config.env deye-inverter-mqtt r 40`

2. Set active power regulation factor to 100%:
`docker run --rm --env-file config.env deye-inverter-mqtt w 40 1000`

gianricod, 2023-01-22:
thanks a lot
i'll give it a try asap

seepv, 2023-01-28:
Hi @gianricod 

could you limit the current of your deye micro ?

Does this Modbus-Write-Command:
`docker run --rm --env-file config.env deye-inverter-mqtt w 40 1000`
work for you or do you get an error?

gianricod, 2023-01-28:
hello @seepv  
i have not tried that yet 
i will post here as soon as i test that
what kind of errors you see?

seepv, 2023-01-28:
No errors anymore.

With [pysolarmanv5](https://github.com/jmccrohan/pysolarmanv5) you could limit a Micro-Inverter.

If you like to limit an Deye-Microinverter SunX00G3 with Device S/N 41xxxxxxxx, take a look at:
[https://github.com/jmccrohan/pysolarmanv5/issues/24](https://github.com/jmccrohan/pysolarmanv5/issues/24)

closed 2023-02-06 by gianricod