kbialek/deye-inverter-mqtt #21: DeyeModbus - ERROR - Modbus frame is too short or empty

slev1n, 2023-01-29:
I'm using a Deye SUN1000G3-EU-230
with Datalogger Serial 41684xxxxx
Firmware version
MW3_15U_5406_1.49R

I'm getting this error;
2023-01-29 08:51:01,911 - DeyeDaemon - INFO - Reading start
2023-01-29 08:51:01,944 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc830200000000000000000000000000000103003c001485c95315
2023-01-29 08:51:01,949 - DeyeConnector - DEBUG - Response frame: a51000101500c9c22576f80201032d0000790800007106d6630600bd15
2023-01-29 08:51:01,950 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 08:51:01,950 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 08:51:01,958 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc8302000000000000000000000000000001030050001044177015
2023-01-29 08:51:01,978 - DeyeConnector - DEBUG - Response frame: a51000101500cac22576f80201032d0000790800007106d6630600be15
2023-01-29 08:51:01,978 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 08:51:01,979 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 08:51:01,982 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc830200000000000000000000000000000103006d0008d5d1d015
2023-01-29 08:51:01,997 - DeyeConnector - DEBUG - Response frame: a51000101500cbc22576f80201032d0000790800007106d6630600bf15
2023-01-29 08:51:01,998 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 08:51:01,998 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 08:51:01,999 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'deye_inverter'
2023-01-29 08:51:02,026 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-01-29 08:51:02,026 - paho.mqtt.client - DEBUG - Sending DISCONNECT
2023-01-29 08:51:02,027 - DeyeDaemon - INFO - Reading completed

Any idea whats wrong here?

Thanks!

kbialek, 2023-01-29:
hi @slev1n 

That is interesting. Thanks for reporting.

Perhaps your logger uses modbus address that is not equal to 1. So far I assumed that it is always 1.
I've changes the address in the code to 2 and tested it with my inverter locally. Obviously it does not work and I get similar and short responses as you did.
The difference I see, is that your logger returned `06` instead of so far expected `01` in the address field, while mine returns `05` there.
a51000101500c9c22576f80201032d0000790800007106d663**06**00bd15  

If you are eager to dig more into that, I can build a beta version that will allow to specify the modbus address in the configuration file.

slev1n, 2023-01-29:
@kbialek A beta would be nice, thanks for your fast support:)

kbialek, 2023-01-29:
@slev1n 

There you have: `ghcr.io/kbialek/deye-inverter-mqtt:2023.01.3-beta1`

Add `DEYE_MODBUS_ADDRESS` to your `config.env`

Good luck!

slev1n, 2023-01-29:
@kbialek 
The error messages with `DEYE_MODBUS_ADDRESS=2` changed a little bit, but still no values

`docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt:2023.01.3-beta1
2023-01-29 13:45:22,063 - DeyeDaemon - INFO - Reading start
2023-01-29 13:45:22,097 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc830200000000000000000000000000000203003c001485fa8515
2023-01-29 13:45:26,415 - DeyeConnector - DEBUG - Response frame: a5100010150038c22576f80201f07100003a2300008506d66306004d15
2023-01-29 13:45:26,416 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 13:45:26,416 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 13:45:26,427 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc8302000000000000000000000000000002030050001044247e15
2023-01-29 13:45:26,515 - DeyeConnector - DEBUG - Response frame: a5100010150038c22576f80201f07100003a2300008506d66306004d15
2023-01-29 13:45:26,516 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 13:45:26,517 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 13:45:26,529 - DeyeConnector - DEBUG - Request frame: a5170010450000b450bc830200000000000000000000000000000203006d0008d5e2e215
2023-01-29 13:45:26,539 - DeyeConnector - DEBUG - Response frame: a510001015003ac22576f80201f17100003a2300008506d66306005015
2023-01-29 13:45:26,540 - DeyeModbus - ERROR - Response frame is too short
2023-01-29 13:45:26,540 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-01-29 13:45:26,542 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'deye_inverter'
2023-01-29 13:45:26,546 - paho.mqtt.client - DEBUG - Received CONNACK (1, 0)
2023-01-29 13:45:26,547 - paho.mqtt.client - DEBUG - Sending DISCONNECT
2023-01-29 13:45:26,548 - DeyeDaemon - INFO - Reading completed
`

kbialek, 2023-01-29:
@slev1n Please try with other addresses too

slev1n, 2023-01-29:
@kbialek tried 0-9 without success :(

slev1n, 2023-01-29:
@kbialek OMG. My fault: unfortunately i used the inverter serial instead the device serial :man_facepalming:

kbialek, 2023-01-29:
@slev1n So you say it is working now, right?

slev1n, 2023-01-29:
@kbialek Yes, with standard address 1, sry for your time waste. Can be closed.

kbialek, 2023-01-29:
Haha, no worries :smiley:

closed 2023-01-29 by kbialek