kbialek/deye-inverter-mqtt #25: Response/Modbus frame is too short - Deye SUN600G3-230-EU

Diavolezza, 2023-02-05:
Hello,
after installing and running I get the following output:
2023-02-05 12:26:13,094 - DeyeDaemon - INFO - Reading start
2023-02-05 12:26:14,937 - DeyeModbus - ERROR - Response frame is too short
2023-02-05 12:26:14,938 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-02-05 12:26:14,959 - DeyeModbus - ERROR - Response frame is too short
2023-02-05 12:26:14,959 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-02-05 12:26:14,983 - DeyeModbus - ERROR - Response frame is too short
2023-02-05 12:26:14,983 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-02-05 12:26:15,990 - DeyeDaemon - INFO - Reading completed

My config file is:
DEYE_LOGGER_IP_ADDRESS=192.168.2.149
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=#####my number######
MQTT_HOST=127.0.0.1
MQTT_PORT=1883
MQTT_USERNAME=deye
MQTT_PASSWORD=####my mqtt password#####
MQTT_TOPIC_PREFIX=deye
LOG_LEVEL=INFO
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=micro

My inverter is a Deye SUN600G3-230-EU

What am I doing wrong?
Thank you!

kbialek, 2023-02-05:
Please double check if your `DEYE_LOGGER_SERIAL_NUMBER` is correct. Also see this issue https://github.com/kbialek/deye-inverter-mqtt/issues/21 Otherwise please enable debug logs and share here.

Diavolezza, 2023-02-06:
Thank you for the hint. I provided the inverter number, not that of the logger.

closed 2023-02-06 by Diavolezza