kbialek/deye-inverter-mqtt #27: Micropython port of your code now under test

JBlohm, 2023-02-08:
Dear Krzysztof,

I have ported your code to Micropython on an ESP8266.
Code ist under test now. Probably not finished and beautiful yet, but basic functionality is there.
Have fun sharing: https://github.com/JBlohm/miropython-deye-inverter-mqtt-esp8266

Regards
Joern

kbialek, 2023-02-08:
@JBlohm Nice! Thanks for the heads up! How big is the scope of change? I see you had to replace the logging library.

JBlohm, 2023-02-08:
Well, Micropython is really micro, I had to replace logging (with kind of a stub lib), some changes in handling dictionaries (see sensors and daemon) and I needed to replace MQTT handling with std. umqtt.simple lib. Error handling is not as sophisticated as in full Python3. But considering the (intentionally chosen) minimalist ESP8266 the porting worked well. Interestingly there was no need to change the socket stuff. Your code is well written, which helped here. This whole idea was more like "Let's see if this is possible" kind of project. It worked out in the end. Took me about 6-8 hours up until now. I intend to fix bugs as they popup (not sure if the code currently consumes memory or so) and maybe back-port changes you will do in the future in case it fits into this small microcontroller. (New sensors maybe.) And I have some more ideas in the queue with register 40(dec). But that has to wait until batteries get cheaper.

closed 2023-03-26 by github-actions[bot]