kbialek/deye-inverter-mqtt #28: Support availability topic

ConsoleCriminal, 2023-02-15:
Publish the reachability of the inverter on an MQTT topic so that stale values are visible to be stale for consumers.
Also, probably publish online/offline state of dye-inverter-mqtt so that it is visible to decide whether the inverter is just unreachable (because there's no sun) or deye-inverter-mqtt is actually not running.

ConsoleCriminal, 2023-02-15:
(Will work on that and forked for the purpose.)

kbialek, 2023-02-15:
@ConsoleCriminal How do you plan to implement it? I can think of at least two separate statuses:
1. daemon status - online/offline
2. logger connectivity status - connected/disconnected
What is your approach?

ConsoleCriminal, 2023-02-15:
I used to think - piece of cake, set last will topic for the one and update
the status on success talking to the inverter or socket exception when
unreachable. After reviewing the code, I feel both is a larger venture,
though. :(

kbialek, 2023-02-15:
Imho, adding a "classic" availability topic with retained last will would be a good start.
The current implementation is over simplified. It connects to the mqtt broker only when metrics are published.
This would have to be changed. The connection must be created once, with appropriate keep-alive setting and then the paho loop must be started. I'm not sure if paho will automatically recover after losing connection to the broker. That would have to be tested too. So yes, a longer trip that it seems to be at the first sight :slightly_smiling_face:

ConsoleCriminal, 2023-02-16:
paho does that quite well. However, time is an issue this time of year as my inverter powers down when there's not enough sunlight. Which at present opens a time window during lunch break only. Might be moving this to summer. :(

kbialek, 2023-03-05:
fyi, I've implemented the status topics :slightly_smiling_face: See https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.03.1

closed 2023-04-19 by github-actions[bot]