kbialek/deye-inverter-mqtt #30: Azure IOT Hub Support

philippbussche, 2023-02-24:
Hi, 
I was doing some research on Azure's IOT Hub capabilities and since I also own a Deye Sun 600 I made some adjustments to send the inverter data to Azure's IOT Hub. I am considering this only as experimental work but maybe somebody else finds this useful. 
Also I added a switch to simulate the communication with the inverter as I realized I could otherwise not work on this after 5pm ;)

https://github.com/kbialek/deye-inverter-mqtt/compare/main...philippbussche:deye-inverter-mqtt:feature/azureiothub

Thanks
Philipp.

kbialek, 2023-02-24:
Hi! @philippbussche 

Thanks for sharing. It looks interesting. 
In the essence, I see the following changes implemented in your fork:
1. Add SSL support
2. Use alternative mqtt message format and topic naming
3. Add option to specify topic suffix
4. Use configurable mqtt client id

Besides of p.2. all other points are rather generic. Perhaps you'd like to submit them as PRs?

closed 2023-04-11 by github-actions[bot]

kbialek, 2023-04-11:
@philippbussche It should be easier to implement such new features with the current, a bit more modular codebase. I'm happy to provide you guidance in this regard when you will find time.