kbialek/deye-inverter-mqtt #31: SOC%

michelissa, 2023-03-04:


kbialek, 2023-03-04:
What "SOC%" is?

michelissa, 2023-03-04:
It is the percentage of the batteries

kbialek, 2023-03-05:
If you are willing to help, then we can try. First we have to identify registers that are holding this value. According to the spec that might me regs 184,185. I assume, that you have already created your `config.env` and the service works for you. If so, please run these commands and send me the results. 

```
docker run --rm --env-file config.env deye-inverter-mqtt r 184
docker run --rm --env-file config.env deye-inverter-mqtt r 185
```

kbialek, 2023-03-16:
These values does not seem to have much sense to me. SOC registers differ depending on the inverter type. What is your inverter type @hackepeterOli ?

hackepeterOli, 2023-03-16:
@kbialek The inverter is: SUN-10K-SG04LP3-EU.
The other values I get are unfortunately not correct either:

active_power | 429496729.5 | 2023-03-16 14:45:12
apparent_power | 6553.5 | 2023-03-16 14:45:12
current | 5.0 | 2023-03-16 14:45:12
day_energy | 25.5 | 2023-03-16 14:45:12
freq | 10.0 | 2023-03-16 14:45:12
igbt_temp | -100.0 | 2023-03-16 14:45:12
logger_status | online | 2023-03-16 14:45:12
operating_power | 0.1 | 2023-03-16 14:45:12
power | 5.0 | 2023-03-16 14:45:12
radiator_temp | 6453.5 | 2023-03-16 14:45:12
reactive_power | 6553.5 | 2023-03-16 14:45:12
status | online | 2023-03-15 22:01:40
total_energy | 74141287.8 | 2023-03-16 14:45:12
total_power | 6553.5 | 2023-03-16 14:45:12
uptime | 5891.0 | 2023-03-16 14:45:12
voltage | 1.0 | 2023-03-16 14:45:12

kbialek, 2023-03-16:
@hackepeterOli 

Let's try the following:
1. Daily energy production:
`docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 529`

2. Battery SOC:
`docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 588`

hackepeterOli, 2023-03-16:
Perfect, that looks great:
root@rofhem:~/deye-inverter-mqtt# docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 529
int: 147, l: 147, h: 0
root@rofhem:~/deye-inverter-mqtt# docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 588
int: 96, l: 96, h: 0

SOC is 95,6, energy production is 14,7. Could you get the other values also? - and maybe create a configuration for this type of inverter?

kbialek, 2023-03-16:
@hackepeterOli Yes, adding it to my todo list. It will report more or less the metrics defined here https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/deye_sg04lp3.yaml

kbialek, 2023-03-18:
@hackepeterOli In case you are still interested in, and not yet moved to HomeAssistant :slightly_smiling_face: 
here is an experimental docker image with SUN-10K-SG04LP3-EU support

Image ID: `ghcr.io/kbialek/deye-inverter-mqtt:2023.03.3-exp-sg04lp3`

In your `config.env` set `DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery`

Please test it and let me know if it works. Especially if all topics report reasonable values.

hackepeterOli, 2023-03-18:
@kbialek It works, you are the best! Can i donate your work a little bit?

current | 14.7 | 2023-03-18 19:13:26
daily_charge | 6.9 | 2023-03-18 19:13:26
daily_discharge | 7.7 | 2023-03-18 19:13:26
daily_energy_bought | 5.5 | 2023-03-18 19:13:26
daily_energy_sold | 0.3 | 2023-03-18 19:13:26
day_energy | 12.2 | 2023-03-18 19:13:26
logger_status | online | 2023-03-18 19:13:26
power | 765.0 | 2023-03-18 19:13:26
radiator_temp | 125.0 | 2023-03-18 19:13:26
soc | 39.0 | 2023-03-18 19:13:26
status | online | 2023-03-18 19:09:19
temperature | 118.2 | 2023-03-18 19:13:26
total_charge | 6.9 | 2023-03-18 19:13:26
total_discharge | 13.2 | 2023-03-18 19:13:26
total_energy | 12.3 | 2023-03-18 19:13:26
total_energy_bought | 5.6 | 2023-03-18 19:13:26
total_energy_sold | 0.3 | 2023-03-18 19:13:26
total_power | 14.0 | 2023-03-18 19:12:02
voltage | 52.0 | 2023-03-18 19:13:26

kbialek, 2023-03-18:
@hackepeterOli 

> Can i donate your work a little bit?

Haha. Thanks. Star the project and spread the word.

The table you posted doesn't show full mqtt topic names, so I'm not sure how did you map that. Anyway, I see the following potential issues:
1. The temperatures are off by 100
2. Totals are pretty small. Does it match the values you see in the Solarman app?

hackepeterOli, 2023-03-18:
@kbialek Thank you very much! Your work helps with the energy transition! Solar power is currently only supplied via the gen port from a second deye inverter. I will complete the system in the next few days.

I simply send the values via mqtt to a FHEM installation.

current = ok i think
daily_charge = ok
daily_discharge = ok
daily_energy_bought = ok
daily_energy_sold = ok
day_energy = ok
logger_status = ok
power = maybe ok
radiator_temp not ok
soc = ok
status = ok
temperature = not ok
total_charge = ok
total_discharge = i can not find it in solarman
total_energy = ok
total_energy_bought = ok
total_energy_sold = ok
total_power =  i can not find it in solarman
voltage = ok

kbialek, 2023-03-19:
@hackepeterOli Hey! Try this beta image. The temperature metrics should be looking good now. `ghcr.io/kbialek/deye-inverter-mqtt:2023.03.3-beta-1` I will release a new version, once you confirm it works fine

hackepeterOli, 2023-03-19:
@kbialek You are my man, now it looks good:

current | 0.0 | 2023-03-19 19:08:34
daily_charge | 4.8 | 2023-03-19 19:08:34
daily_discharge | 6.3 | 2023-03-19 19:08:34
daily_energy_bought | 9.0 | 2023-03-19 19:08:34
daily_energy_sold | 0.2 | 2023-03-19 19:08:34
day_energy | 8.3 | 2023-03-19 19:08:34
logger_status | online | 2023-03-19 19:08:34
power | 44.0 | 2023-03-19 19:08:34
radiator_temp | 25.0 | 2023-03-19 19:08:34
soc | 25.0 | 2023-03-19 19:08:34
status | online | 2023-03-19 19:08:34
temperature | 17.0 | 2023-03-19 19:08:34
total_charge | 13.1 | 2023-03-19 19:08:34
total_discharge | 21.6 | 2023-03-19 19:08:34
total_energy | 20.6 | 2023-03-19 19:08:34
total_energy_bought | 17.5 | 2023-03-19 19:08:34
total_energy_sold | 0.6 | 2023-03-19 19:08:34
total_power | 713.0 | 2023-03-19 19:08:34
voltage | 234.9 | 2023-03-19 19:08:34

Here is a Solarman Image, what i am miss is "Produktion" and/or "Verbrauch heute"
[https://abload.de/img/solarmanneda1.jpg](url)

kbialek, 2023-03-19:
@hackepeterOli I think that your mqtt client squashes topics having the same suffix, e.g. `current` is shown only once, but there are multiple topcis ending with `/current`. Basically you should see all metrics that are documented here:
* https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg04lp3.md
* https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg04lp3_battery.md

Can you check your mqtt client? Or alternatively, try out `mosquitto_sub` command line client?

> Here is a Solarman Image, what i am miss is "Produktion" and/or "Verbrauch heute"

This we can check after you make all mqtt topics properly visible in the client.

hackepeterOli, 2023-03-19:
I have checked the logs...:

2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 logger_status: online
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 0.0
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 0.0
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 14.5
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 7.3
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 0.0
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 0.0
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 day_energy: 8.3
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_energy: 20.6
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 daily_charge: 4.8
2023-03-19 19:40:58 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 daily_discharge: 6.3
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_charge: 13.1
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_discharge: 21.6
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 15.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 52.2
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 soc: 25.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 0.3
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 temperature: 16.9
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_power: 947.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 235.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 236.2
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 voltage: 235.2
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 daily_energy_bought: 9.6
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_energy_bought: 18.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 daily_energy_sold: 0.2
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 total_energy_sold: 0.6
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 0.1
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 0.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 current: 654.4
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 58.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 47.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 power: 65348.0
2023-03-19 19:40:59 MQTT2_DEVICE MQTT2_deye_inverter_2388186981 radiator_temp: 25.0

and fixed the problem:
ac/daily_energy_bought | 10.0 | 2023-03-19 20:07:27
ac/daily_energy_sold | 0.2 | 2023-03-19 20:07:27
ac/l1/voltage/current | 0.1 | 2023-03-19 20:07:27
ac/l1/voltage/power | 59.0 | 2023-03-19 20:07:27
ac/l1/voltage/voltage | 235.5 | 2023-03-19 20:07:27
ac/l2/voltage/voltage | 236.5 | 2023-03-19 20:07:27
ac/l2/voltage/voltage/current | 655.1 | 2023-03-19 20:07:27
ac/l2/voltage/voltage/power | 65464.0 | 2023-03-19 20:07:27
ac/l3/voltage/voltage/current | 654.9 | 2023-03-19 20:07:27
ac/l3/voltage/voltage/power | 65461.0 | 2023-03-19 20:07:27
ac/l3/voltage/voltage/voltage | 235.5 | 2023-03-19 20:07:27
ac/total_energy_bought | 18.5 | 2023-03-19 20:07:27
ac/total_energy_sold | 0.6 | 2023-03-19 20:07:27
ac/total_power | 835.0 | 2023-03-19 20:07:27
battery/current | 0.5 | 2023-03-19 20:07:27
battery/daily_charge | 4.8 | 2023-03-19 20:07:27
battery/daily_discharge | 6.3 | 2023-03-19 20:07:27
battery/power | 24.0 | 2023-03-19 20:07:27
battery/soc | 25.0 | 2023-03-19 20:07:27
battery/temperature | 16.8 | 2023-03-19 20:07:27
battery/total_charge | 13.1 | 2023-03-19 20:07:27
battery/total_discharge | 21.6 | 2023-03-19 20:07:27
battery/voltage | 52.2 | 2023-03-19 20:07:27
day_energy | 8.3 | 2023-03-19 20:07:27
dc/pv1/current | 0.0 | 2023-03-19 20:07:27
dc/pv1/power | 0.0 | 2023-03-19 20:07:27
dc/pv1/voltage | 14.6 | 2023-03-19 20:07:27
dc/pv2/current | 0.0 | 2023-03-19 20:07:27
dc/pv2/power | 0.0 | 2023-03-19 20:07:27
dc/pv2/voltage | 7.4 | 2023-03-19 20:07:27
logger_status | online | 2023-03-19 20:07:27
radiator_temp | 25.0 | 2023-03-19 20:07:27
total_energy | 20.6 | 2023-03-19 20:07:27

 i am miss still "Produktion" and/or "Verbrauch heute"

kbialek, 2023-03-19:
@hackepeterOli These rows make no sense to me, can you check it?

> ac/l2/voltage/voltage/current | 655.1 | 2023-03-19 20:07:27
> ac/l2/voltage/voltage/power | 65464.0 | 2023-03-19 20:07:27
> ac/l3/voltage/voltage/current | 654.9 | 2023-03-19 20:07:27
> ac/l3/voltage/voltage/power | 65461.0 | 2023-03-19 20:07:27
> ac/l3/voltage/voltage/voltage | 235.5 | 2023-03-19 20:07:27

"Produktion" is here most likely

> day_energy | 8.3 | 2023-03-19 20:07:27

And  "Verbrauch heute" is here

> ac/total_energy_bought | 18.5 | 2023-03-19 20:07:27

Does it make sense?

hackepeterOli, 2023-03-19:
@kbialek - I had a naming problem and fixed it: 

ac/daily_energy_bought | 10.5 | 2023-03-19 20:36:12
ac/daily_energy_sold | 0.2 | 2023-03-19 20:36:12
ac/l1/current | 0.1 | 2023-03-19 20:36:12
ac/l1/power | 58.0 | 2023-03-19 20:36:12
ac/l1/voltage | 235.0 | 2023-03-19 20:36:12
ac/l2/current | 654.4 | 2023-03-19 20:36:12
ac/l2/power | 65349.0 | 2023-03-19 20:36:12
ac/l2/voltage | 235.9 | 2023-03-19 20:36:12
ac/l3/current | 0.0 | 2023-03-19 20:36:12
ac/l3/power | 46.0 | 2023-03-19 20:36:12
ac/l3/voltage | 235.4 | 2023-03-19 20:36:12
ac/total_energy_bought | 19.0 | 2023-03-19 20:36:12
ac/total_energy_sold | 0.6 | 2023-03-19 20:36:12
ac/total_power | 695.0 | 2023-03-19 20:36:12
battery/current | 0.3 | 2023-03-19 20:36:12
battery/daily_charge | 4.8 | 2023-03-19 20:36:11
battery/daily_discharge | 6.4 | 2023-03-19 20:36:11
battery/power | 17.0 | 2023-03-19 20:36:11
battery/soc | 25.0 | 2023-03-19 20:36:12
battery/temperature | 16.7 | 2023-03-19 20:36:12
battery/total_charge | 13.1 | 2023-03-19 20:36:11
battery/total_discharge | 21.6 | 2023-03-19 20:36:11
battery/voltage | 52.2 | 2023-03-19 20:36:12
day_energy | 8.3 | 2023-03-19 20:36:11
dc/pv1/current | 0.0 | 2023-03-19 20:36:11
dc/pv1/power | 0.0 | 2023-03-19 20:36:11
dc/pv1/voltage | 14.5 | 2023-03-19 20:36:11
dc/pv2/current | 0.0 | 2023-03-19 20:36:11
dc/pv2/power | 0.0 | 2023-03-19 20:36:11
dc/pv2/voltage | 7.5 | 2023-03-19 20:36:11
logger_status | online | 2023-03-19 20:36:11
radiator_temp | 25.0 | 2023-03-19 20:36:12
total_energy | 20.6 | 2023-03-19 20:36:11

but this Values are not right:
ac/l2/current | 655.1 | 2023-03-19 20:07:27
ac/l2/power | 65464.0 | 2023-03-19 20:07:27
ac/l3/current | 654.9 | 2023-03-19 20:07:27
ac/l3/power | 65461.0 | 2023-03-19 20:07:27
ac/l3/voltage | 235.5 | 2023-03-19 20:07:27

No, "Produktion" should be 3.29, this is "Produktion-Heute":
day_energy | 8.3 | 2023-03-19 20:07:27

It is too late for me, and i have corona :(
You are right with this:
And "Verbrauch heute" is here
ac/total_energy_bought | 18.5 | 2023-03-19 20:07:27

hackepeterOli, 2023-03-19:
@kbialek 

this is my mapping:

`deye_inverter_2388186981:deye/logger_status:.* logger_status
deye_inverter_2388186981:deye/day_energy:.* day_energy
deye_inverter_2388186981:deye/total_energy:.* total_energy
deye_inverter_2388186981:deye/ac/l1/voltage:.* ac/l1/voltage
deye_inverter_2388186981:deye/ac/l1/current:.* ac/l1/current
deye_inverter_2388186981:deye/ac/l1/power:.* ac/l1/power
deye_inverter_2388186981:deye/ac/l2/voltage:.* ac/l2/voltage
deye_inverter_2388186981:deye/ac/l2/current:.* ac/l2/current
deye_inverter_2388186981:deye/ac/l2/power:.* ac/l2/power
deye_inverter_2388186981:deye/ac/l3/voltage:.* ac/l3/voltage
deye_inverter_2388186981:deye/ac/l3/current:.* ac/l3/current
deye_inverter_2388186981:deye/ac/l3/power:.* ac/l3/power
deye_inverter_2388186981:deye/ac/freq:.* ac/freq
deye_inverter_2388186981:deye/uptime:.* uptime
deye_inverter_2388186981:deye/dc/pv1/voltage:.* dc/pv1/voltage
deye_inverter_2388186981:deye/dc/pv1/current:.* dc/pv1/current
deye_inverter_2388186981:deye/dc/pv1/power:.* dc/pv1/power
deye_inverter_2388186981:deye/dc/pv2/voltage:.* dc/pv2/voltage
deye_inverter_2388186981:deye/dc/pv2/current:.* dc/pv2/current
deye_inverter_2388186981:deye/dc/pv2/power:.* dc/pv2/power
deye_inverter_2388186981:deye/dc/pv3/voltage:.* dc/pv3/voltage
deye_inverter_2388186981:deye/dc/pv3/current:.* dc/pv3/current
deye_inverter_2388186981:deye/dc/pv3/power:.* dc/pv3/power
deye_inverter_2388186981:deye/dc/pv4/voltage:.* dc/pv4/voltage
deye_inverter_2388186981:deye/dc/pv4/current:.* dc/pv4/current
deye_inverter_2388186981:deye/dc/pv4/power:.* dc/pv4/power
deye_inverter_2388186981:deye/dc/total_power:.* dc/total_power
deye_inverter_2388186981:deye/operating_power:.* operating_power
deye_inverter_2388186981:deye/ac/apparent_power:.* ac/apparent_power
deye_inverter_2388186981:deye/ac/active_power:.* ac/active_power
deye_inverter_2388186981:deye/ac/reactive_power:.* ac/reactive_power
deye_inverter_2388186981:deye/radiator_temp:.* radiator_temp
deye_inverter_2388186981:deye/igbt_temp:.* igbt_temp
deye_inverter_2388186981:deye/status:.* status
deye_inverter_2388186981:deye/battery/daily_charge:.* battery/daily_charge
deye_inverter_2388186981:deye/battery/daily_discharge:.* battery/daily_discharge
deye_inverter_2388186981:deye/battery/total_charge:.* battery/total_charge
deye_inverter_2388186981:deye/battery/total_discharge:.* battery/total_discharge
deye_inverter_2388186981:deye/battery/power:.* battery/power
deye_inverter_2388186981:deye/battery/voltage:.* battery/voltage
deye_inverter_2388186981:deye/battery/soc:.* battery/soc
deye_inverter_2388186981:deye/battery/current:.* battery/current
deye_inverter_2388186981:deye/battery/temperature:.* battery/temperature
deye_inverter_2388186981:deye/ac/total_power:.* ac/total_power
deye_inverter_2388186981:deye/ac/daily_energy_bought:.* ac/daily_energy_bought
deye_inverter_2388186981:deye/ac/total_energy_bought:.* ac/total_energy_bought
deye_inverter_2388186981:deye/ac/daily_energy_sold:.* ac/daily_energy_sold
deye_inverter_2388186981:deye/ac/total_energy_sold:.* ac/total_energy_sold`

kbialek, 2023-03-20:
@hackepeterOli 
These values look suspicious to me.

> ac/l1/power | 58.0 | 2023-03-19 20:36:12
ac/l2/current | 654.4 | 2023-03-19 20:36:12
ac/l2/power | 65349.0 | 2023-03-19 20:36:12
ac/l3/power | 46.0 | 2023-03-19 20:36:12

Could you run the service again with DEBUG logs enabled? (Set "LOG_LEVEL=DEBUG" in config.env) This will show us the values read by the service.

> No, "Produktion" should be 3.29, this is "Produktion-Heute":
> day_energy | 8.3 | 2023-03-19 20:07:27

There are two possibilities:
1. Solarman app calculates "Produktion" basing on other metrics
2. "Produktion" is not yet mapped in the code. Unfortunately I do not have a protocol spec for this type of inverter, so I cannot check this.

> It is too late for me, and i have corona :(

Get well soon!

hackepeterOli, 2023-03-21:
@kbialek I checked the values in the Home Assistant. They look exactly the same there. However, here are the logs

`2023-03-21 21:57:17,152 - DeyeDaemon - INFO - Reading start
2023-03-21 21:57:17,152 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_battery, range: 0202-022e]
2023-03-21 21:57:17,155 - DeyeConnector - DEBUG - Request frame: a517001045000065d7588e02000000000000000000000000000001030202002d25af9915
2023-03-21 21:57:21,100 - DeyeConnector - DEBUG - Response frame: a56d001015002465d7588e0201075132000301000035d7e76301035a00430042010c0000016700000045000201850000000b0000008102ac000000610000000000000000016d00000061016d0000000004e20564000000000000000000000000000000000a000001000d0000000000000000000000006b7e0c15
2023-03-21 21:57:21,100 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_battery, range: 024a-024f]
2023-03-21 21:57:21,104 - DeyeConnector - DEBUG - Request frame: a517001045000065d7588e0200000000000000000000000000000103024a0006e5a67115
2023-03-21 21:57:27,249 - DeyeConnector - DEBUG - Response frame: a51f001015002565d7588e02010e5132000901000035d7e76301030c048014040015000002d105816f8d9515
2023-03-21 21:57:27,250 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3, range: 0256-027c]
2023-03-21 21:57:27,253 - DeyeConnector - DEBUG - Request frame: a517001045000065d7588e020000000000000000000000000000010302560027e4786f15
2023-03-21 21:57:32,502 - DeyeConnector - DEBUG - Response frame: a561001015002665d7588e0201135132000f01000035d7e76301034e0943094d0948000000000000ff49fff3ff70feac000013880061001f004400060006000600650000005f00c40000000000650000005f00c40000094a093f093c003c0000002800db001f009e01981e4cc515
2023-03-21 21:57:32,503 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3, range: 02a0-02a7]
2023-03-21 21:57:32,506 - DeyeConnector - DEBUG - Request frame: a517001045000065d7588e020000000000000000000000000000010302a0000845961915
2023-03-21 21:57:37,663 - DeyeConnector - DEBUG - Response frame: a523001015002765d7588e0201185132001401000035d7e7630103100000000000000000009200000049000086465515
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV1 Power: 0.0
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV2 Power: 0.0
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV1 Voltage: 14.6
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV2 Voltage: 7.3
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV1 Current: 0.0
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - PV2 Current: 0.0
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - Daily Production: 9.7
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - Total Production: 36.5
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - Daily Battery Charge: 6.7
2023-03-21 21:57:37,664 - DeyeDaemon - DEBUG - Daily Battery Discharge: 6.6
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Total Battery Charge: 26.8
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Total Battery Discharge: 35.9
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Battery Power: 721.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Battery Voltage: 51.2
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Battery SOC: 21.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Battery Current: 14.1
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Battery Temperature: 15.2
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Total Grid Power: 196.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Grid Voltage L1: 237.1
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Grid Voltage L2: 238.1
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Grid Voltage L3: 237.6
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Daily Energy Bought: 6.9
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Total Energy Bought: 38.9
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Daily Energy Sold: 0.2
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Total Energy Sold: 1.1
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Current L1: 0.6
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Current L2: 0.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Current L3: 0.4
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Inverter L1 Power: 219.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Inverter L2 Power: 31.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - Inverter L3 Power: 158.0
2023-03-21 21:57:37,665 - DeyeDaemon - DEBUG - DC Temperature: 25.0
2023-03-21 21:57:37,666 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'online'
2023-03-21 21:57:37,666 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m35), 'b'deye/logger_status'', ... (6 bytes)
2023-03-21 21:57:37,670 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 35)
2023-03-21 21:57:37,670 - DeyeMqttClient - INFO - Logger is online
2023-03-21 21:57:37,670 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/power', value: '0.0'
2023-03-21 21:57:37,670 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m36), 'b'deye/dc/pv1/power'', ... (3 bytes)
2023-03-21 21:57:37,674 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 36)
2023-03-21 21:57:37,674 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv2/power', value: '0.0'
2023-03-21 21:57:37,675 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m37), 'b'deye/dc/pv2/power'', ... (3 bytes)
2023-03-21 21:57:37,678 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 37)
2023-03-21 21:57:37,679 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/dc/pv1/voltage', value: '14.6'
[... cut ...]
2023-03-21 21:57:37,793 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m66), 'b'deye/ac/l3/power'', ... (5 bytes)
2023-03-21 21:57:37,796 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 66)
2023-03-21 21:57:37,796 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/radiator_temp', value: '25.0'
2023-03-21 21:57:37,797 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m67), 'b'deye/radiator_temp'', ... (4 bytes)
2023-03-21 21:57:37,800 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 67)
2023-03-21 21:57:37,800 - DeyeDaemon - INFO - Reading completed
`

hackepeterOli, 2023-03-22:
Yes, we can close it, many thx!

closed 2023-03-22 by kbialek

toleabivol, 2023-09-13:
Hello @kbialek ,
I also tried the image above `ghcr.io/kbialek/deye-inverter-mqtt:2023.03.3-exp-sg04lp3` and it seems to work very well but the pnly issue I see now is that when the battery is charging it is publishing always a high number like 64kW.

[image]

When it is discharging then the value is correct as you see in my screenshot ~1kW .

toleabivol, 2023-09-14:
Update: Works fine if I use the latest docker image and not the one specified above.