kbialek/deye-inverter-mqtt #35: active_power_regulation

my-engel, 2023-03-05:
Hello,
if you change both files, the power control values are also output on the micro...
self.modbus.read_registers(0x20, 0x2f) and active_power_regulation
[active_power_regulation.zip](https://github.com/kbialek/deye-inverter-mqtt/files/10892431/active_power_regulation.zip)

greetings Uwe

kbialek, 2023-03-05:
Yes, but why? This is a configuration parameter, so why to monitor it?

my-engel, 2023-03-05:
Hello,
there are users who switch to battery overnight at the entrance.
In order to then only cover the base load, throttle the power control to the desired value. The changeover then takes place in the morning and evening.

VG Uwe

kbialek, 2023-03-05:
And that switch changes active power regulation factor, right? I'm not familiar with this process. Explain it to me please.

my-engel, 2023-03-05:
you do for example:
docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt w 40 50
and sets the 600s to 50%. That means you set it from 2x 300W to 2x 150W, i.e. per input. If a battery is now connected to an input overnight, you can generate 150W base load and not feed anything into the public grid. Let's say you need 150W

kbialek, 2023-03-05:
Thanks. I get it, so it's a zero-export setup. I can add this sensor for sure. I can probably also add a support for a command topic to allow setting this parameter over mqtt. Would it make sense in your use case?

kbialek, 2023-03-11:
hi @my-engel 

This beta image `ghcr.io/kbialek/deye-inverter-mqtt:2023.03.2-beta-1` has an option to set the logger/inverter time once it becomes available online. This feature must be activated in the configuration by adding `DEYE_FEATURE_SET_TIME=true`
Could you test it?

@Antagor That may be interesting for you too.

my-engel, 2023-03-11:
Thanks,

I will test ...
FYI because you are calculating the date and time and writing it to register 22. Register 22 contains only month and year.
'w 22 5891' for year and month
'w 23 518' for day and hour
"w 24 5166" for minute and second
greetings Uwe

my-engel, 2023-03-11:
It's enough month and year...
Time and date are beautiful.
But month and year are enough....

kbialek, 2023-03-12:
I guess you are referring to this code snippet
```python
        self.__modbus.write_registers(22, [
            # year and month
            256 * (now.year % 100) + now.month,
            # day and hour
            256 * now.day + now.hour,
            # minute and seconds
            256 * now.minute + now.second
        ])
```

It writes three registers in a row, starting from register 22.
This approach requires only one modbus request instead of three.

my-engel, 2023-03-12:
Yes, I meant this code.
and yes it works.
It writes start/reset UTC time and date to the registers.
The daily yield data is reset.
The time and date are retained for the entire time until the DC is switched off and do not continue to count. You could use these for other things like evaluating the runtime of the day, but that's another topic.

Now only the power control registers self.modbus.read_registers(0x20, 0x2f) have to be read out.

Thank you, regards Uwe

kbialek, 2023-03-12:
> Yes, I meant this code.
> and yes it works.
> It writes start/reset UTC time and date to the registers.
> The daily yield data is reset.
> The time and date are retained for the entire time until the DC is [...]

Perfect :+1:

Antagor, 2023-03-12:
You are right. Very interesting.
I'm knoked out since a few days.
I will test it as soon as i can and give you a reply.
Marc

Antagor, 2023-03-22:
Wow. you changed a lot in your program!
integrated hybrid with and without battery!

i run the daily reset since a few days without errors! Great! Thank you
very much. I did it manualy in IOBroker....

Question: for me it is not relevant - because mine switches of by night.
When do you reset the time if the inverter run 24/7 with battery? I
understand your code that you check if the actual date and the date in
reg 22-24 is equal you don`t set the time. i have not tested jet - what
happens when docker will be restarted?

Thanks for your work

Antagor

kbialek, 2023-03-23:
> Question: for me it is not relevant - because mine switches of by night. When do you reset the time if the inverter run 24/7 with battery? 

The time is set only when the inverter is becoming available online. So, if the inverter is on battery, and thus always online, the time will not be set.

> I understand your code that you check if the actual date and the date in reg 22-24 is equal you don`t set the time.

No the code does not check time equality. It sets the time unconditionally.

> i have not tested jet - what happens when docker will be restarted?

The time will be set after the first successful communication with the inverter

kbialek, 2023-05-31:
> Having a command topic to set values (for active power regulation) would be great.
> Is it still possible to implement this feature?

It is. Once I find time for that.

closed 2023-06-09 by kbialek