kbialek/deye-inverter-mqtt #38: error messages during image start

parkpirat, 2023-03-11:
Hallo,
I'm really like this project and I'm interesting in to use it. 
During the first tries I'm getting the following errors:

"......Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 19, in <module>
    import logging
  File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
    _startTime = time.time()......"

During that time, the Inverter was just online, not producing. Could it be related together? Or what else could be the problem?

Regards,
Parkpirat

kbialek, 2023-03-11:
So the logging library fails to start? Please provide full startup log.

parkpirat, 2023-03-11:
How to get a full startup log? There is no change in the messages after modifying the config.enf  from INFO to DEBUG/ERROR.

kbialek, 2023-03-11:
It's everything that gets printed by the docker container until the errors appears.

parkpirat, 2023-03-11:
The errors occurs more or less imediately after entering this:
pi@raspberrypi:~ $ docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt

Unable to find image 'ghcr.io/kbialek/deye-inverter-mqtt:latest' locally
latest: Pulling from kbialek/deye-inverter-mqtt
c6556b3b6858: Pull complete
c3bc830afdce: Pull complete
350038f6bba7: Pull complete
701ac2a102f5: Pull complete
2ea74c63d047: Pull complete
afd8f4185ebb: Pull complete
12e7d5e8ac50: Pull complete
c4378880d2b8: Pull complete
Digest: sha256:bddf7ad32462452156d9f36e1e7e9dfb2f9a190c1191d6916c5c9b55e32ae7c7
Status: Downloaded newer image for ghcr.io/kbialek/deye-inverter-mqtt:latest
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 19, in <module>
    import logging
  File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>
    _startTime = time.time()
PermissionError: [Errno 1] Operation not permitted

kbialek, 2023-03-11:
Here they have a similar problem https://github.com/certbot/certbot/issues/9578
You can try to follow their workaround and run the image as privileged
What hardware and operating system are you using?

parkpirat, 2023-03-11:
I'm using a Raspberry Pi 4 Model B Rev 1.4 with Raspbian GNU/Linux 10 (buster).

kbialek, 2023-03-11:
That's interesting. It seems like I'm using this same setup. Can you confirm?
`r=$(awk '/^Revision/ { print $3 }' /proc/cpuinfo) && curl -L https://perturb.org/rpi?rev=$r`
```
Revision: c03114
Model   : 4 Model B
Memory  : 4 GB
Overvolt: No
Released: Q2 2020
Notes   : (Mfg by Sony)
```

and 

`uname -a`
```
Linux quark 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

parkpirat, 2023-03-11:
Revision: d03114
Model   : 4 Model B
Memory  : 8 GB
Overvolt: No
Released: Q2 2020
Notes   : (Mfg by Sony)

and

Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

kbialek, 2023-03-11:
Okay I found it in my notes :slightly_smiling_face: 

You have to install `libseccomp2`

See this thread
https://community.home-assistant.io/t/migration-to-2021-7-fails-fatal-python-error-init-interp-main-cant-initialize-time/320648

here are the commands
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 04EE7237B7D453EC 648ACFD622F3D138
echo "deb http://deb.debian.org/debian buster-backports main" | sudo tee -a /etc/apt/sources.list.d/buster-backports.list
sudo apt update
sudo apt install -t buster-backports libseccomp2
```

parkpirat, 2023-03-11:
Thank you very much for your fast and strong support.
The commands were executed and no errors occurred during this. :-)

When I start the image now, the cursor "comes" not back to pi@raspberrypi:
So it looks like it's not finished somehow.
How can I check starting was finished and successfully?

pi@raspberrypi:~ $ docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt
[image]

kbialek, 2023-03-11:
Try restarting docker service after installing libseccom2

parkpirat, 2023-03-12:
Hello again :-),
after reboot (after sudo apt-get update && sudo apt-get upgrade) it's still the same. After starting the docker image, the command prompt is not coming back.

One additional question:
In the documentation there are a lot of parameters inside Configuration; 13 different points are there. In my config.env (downloaded from here) I've only the following:

kbialek, 2023-03-12:
> Hello again :-), after reboot (after sudo apt-get update && sudo apt-get upgrade) it's still the same. After starting the docker image, the command prompt is not coming back.
> 
> a) Is there a "easy way" to proof that the image is doing something?

Let's check if your docker service runs images correctly. This command downloads and runs very slim linux image and drops you into the container shell
`docker run --rm -ti alpine:3.16 sh`

If everything is okay, you should see something like this
```
Unable to find image 'alpine:3.16' locally
3.16: Pulling from library/alpine
beefe5ad637c: Pull complete 
Digest: sha256:1bd67c81e4ad4b8f4a5c1c914d7985336f130e5cefb3e323654fd09d6bcdbbe2
Status: Downloaded newer image for alpine:3.16
/ #
```

> 
> b) One additional question: In the documentation there are a lot of parameters inside Configuration; 13 different points are there. In my config.env (downloaded from here) I've only the following 11 points:
> 
> ```
>  DEYE_LOGGER_IP_ADDRESS=xxx
>  DEYE_LOGGER_PORT=8899
>  DEYE_LOGGER_SERIAL_NUMBER=xxx
> 
>  MQTT_HOST=xxx
>  MQTT_PORT=1883
>  MQTT_USERNAME=test
>  MQTT_PASSWORD=test
>  MQTT_TOPIC_PREFIX=deye
> 
>  LOG_LEVEL=ERROR
>  DEYE_DATA_READ_INTERVAL=60
>  DEYE_METRIC_GROUPS=micro
> ```
> 
> Can it be a problem that MQTT_AVAILIBILITY_TOPIC and MQTT_LOGGER_STATUS_TOPIC are not there?

Your config is valid. These additional parameters are optional and have default values

> 
> c) I placed the config.env directly in home/pi/... Could this create the problem?

No, it is not a problem as long as you specify a correct path to it in the docker run command. Normally docker will report an error when the given file does not exist

parkpirat, 2023-03-12:
So, I'm back and much happier. :-)
It look likes the sending started.
I wasn't aware, that the prefix is so important for the image. So I changed the deye prefix to solaranzeige (this is what the mqtt broker is expecting) and I can see it's running.

Thank you very much for your strong help again.
[image]

kbialek, 2023-03-12:
Awesome! So what prefix have you used before? Perhaps I can add some validation to avoid similar problems in the future.

parkpirat, 2023-03-12:
Like described above: First I used the original untouched prefix (deye) . Later I changed to solaranzeige (this is what the other broker needs).
The history behind: I have a Deye SUN1600G3 inverter and a shelly3EM. I would also like to use the "Solaranzeige" running on Raspi. But Deye isn't supported there. I startet to use "Solaranzeige" with Shelly. Later I found your solution to get the data of the Deye via MQTT. MQTT is also supported in "Solaranzeige". But there the data has to come with "Solaranzeige/Anzeige/1" if I understood right.

kbialek, 2023-03-12:
Do you mean this https://solaranzeige.de platform?
I understand that you had to configure mqtt topic prefix required by the platform.
What I do not understand is, why the docker container was not starting up properly before you did this change (as you described it here https://github.com/kbialek/deye-inverter-mqtt/issues/38#issuecomment-1464924587) Can you explain this to me? Thanks!

parkpirat, 2023-03-13:
Yes, I was talking about [https://solaranzeige.de](url)/.

Unfortunately I can't get the bad result again. I changed back "MQTT_TOPIC_PREFIX=deye" without problem. Now the image is starting well every time. So I'm sorry, it looks I can't force the problem again and the image is starting every time well.

If you need further information, I can prepare it on the weekend for you. During the week I don't find time for playing with the raspberry.

Now I've to find why the data won't save in influxDB. When I check there with "mosquitto_sub -h localhost -v -t +/#", I can see the data from the Deye.

By the way: Maybe you should think about to support [https://solaranzeige.de](url) directly and implement the different Deyes there?

kbialek, 2023-03-13:
> If you need further information, I can prepare it on the weekend for you. During the week I don't find time for playing with the raspberry.

No need to do that. Thanks

> By the way: Maybe you should think about to support [https://solaranzeige.de](url) directly and implement the different Deyes there?

I belive that MQTT is the most appropriate and versatile integration protocol. It is supported by plenty of smart home and IoT platforms. This way everyone can choose the solution that suits them best. That is also the reason why I will not implement a direct integration with any specific platform or database.

parkpirat, 2023-03-22:
Thank you for your support. I'll close the issue now.

closed 2023-03-22 by parkpirat