kbialek/deye-inverter-mqtt #41: Compatible inverters

kbialek, 2023-03-19:
Please report here inverter models that work with this tool. This will let me build a database of compatible inverters. Thanks!

aresius, 2023-03-20:
It seems to work with
Model: Deye SUN300G3-EU-230
Firmware: MW3_16U_5406_1.53

parkpirat, 2023-03-22:
1. Deye SUN1600G3-EU-230
2. micro

-> in general working, not finally tested all collected data
Did you use the file below as base for the different parameters? This file I've received from Deye service last time.

[DEYE_Modbus储能-组串-微逆宁波德业V118.pdf](https://github.com/kbialek/deye-inverter-mqtt/files/11036325/DEYE_Modbus.-.-.V118.pdf)

kbialek, 2023-03-22:
> Did you use the file below as base for the different parameters? This file I've received from Deye service last time.

Yes, exactly

K3vb3rt, 2023-04-16:
Model: Deye SUN800G3-EU-230
Firmware: MW3_16U_5406_1.57

Thanks!

ivanK0dy, 2023-04-16:
Hi @kbialek,

As inverters are just rebranded but same HW, I tested on FU-SUN-5K-G05 (Fuji Solar) and get most of information but I miss data from Mobus connected Meter which data are on registers 150-218: 
PDF Document for Deye at: https://diysolarforum.com/attachments/modbus%E5%82%A8%E8%83%BD-%E7%BB%84%E4%B8%B2-%E5%BE%AE%E9%80%86%E5%AE%81%E6%B3%A2%E5%BE%B7%E4%B8%9Av118-pdf.110709/

Registers dec 150 - 218 ? As I can see your script only pull range 003c-0075.
If i try to read it with your toll I get:
Day Load Power:
docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 200
int: 1194, l: 170, h: 4

Meter active power: registers 203 (low) and 204(high) I get:
int: 65027, l: 3, h: 254
int: 65535, l: 255, h: 255

How to pull this second range? Or I should create new instance with new range and only read new registers from that range.
Thank you

edvandreas, 2023-04-16:
Logger Model: Stick Logger LSW-3 (Wifi)

Inverter Model: Deye Sun 12K SG04LP3 EU 
Firmware: 2005-1128-1807

Works great, reading the data every 10 seconds without problems. Thank you!

devil33, 2023-04-18:
Works here with 

Model: Deye SUN600G3-EU-230
Firmware: MW3_16U_5406_1.57

Thank you so much!

RafaelDuncan, 2023-05-09:
> Logger Model: Stick Logger LSW-3 (Wifi)
> 
> Inverter Model: Deye Sun 12K SG04LP3 EU Firmware: [2005-1128-1807](tel:2005-1128-1807)
> 
> Works great, reading the data every 10 seconds without problems. Thank you!

Hello, is it also possible to write data to the Deye Sun 12K SG04LP3 EU?  I want to be able to control the adjustments from Home Assistant

marcniedersachsen, 2023-06-20:
Thank you very much for this great implementation. I have successfully tested these two ("brand new ?") Micro Inverters:

Deye SUN-M60g3-EU-230
and
Deye SUN-M80g3-EU-230

Metric group = Micro,  (I haven't tested ",settings". Guess that will work too :-)

lpunkt66, 2023-06-23:
**Works for me with version 2023.06.2**  👍 

Model: Deye SUN800G3-EU-230
Firmware: MW3_16U_5406_1.57
Metric group: micro

Device serial number 3912xxxxxx
Inverter serial number 2303xxxxxx

**Did't work with 2023.05.4**

Thanks very much!!!

fakehec, 2023-07-15:
Model:Deye 30kW SUN-30K-SG01HP3-EU-BM3 hybrid with battery
Metric: deye_sg04lp3, deye_sg04lp3_battery,deye_sg04lp3_ups

Logger serial number: 2770xxxxxx
Inverter serial number: 2304xxxxxx
Works ok everything except data from PV3

Please implement PV3 line data in deye_sg04lp3

Verkalets, 2023-07-17:
@kbialek thanks!

toleabivol, 2023-09-14:
DEYE SUN-8K-SG04LP3-EU
Metrics : deye_sg04lp3,deye_sg04lp3_battery

Harry4711, 2023-09-20:
Hi congratulation to the project.
Model: Deye SUN800G3-EU-230
Firmware: MW3_16U_5406_2.32-D1
I did create a "deye_sensors_deye_2mppt.py" with the appropriate Value for the above model and intergrated it in your "deye_sensors.py" file as you did with other "deye_sensors_deye_*.py" files.
It worked well in an Home Automation environment developed with FHEM
If you are intereset, I could upload the file and send further information.
I did create own groups to not have to use the "micro" one to be more specific for this model.
Best regard

fschwanzara, 2023-10-12:
Model: Deye SUN600G3-EU-230
Firmware: MW3_16U_5406_2.32-D1
Metric Group: Micro

**Error:**
```
DeyeModbusTcp - ERROR - Response frame has invalid starting byte
DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 41542b595a434d505645523d4d57335f3136555f353430365f322e33322d44310d0a0d0a
DeyeTcpConnector - DEBUG - Request frame: a51700104500008fde70e90200000000000000000000000000000103003c003945d4c615
DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
DeyeDaemon - INFO - Reading start
```

I just wanted to get the MQTT messages to send it over to influxDB/Grafana

Harry4711, 2023-10-12:
TCP ist Not supported any more with this Software version, pleased use AT !!
See documentation

voipmonitor, 2024-01-20:
Hello, 

I have SUN-50K-SG01HP3-EU-BM4   which has 4 strings - I have two battery towers with two BMS - I have modified deye_sensors_deye_sg01hp3.py and added BMS1, BMS2, bettery1 and battery2 registry and fixed one signed/unsigned parameter. 
[deye_sensors_deye_sg01hp3.zip](https://github.com/kbialek/deye-inverter-mqtt/files/13998908/deye_sensors_deye_sg01hp3.zip)

mrlordziemniak, 2024-02-17:
Hi i have DTSD422-D3 i can read data, but for example current CT1 when energy is with - value there is still + how to mengae this?
and on all Total Positive Energy CT1 CT2 etc is not correct value, any ideas why?

xGDNAx, 2024-03-15:
WR BW-MI600 Bosswerk 
MW3_16U_5406_2.32-D1

Micro, settings

omgfreezeur, 2024-03-28:
Deye SUN-M160**G4**-EU-Q0
SW: MW3_16U_5406_1.59

Micro, settings

I receive all data via MQTT Broken in HA without any errors.
Unfortunately, I have not yet found a way to send a command to set the active_power_regulation via MQTT.
Can anyone give me a hint?

xGDNAx, 2024-03-28:
https://github.com/kbialek/deye-inverter-mqtt/issues/154

SandimL, 2024-04-13:
Its working with:

Model: Deye [SUN-4K-G](https://pt.deyeinverter.com/product/single-phase-string-inverter/sun3-6-5-6kg.html)
Firmware: LSW3_15_FFFF_1.0.9E
Metric groups: string

Thank you!

jpgnz, 2024-05-19:
Model: Deye [SUN-16K-SG01LP1-EU](https://www.deyeinverter.com/product/hybrid-inverter-1/sun12-14-16ksg01lp1-1216kw-single-phase-3-mppt-hybrid-inverter.html)
DEYE_METRIC_GROUPS=deye_hybrid,deye_hybrid_battery,deye_hybrid_timeofuse,settings

Looks to be complete, have left all features at defaults, eg:

```
# DEYE_FEATURE_MQTT_PUBLISHER=true
# DEYE_FEATURE_TIME_OF_USE=false
# DEYE_FEATURE_ACTIVE_POWER_REGULATION=false
# DEYE_FEATURE_SET_TIME=false
```

matt-kubica, 2024-05-23:
Deye `SUN-20K-SG01HP3-EU-AM2`  (version of [SUN-6/8/10/12/15/20K-SG01HP3-EU-AM2](https://deye.com/product/sun-6-8-10-12-15-20k-sg01hp3-eu-am2-6-20kw-three-phase-2-mppt-hybrid-inverter-low-voltage-battery/)) seems to be working fine with `deye_sg01hp3, deye_sg01hp3_battery, deye_sg01hp3_bms, deye_sg01hp3_ups, settings` metric groups.

Haven't tried writes and features besides mqtt reporting. Configuration was mostly default (apart from the envs I had to change).

TRIROG, 2024-07-14:
Confirmed  working DEYE SUN-12K-SG04LP3-EU
With TCP settings.
DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery,deye_sg04lp3_ups,deye_sg04lp3_timeofuse,deye_sg04lp3_generator

Thank you! Awesome work!

CarstenGrohmann, 2024-08-05:
## Deye SUN300G3-EU-230

all firmware releases:
```
DEYE_METRIC_GROUPS=micro
DEYE_FEATURE_ACTIVE_POWER_REGULATION=false
DEYE_FEATURE_SET_TIME=true
```

### Firmware: MW3_16U_5406_1.53 till MW3_16U_5406_1.57
Metric groups: micro
Modbus/TCP (default)

### Firmware: MW3_16U_5406_2.32-D1
Metric groups: micro
Modbus/AT

BjoernSch, 2024-08-18:
SUN-M80G4-EU-Q0
MW3_16U_5406_1.62

Metric groups: micro, settings_micro
Modbus/AT

Metric groups: micro, settings_micro
Modbus/TCP (default)

Settings was not really tested.

Bhoft, 2024-08-25:
SUN-M160G4-EU-Q0
MW3_16U_5406_1.59

Metric groups: micro,settings_micro
DEYE_LOGGER_PROTOCOL: at

Everythings works perfectly

closed 2024-10-10 by github-actions[bot]

kingp0dd, 2025-01-29:
How can we add new sensors to the metric groups? There are additional (grid and load consumption) values that worked on Deye string inverter

Deye SUN-6K-SG05P1-EU-AM2

https://github.com/davidrapan/ha-solarman/commit/12c5d45704764a802a9d6c8a012e6ad28bc760dd

https://github.com/davidrapan/ha-solarman/commit/20e04f14a7fc366831e65e61ed86812e185a0059

Pims8383, 2025-12-30:
Model : SUN-M100G4-EU-EQ
Firm : MW3_16U_5406_1.62
Metric : micro, settings, deye_micro_systemtime

Working perfectly

completelyClueless, 2026-05-10:
Model: 20K-SG05LP3-EU-SM2

Metrics: *SG04LP3*

joshuwapoot, 2026-08-20:
SUN-5K-SG04LP1
LSW3_32_5406_SS_04_00.00.00.0B

yaqjav, 2026-09-18:
Model: Deye SUN-15K-SG05LP3-EU-SM2
Firmware: LSW3_32U_5406_1.0B
Metric groups: deye_hybrid, deye_hybrid_three_phase, deye_battery_multi