kbialek/deye-inverter-mqtt #45: time lag between time in front of terminal window and real time

parkpirat, 2023-03-22:
Hi,
I observed a time lag between time in terminal window with running docker image and the real time.
This is shown in the terminal:

     2023-03-22 05:12:36,479 - DeyeMqttClient - INFO - Logger is online
     2023-03-22 05:12:36,534 - DeyeDaemon - INFO - Reading completed
     2023-03-22 05:13:36,543 - DeyeDaemon - INFO - Reading start
     2023-03-22 05:13:36,543 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]

The time in reality was 06:12 or 06:13 at this time.
The raspberry shows the correct time with  "date".

Where is the 1hr. difference coming from?

kbialek, 2023-03-22:
@parkpirat Shortly speaking, the difference comes from the daylight saving.
This docker container runs with UTC timezone by default. It ignores the timezone you have set on your raspi.
If you wish to change it, start the container with additional environment variable `-e TZ=Europe/Berlin`

parkpirat, 2023-03-22:
Thank you. Problem solved. I start now with:
docker run -e TZ=Europe/Berlin --rm --env-file  config.env ghcr.io/kbialek/deye-inverter-mqtt

closed 2023-03-22 by parkpirat