kbialek/deye-inverter-mqtt #46: howto use this docker image with 2 deye 10k inverters (1 x Master 1 x Slave)

Gregor4711, 2023-03-22:
first exelent work, many thanks for your effort and make it public.

 how to config env to use this with 2 logger (=> 2 Inverter?)

mastertape, 2023-03-22:
i ´ve the same question - with 2 x deye 600 eu

kbialek, 2023-03-22:
You have to create two config files, one for each inverter. Then you can run two instances of the service. For example

```
docker run --rm --name mqtt-inv1 --env-file config-inv1.env ghcr.io/kbialek/deye-inverter-mqtt
docker run --rm --name mqtt-inv2 --env-file config-inv2.env ghcr.io/kbialek/deye-inverter-mqtt
```

This command will create two docker containers, one for each of your containers. Additionally they will be named `mqtt-inv1` and `mqtt-inv2`, so you can figure out which one is which e.g. when listing them with `docker ps` command.

Of course be sure to stop the old container before you do that

closed 2023-05-07 by github-actions[bot]