kbialek/deye-inverter-mqtt #47: regrouping data

Gregor4711, 2023-03-23:
I'd like to groupe the data like logger status, status, total_energy etc. in an "info" topic.

which module is customizing the mqtt data structure?

kbialek, 2023-03-23:
What do you mean by grouping exactly? 
1. Build a data structure containing all those values (e.g. JSON) and send it over "info" topic?

or

2. Modify existing topic hierarchy, e.g. map logger_status to info/logger_status, total_energy to info/total_energy, and so on?

Gregor4711, 2023-03-25:
it is more second topic, but in the meanwile I was able to sort it out via Nodered. 
I wrote a script which provide the data structure as needed in mqtt => elastic => grafana.

A first check for pv data like PV1, PV2, total values, etc. looks fine.
Need to check battery values, current=653,6 and power= 65461.0 looks wired, but SOC= 99 is fine

kbialek, 2023-03-25:
@Gregor4711 It's great that you found a solution that suits you.

Yes, I can confirm that some values are looking strange for sg04lp3 inverter. I've seen it already before while adding the support for this inverter together with @hackepeterOli here https://github.com/kbialek/deye-inverter-mqtt/issues/31#issuecomment-1475372788

Since I haven't had modbus documentation for this inverter, I took the mapping that's used by the corresponding HomeAssistant integration. Perhaps the mapping is not entirely correct.

If you would have time, you can try to read the registers manually. I suspect addresses between 560 and 640 are worth trying. Maybe you will find appropriate addresses this way. Then we could fix the mapping.

A read command looks like this
```
docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r <reg_address>
```

Gregor4711, 2023-03-25:
sure I can do. Can you confirm checking of registers on this way will not harm the inverter nor the firmware of inverter?

Since I'm not an modbus expert, what is happen / shown when I call the docker with register address?
Or in other words for what we looking for? When I meet an proper register?

kbialek, 2023-03-25:
> sure I can do. Can you confirm checking of registers on this way will not harm the inverter nor the firmware of inverter?

Reads are safe, as they do not affect the operation of the inverter

> Since I'm not an modbus expert, what is happen / shown when I call the docker with register address?
> Or in other words for what we looking for? When I meet an proper register?

Modbus is a serial communication protocol. It offers a handful of features. One of them is reading 16-bit registers from 16-bit address space. That is the feature used by the integration. Each value that is reported over mqtt is basically first read from a certain register (or a pair of registers, in case of totals). The current and the voltage values are typically returned as multiplied by 10 - for fractional precision. The power is usually not multiplied.

As you've noticed:
> current=653,6 and power= 65461.0 looks wired

So, what we are looking for, are meaningful values that we can correlate with corresponding values, that you can see in the Solarman app.

The read command will print 16-bit value followed by and H and L byte values. But we are interested in the first 16-bit value.

Gregor4711, 2023-03-25:
first patch, may also other can have a look. 

only register  588 is recognizable for me, can be=> SOC 99% battery, wich is also close to your "wired" value 65466 :)

pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 560
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 561
int: 9, l: 9, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 562
int: 1, l: 1, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 563
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 564
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 565
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 566
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 567
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 568
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 569
int: 2, l: 2, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 570
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 571
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 572
int: 13816, l: 248, h: 53
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 573
int: 44, l: 44, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 574
int: 2048, l: 0, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 575
int: 5187, l: 67, h: 20
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 576
int: 6000, l: 112, h: 23
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 577
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 578
int: 6000, l: 112, h: 23
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 579
int: 4986, l: 122, h: 19
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 580
int: 722, l: 210, h: 2
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 581
int: 361, l: 105, h: 1
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 582
int: 631, l: 119, h: 2
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 583
int: 2, l: 2, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 584
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 585
int: 250, l: 250, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 586
int: 1245, l: 221, h: 4
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 587
int: 5186, l: 66, h: 20
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 588
int: 99, l: 99, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 589
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 590
int: 65466, l: 186, h: 255

Gregor4711, 2023-03-25:
second patch, nothing special found

Register 592 => 800 can't find any relation to value from Inverter or Solarman

pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 591
int: 65375, l: 95, h: 255
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 592
int: 800, l: 32, h: 3
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 593
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 594
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 595
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 596
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 597
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 598
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 599
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 600
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 601
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 602
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 603
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 604
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 605
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 606
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 607
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 608
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 609
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 610
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 611
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 612
int: 2, l: 2, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 613
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 614
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 615
int: 1, l: 1, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 616
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 617
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 618
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 619
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 620
int: 0, l: 0, h: 0

Gregor4711, 2023-03-25:
last patch

Register 633->635 (7,7,6) are same like infoscreen in deye Inverter listet as "INV_P:" 7W, 7W, 6W (no glue what it should be. There is also an "AC_T:" at the screen with 43.0C in my case but can't see in values here.

hope that helps. In case of ony further register scanning count on me :)

pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 621
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 622
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 623
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 624
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 625
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 626
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 627
int: 2300, l: 252, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 628
int: 2299, l: 251, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 629
int: 2300, l: 252, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 630
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 631
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 632
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 633
int: 7, l: 7, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 634
int: 7, l: 7, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 635
int: 6, l: 6, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 636
int: 20, l: 20, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 637
int: 20, l: 20, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 638
int: 5000, l: 136, h: 19
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 639
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 640
int: 0, l: 0, h: 0

kbialek, 2023-03-25:
You correctly identified some of the registers. They are already known and mapped. You can see the list here:
* https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg04lp3.md
* https://github.com/kbialek/deye-inverter-mqtt/blob/main/docs/metric_group_deye_sg04lp3_battery.md

What is new to me:
1. These values look like Grid AC voltages multiplied by 10:
```
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 627
int: 2300, l: 252, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 628
int: 2299, l: 251, h: 8
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 629
int: 2300, l: 252, h: 8
```

Currently AC voltages are mapped to these registers
```
Grid Voltage L1 | 598 | ac/l1/voltage | V
Grid Voltage L2 | 599 | ac/l2/voltage | V
Grid Voltage L3 | 600 | ac/l3/voltage | V
```

However in your test they all reported zeros.
```
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 598
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 599
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 600
int: 0, l: 0, h: 0
```

2. Registers 630, 631, 632 are supposed to return AC currents, but they are all zero.
```
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 630
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 631
int: 0, l: 0, h: 0
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 632
int: 0, l: 0, h: 0
```

3. This is battery current, the value 65375 may look strange, but maybe it's a signed integer, then it would represent -16A. Does it make sense?
```
pc1$ sudo docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 591
int: 65375, l: 95, h: 255
```

Here I have a couple of questions:
1. What is your inverter model exactly?
2. Do you have a zero-export setup?
3. In what mode was your inverter when you did the test? (charging the battery, discharging the battery, exporting to the grid, importing from the grid)?
5. Does the Solarman app sometimes show negative current values? (I wonder if current values close to 65535 are maybe representing signed integers)

closed 2023-05-09 by github-actions[bot]