kbialek/deye-inverter-mqtt #49: wrong values or empty registers (SUN1600G3, metric_group_micro)

parkpirat, 2023-03-25:
Hi,
I've had a deeper look in the values from my inverter. Below you can find the registers with problems
(based by metric_group_micro.md, Metric | Modbus address | MQTT topic suffix | Unit).

Inverter type: SUN1600G3 -> metric-group_micro

a) Uptime | 62 | uptime | minutes. ----> shows all time: 0, 
---> register 62 means in Deye document: Today_Gen_PowerWh

b) DC Total Power | computed | dc/total_power | W     -----> shows: no data

c) Operating Power | 80 | operating_power | W     -----> shows: no data
---> register 80 in Deye document: looks like somehow related to register 81 (low and high word)

d) AC Active Power | 86,87 | ac/active_power | W     -----> shows: no data

e) Radiator temperature | 90 | radiator_temp | °C.       -----> shows: no data

Could someone verify this, please?

kbialek, 2023-03-25:
> a) Uptime | 62 | uptime | minutes. ----> shows all time: 0,
> ---> register 62 means in Deye document: Today_Gen_PowerWh

I've got a string inverter, and originally, I've recognized this register to report the uptime. And, yes, it does not match the documentation. Apparently, it does not make sense for micro, so my proposal is to remove it from the `micro` group

> b) DC Total Power | computed | dc/total_power | W -----> shows: no data

This is a sum of `dc/pv1/power`, `dc/pv2/power`, `dc/pv3/power`, `dc/pv4/power`, which are in turn computed as `dc/pv1/voltage` * `dc/pv1/current`, etc. What values do you get here?

> c) Operating Power | 80 | operating_power | W -----> shows: no data
> ---> register 80 in Deye document: looks like somehow related to register 81 (low and high word)

It's not marked with "MI" in the documentation, so I assume it's not supported for micro inverters. My suggestion is to remove it from the `micro` group

> d) AC Active Power | 86,87 | ac/active_power | W -----> shows: no data

In the documentation, they are marked with "MI", but the meaning is not explained.

> e) Radiator temperature | 90 | radiator_temp | °C. -----> shows: no data

In the documentation, they are marked with "MI". I'd expect it to work.

parkpirat, 2023-03-26:
I'm a little bit confused now and I've to say sorry.
After set the logging in your image to DEBUG, I could see that all the data are there. So my problem starts somewhere behind this image. So I'll focus now to the next step (saving the data in fluxdb; Mosquitto broker).

I'll close this here. No change needed.

Sorry and thank you again.

[image]

closed 2023-03-26 by parkpirat