kbialek/deye-inverter-mqtt #55: manifest for armv6

SeboCoder, 2023-04-05:
Hey, would it be possible to change the manifest to allow for armv6 processors so i can install it on my Raspberry pi zero w
thanks

kbialek, 2023-04-05:
The base image exists for this arch, so stay tuned

SeboCoder, 2023-04-05:
Thanks for your fast response!
How long would it take you to implement this? Just that i can plan a little bit.

kbialek, 2023-04-05:
It's ready. Test it and confirm if it works `ghcr.io/kbialek/deye-inverter-mqtt:2023.04.1-beta-1`

kbialek, 2023-04-05:
okay, the first beta is broken :smile: go ahead and test the second version `2023.04.1-beta-2`

SeboCoder, 2023-04-05:
tested beta2 got following error on docker run

Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 43, in <module>
    main()
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 39, in main
    daemon_main()
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 95, in main
    daemon = DeyeDaemon(config)
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 40, in __init__
    mqtt_client = DeyeMqttClient(config)
  File "/opt/deye_inverter_mqtt/deye_mqtt.py", line 37, in __init__
    self.__mqtt_client.connect(config.mqtt.host, config.mqtt.port, keepalive=60)
  File "/opt/deye_inverter_mqtt/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/opt/deye_inverter_mqtt/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/opt/deye_inverter_mqtt/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/usr/local/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: timed out

kbialek, 2023-04-05:
The good news is - it seems to work at arm/v6
The bad news is - it can't connect to your mqtt broker

SeboCoder, 2023-04-05:
Tested it multiple times with different configurations, still same error. I have a second docker container with this image running on the same network on a x64 system and its working just fine. 
What else could be the problem?

kbialek, 2023-04-05:
Hard to say, I'd start from testing the connectivity from within the container. For example, in this way:
1. SSH into your pi zero
2. Start the container in shell mode `docker run --rm -ti --entrypoint /bin/sh ghcr.io/kbialek/deye-inverter-mqtt:2023.04.1-beta-2`
3. Install `telnet` by running `apk update && apk add busybox-extras`
4. Check connectivity: `telnet <broker_ip> <broker_port>`

For me it works and shows the following result:
```
/opt/deye_inverter_mqtt # telnet 192.168.2.6 1883
Connected to 192.168.2.6
```

Otherwise you will get something like this:
```
telnet: can't connect to remote host (192.168.2.6): Connection refused
```

SeboCoder, 2023-04-05:
Network connectivity seems to work:

Config:
[image]

telnet:

`
/opt/deye_inverter_mqtt # telnet 192.168.0.89 1883
`
`
Connected to 192.168.0.89
`

SeboCoder, 2023-04-09:
Do you have an idea what else could be wrong?

kbialek, 2023-04-10:
Enable DEBUG logs. Maybe this will tell us more

SeboCoder, 2023-04-11:
I changed log level to Debug in config.env: `LOG_LEVEL=DEBUG`

then i recreated the container and then used `docker logs <container-id>` to get the logs 
but they are still the same:
[image]

Am i doing something wrong?

kbialek, 2023-04-11:
I don't think that you are doing anything wrong.

The log above shows that paho mqtt client can't create a connection to the broker. Paho is a 3rd party library that I use in my solution. Perhaps it has a bug or is misconfigured.

Please show more logs with the timestamps. That's why I asked for DEBUG logs. This will help to identify timing dependencies.

You can also test Paho client itself on your own.
1. Get into the container as you did for the test with telnet before
2. Execute `python` command to start python cli
3. Then `import paho.mqtt.client as paho`
4. Create mqtt client `c = paho.Client()`
5. Connect to your broker `c.connect('192.168.0.89', 1883)`

If it prints `0` then the connection worked.

SeboCoder, 2023-04-11:
Thank you very much for your response.

I found the problem.
After I tried your suggestions i wanted to check if the environment variables are set correct from inside the container and there i found that somehow it created the container always with an old config.env file. I always edited the file and then run the container with the same command. 
`sudo docker run -d --name deye-mqtt --env-file config.env --restart unless-stopped ghcr.io/kbialek/deye-inverter-mqtt:2023.04.1-beta-2`
but it seems that it always used old env variables with a wrong ip address

So i fixed it by copying my config.env file and renaming it config2.env and then use the command:
`sudo docker run -d --name deye-mqtt --env-file config2.env --restart unless-stopped ghcr.io/kbialek/deye-inverter-mqtt:2023.04.1-beta-2`

And now it seems to finally work. Container starts and stays up. Have to try it with the inverter in the next days.

I am so sorry if I wasted your time. But thank you very much for your patience. 
I`ll let you know when I tested it with my inverter so you can add it to the "tested inverters" (Deye SUN800G3-EU230)

SeboCoder, 2023-04-11:
well i used that command but without the `-v` flag. Do you think that was the problem?

kbialek, 2023-04-11:
No, I don't think so

SeboCoder, 2023-04-11:
hm interesting

SeboCoder, 2023-04-11:
well that looks how it should be. 🤔
maybe something else was wrong on my side

closed 2023-04-12 by kbialek