kbialek/deye-inverter-mqtt #57: Deye Connectivity- Serial Number does not match

manalishi70, 2023-04-06:
Hi Krysztof,

I try to connect to my DEYE SG04LP3-12K Inverter and get this message in the logfile:

Logger Serial Number does not match. Check your configuration

config.env:

DEYE_LOGGER_SERIAL_NUMBER=274409XXXXX

Wrong syntax or do I miss something here?

kbialek, 2023-04-06:
Please double check that you use **Logger SN** and not the Inverter SN

manalishi70, 2023-04-10:
Sure!

kbialek, 2023-04-15:
okay, enable and show the debug logs

AST70, 2023-05-02:
Hi folks,

I do have the same error message like manalishi70, but using a SUN600-G3-EU

`{"log":"2023-05-02 08:19:33,725 - paho.mqtt.client - DEBUG - Sending PINGREQ\n","stream":"stdout","time":"2023-05-02T08:19:33.728351455Z"}
{"log":"2023-05-02 08:19:33,735 - paho.mqtt.client - DEBUG - Received PINGRESP\n","stream":"stdout","time":"2023-05-02T08:19:33.738235156Z"}
{"log":"2023-05-02 08:19:40,692 - DeyeDaemon - INFO - Reading start\n","stream":"stdout","time":"2023-05-02T08:19:40.694669315Z"}
{"log":"2023-05-02 08:19:40,697 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]\n","stream":"stdout","time":"2023-05-02T08:19:40.700533138Z"}
{"log":"2023-05-02 08:19:40,724 - DeyeConnector - DEBUG - Request frame: a517001045000xxxxxxxxxxxxxxxxxxxxxxxxx\n","stream":"stdout","time":"2023-05-02T08:19:40.727931307Z"}
{"log":"2023-05-02 08:19:40,741 - DeyeConnector - DEBUG - Response frame: a510001015004xxxxxxxxxxxxxxxxxxxxxxxxxx\n","stream":"stdout","time":"2023-05-02T08:19:40.744767797Z"}
{"log":"2023-05-02 08:19:40,749 - DeyeModbus - ERROR - Logger Serial Number does not match. Check your configuration file.\n","stream":"stdout","time":"2023-05-02T08:19:40.755261479Z"}
{"log":"2023-05-02 08:19:40,758 - DeyeModbus - ERROR - Modbus frame is too short or empty\n","stream":"stdout","time":"2023-05-02T08:19:40.761906277Z"}
{"log":"2023-05-02 08:19:40,765 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'\n","stream":"stdout","time":"2023-05-02T08:19:40.769365051Z"}
{"log":"2023-05-02 08:19:40,773 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m14), 'b'deye/logger_status'', ... (7 bytes)\n","stream":"stdout","time":"2023-05-02T08:19:40.776460836Z"}
{"log":"2023-05-02 08:19:40,783 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 14)\n","stream":"stdout","time":"2023-05-02T08:19:40.786175542Z"}
{"log":"2023-05-02 08:19:40,790 - DeyeMqttClient - INFO - Logger is offline\n","stream":"stdout","time":"2023-05-02T08:19:40.794439291Z"}
{"log":"2023-05-02 08:19:40,797 - DeyeDaemon - INFO - Reading completed\n","stream":"stdout","time":"2023-05-02T08:19:40.801117089Z"}
`

My systems information: 

> Inverter SN: 2212xxxxxx
> Device(Logger?) SN: 3912xxxxxxx
> Firmware Version: MW3_16U_5406_1.53

The 3912xxxxx SN is what the Solarman App shows for Device - Logger.

BR

kbialek, 2023-05-02:
@AST70 I see the following options:
1. There is a bug in my modbus/tcp implementation. (also frame obfuscation does not help :slightly_smiling_face: )
2. Your logger does not support implemented protocol version

There is an alternative solarman protocol implementation available. https://github.com/jmccrohan/pysolarmanv5 It's a library that one can integrate into his project. It also has a logger scanner implemented. You may want to try it out and check if your logger responds.

closed 2023-06-17 by github-actions[bot]