kbialek/deye-inverter-mqtt #59: Portainer Support

edvandreas, 2023-04-14:
I already have some containers runnung using Portainer.

Can you please add Portainer support or a docker compose file?

Maybe someone is already using Portainer?

Thanks for you efforts!
Best Regards,
Andreas

kbialek, 2023-04-14:
Sure, I've added a docker-compose.yaml.

I've no idea what extra setup is required by Portainer. Anyhow I do not plan to add customization for any specific deployment tools. Imho that's way beyond the scope of this project.

edvandreas, 2023-04-16:
Thank you for your help.

I never had a Docker container with external env file before. I could not really get the config.env file inside the container so I moved the parameters inside the docker compose. Works pretty well.

`version: "3.3"
services:
  deye-mqtt:
    image: ghcr.io/kbialek/deye-inverter-mqtt:latest
    container_name: deye-mqtt
    networks:
      LAN:
        ipv4_address: 192.168.1.215
    environment:
      - PUID=1001 #optional
      - PGID=991 #optional
      - TZ=Europe/Berlin
      - DEYE_LOGGER_IP_ADDRESS=192.168.1.XXX
      - DEYE_LOGGER_PORT=8899
      - DEYE_LOGGER_SERIAL_NUMBER=27408XXXXX
      - MQTT_HOST=192.168.1.XXX
      - MQTT_PORT=1883
      - MQTT_USERNAME=user
      - MQTT_PASSWORD=password
      - MQTT_TOPIC_PREFIX=deye
      - LOG_LEVEL=INFO
      - DEYE_DATA_READ_INTERVAL=10
      - DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery
    restart: unless-stopped

networks:
  LAN:
    external:
      name: LAN`

closed 2023-04-16 by edvandreas