kbialek/deye-inverter-mqtt #61: FU-SUN-5K-G05 - read registers 150-218

kbialek, 2023-04-16:
Hi @kbialek,

As inverters are just rebranded but same HW, I tested on FU-SUN-5K-G05 (Fuji Solar) and get most of information but I miss data from Mobus connected Meter which data are on registers 150-218: 
PDF Document for Deye by "[parkpirat]" few posts earlier..

Registers dec 150 - 218 ? As I can see your script only pull range 003c-0075.
If i try to read it with your toll I get:
Day Load Power:
docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 200
int: 1194, l: 170, h: 4

Meter active power: registers 203 (low) and 204(high) I get:
int: 65027, l: 3, h: 254
int: 65535, l: 255, h: 255

How to pull this second range? Or I should create new instance with new range and only read new registers from that range.
Thank you

_Originally posted by @ivanK0dy in https://github.com/kbialek/deye-inverter-mqtt/issues/41#issuecomment-1510386960_

kbialek, 2023-04-16:
@ivanK0dy

A new metric group must be defined to support this register range. It can be named `string_extended` or something like this. This way it could be optionally enabled by users who are interested in these metrics. The file itself should be named `deye_sensors_string_extended.py` and should look similar to [deye_sensors_deye_sg04lp3.py](https://github.com/kbialek/deye-inverter-mqtt/blob/main/deye_sensors_deye_sg04lp3.py)

Would you be interested in contributing such file?

ivanK0dy, 2023-04-18:
Sure, it something that I would like to have for myself, and would like to contribute to community.

I added second range as you proposed in string_extended but now it just pool that range from "deye_sensors_string_extended.py" , not 1st one from deye_sensors.py:
# AC Phase 1
phase1_voltage_sensor = SingleRegisterSensor(
    "Phase1 Voltage", 0x49, 0.1, mqtt_topic_suffix='ac/l1/voltage', unit='V', groups=['string', 'micro', 'string_extended'])

I added sting_extended to the group list.. Any hint?

ivanK0dy, 2023-04-21:
Sure, let me just re-check all registers and verify that they show correct data so I'll share my findings (y).

closed 2023-06-05 by github-actions[bot]