kbialek/deye-inverter-mqtt #62: Compatability with secure-mqtt

devil33, 2023-04-18:
Hi there,

I just started off using the nice piece of software to store my readings locally.

For my MQTT broker I use mosquitto with port 8883, secure-mqtt.
I would like to know if there are plans to implement

a) Full secure-mqtt support with custom CAs
b) secure-mqtt with options to at least ignore TLS verification errors.

I can type down some python but would not consider myself a programmer.
Nevertheless, I like to offer my help if needed.

Thanks
devil33

kbialek, 2023-04-18:
This will have to wait until I have time to implement it.

kbialek, 2023-04-23:
@devil33 

Could you test this beta image? `ghcr.io/kbialek/deye-inverter-mqtt:2023.04.2-beta-2`

Instructions: https://github.com/kbialek/deye-inverter-mqtt/tree/feature/mqtt-tls#connecting-to-mqtt-broker-over-tls

kbialek, 2023-04-25:
correct, that's mtls

closed 2023-05-27 by kbialek