kbialek/deye-inverter-mqtt #66: timeouts in logger

kptkip, 2023-04-22:
I get the following info in the log-file:

2023-04-22 23:25:56,760 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_battery, range: 0202-022e]
2023-04-22 23:26:46,807 - DeyeConnector - WARNING - Too many connection timeouts
2023-04-22 23:26:46,807 - DeyeModbus - ERROR - No response frame
2023-04-22 23:26:46,807 - DeyeModbus - ERROR - Modbus frame is too short or empty

finfin82, 2023-04-23:
more information missing... what is your refreshrate? etc. same wifi?
the logger seems to be pretty laggy anyway

kptkip, 2023-04-23:
The docker container runs on a NAS in the same Network but connected via LAN.

Here is my config.env:
`DEYE_LOGGER_IP_ADDRESS=192.168.33.X
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=[Serial]
DEYE_METRIC_GROUPS=deye_sg04lp3_battery
DEYE_FEATURE_SET_TIME=true

MQTT_HOST=192.168.33.Y
MQTT_PORT=1884
MQTT_USERNAME=
MQTT_PASSWORD=
MQTT_TOPIC_PREFIX=SmartHome/Deye

LOG_LEVEL=INFO
DEYE_DATA_READ_INTERVAL=30 `

kbialek, 2023-04-23:
hi @kptkip , please follow the instructions here https://github.com/kbialek/deye-inverter-mqtt#network-connectivity-issues

kptkip, 2023-04-24:
Connection works like a charm (s. screenshot attached)
Logs look like this (also attached)
[image]
[image]

kbialek, 2023-04-24:
That's interesting. @kptkip Could you fill in this information?
- Inverter model:
- Logger firmware version:
- deye-inverter-mqtt version:

kptkip, 2023-04-24:
Inverter model: SUN-8K-SG04LP3-EU
Logger firmware version: LSW3_15_FFFF_1.0.96
deye-inverter-mqtt version: 2023.04.1

kbialek, 2023-04-24:
@edvandreas @hackepeterOli you own similar inverters. What is the logger firmware version at your devices? I suspect that @kptkip logger has incompatible firmware or is not properly configured. Maybe modbus/tcp must be enabled or something.

edvandreas, 2023-04-25:
Here are the details from my working wifi stick:
Logger Firmware: LSW3_15_FFFF_1.0.92R
Inverter: SUN-12K-SG04LP3-EU

hackepeterOli, 2023-04-25:
Here is my logger Firmware: 
LSW3_15_FFFF_1.0.9E
you can contact the Support (in solarman app) for an update. Maybe the wifi signal is to bad?! Works a ping-t?

kptkip, 2023-04-25:
According to the numbers it seems, that I have the newest FW of us three, right?
WiFi is brillant - no hazzle.

hackepeterOli, 2023-04-25:
Last week i get a Firmware-Update from Deye…

kptkip, 2023-04-25:
OK, I'll ask them, Do you have a special Contact adress or simply the info@...?

hackepeterOli, 2023-04-25:
[service@deye.com.cn], you must tell the SN from the stick.

kptkip, 2023-04-27:
This is the response from service@deye.com.cn:

_Hello, it is not recommended to update the firmware of the collector. If you need it, I can update your inverter firmware_

kbialek, 2023-04-27:
@kptkip Sad to hear it.
If you are in contact with deye support, you may ask them if modbus/tcp needs to be enabled at the logger
Also let's check some basic stuff:
1. Please make sure that you've set **logger** S/N in the config. Some ppl set **inverter** S/N. That will not work.
2. Have you tried to run it with `deye_sg04lp3` metric group?

gwest7, 2023-05-19:
Probably a different timeout from the above:
```
DeyeConnector - DEBUG - Connection timeout
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/deye_connector.py", line 48, in send_request
    data = client_socket.recv(1024)
ConnectionResetError: [Errno 104] Connection reset by peer
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 43, in <module>
    main()
  File "/opt/deye_inverter_mqtt/./deye_docker_entrypoint.py", line 39, in main
    daemon_main()
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 97, in main
    daemon.do_task()
  File "/opt/deye_inverter_mqtt/deye_daemon.py", line 66, in do_task
    regs |= self.modbus.read_registers(reg_range.first_reg_address, reg_range.last_reg_address)
  File "/opt/deye_inverter_mqtt/deye_modbus.py", line 47, in read_registers
    resp_frame = self.connector.send_request(req_frame)
  File "/opt/deye_inverter_mqtt/deye_connector.py", line 60, in send_request
    self.__log.warn("Connection error", e.message)
AttributeError: 'ConnectionResetError' object has no attribute 'message'
```

Inverter `SUN-5K-SG01LP1-EU`
Logger firmware `LSW3_15_FFFF_1.0.96`
Image `2023.05.2` and `2023.05.3-beta-1`

It runs smoothly for about an hour until the error occurs.

kbialek, 2023-05-19:
@gwest7 thanks for reporting. This is a pretty stupid bug in the error handling. It's not related to the original problem reported in this issue.

palto42, 2023-05-21:
@kbialek I came across the same issue with exception handling yesterday. My local fix to this is:

```
except Exception as e:
                    self.__log.warn("Connection error: %s", getattr(e, "message", e.__class__.__name__))
```

This would log the exception message if the exception has such attribute, otherwise it would just log the exception name.

kptkip, 2023-06-05:
> 

tried both - SN is from the logger - not from the inverter.

I asked deye tech support - but they don't understand, what I want from them

kbialek, 2023-06-05:
@kptkip Can we check if there are response frames being sent by the logger? Could you record the network traffic at port 8899? E.g. using `tcpdump` or `wireshark`?

@palto42 I'm tagging you here, because of your network connectivity expertise.

kptkip, 2023-06-05:
to be honest - I#m not familiar with tcpdump...
THis is the output:
[image]

not really valuable...

kptkip, 2023-06-05:
so you are running this command inside the docker container on port 8899?

kbialek, 2023-06-05:
No, at the host

kbialek, 2023-06-06:
An example of bidirectional communication
```
$ sudo tcpdump -i docker0 port 8899 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on docker0, link-type EN10MB (Ethernet), capture size 262144 bytes
06:12:11.545722 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [S], seq 3718790533, win 64240, options [mss 1460,sackOK,TS val 929913265 ecr 0,nop,wscale 7], length 0
06:12:11.552095 IP 192.168.2.90.8899 > 172.17.0.3.37174: Flags [S.], seq 11502, ack 3718790534, win 4380, options [mss 1460], length 0
06:12:11.552243 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [.], ack 1, win 64240, length 0
06:12:11.552413 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [P.], seq 1:37, ack 1, win 64240, length 36
06:12:11.717820 IP 192.168.2.90.8899 > 172.17.0.3.37174: Flags [.], ack 37, win 4344, length 0
06:12:28.382301 IP 192.168.2.90.8899 > 172.17.0.3.37174: Flags [P.], seq 1:149, ack 37, win 4344, length 148
06:12:28.382613 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [.], ack 149, win 64092, length 0
06:12:28.383057 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [F.], seq 37, ack 149, win 64092, length 0
06:12:28.385099 IP 192.168.2.90.8899 > 172.17.0.3.37174: Flags [.], ack 38, win 4343, length 0
06:12:28.385811 IP 192.168.2.90.8899 > 172.17.0.3.37174: Flags [F.], seq 149, ack 38, win 4343, length 0
06:12:28.385906 IP 172.17.0.3.37174 > 192.168.2.90.8899: Flags [.], ack 150, win 64092, length 0
06:17:11.545843 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [S], seq 2868053599, win 64240, options [mss 1460,sackOK,TS val 930213265 ecr 0,nop,wscale 7], length 0
06:17:11.549402 IP 192.168.2.90.8899 > 172.17.0.3.37324: Flags [S.], seq 14190, ack 2868053600, win 4380, options [mss 1460], length 0
06:17:11.549577 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [.], ack 1, win 64240, length 0
06:17:11.549730 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [P.], seq 1:37, ack 1, win 64240, length 36
06:17:11.640422 IP 192.168.2.90.8899 > 172.17.0.3.37324: Flags [.], ack 37, win 4344, length 0
06:17:11.783443 IP 192.168.2.90.8899 > 172.17.0.3.37324: Flags [P.], seq 1:149, ack 37, win 4344, length 148
06:17:11.783597 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [.], ack 149, win 64092, length 0
06:17:11.784566 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [F.], seq 37, ack 149, win 64092, length 0
06:17:11.786367 IP 192.168.2.90.8899 > 172.17.0.3.37324: Flags [.], ack 38, win 4343, length 0
06:17:11.787127 IP 192.168.2.90.8899 > 172.17.0.3.37324: Flags [F.], seq 149, ack 38, win 4343, length 0
06:17:11.787286 IP 172.17.0.3.37324 > 192.168.2.90.8899: Flags [.], ack 150, win 64092, length 0
```

kbialek, 2023-06-06:
> Now, it works (don't ask me why).

:+1: 

> Do I need two docker instances (with different metric groups) if I want inverter data and battery data?

No, just list the metric groups you want.
This is documented https://github.com/kbialek/deye-inverter-mqtt#configuration

closed 2023-07-22 by github-actions[bot]