kbialek/deye-inverter-mqtt #69: Error if mqtt-service not startet

bismosa, 2023-05-03:
**Describe the bug**
I am using a init.d script to start this automatically after a reboot.
If the mqtt service is not ready i get the following error:

```
Traceback (most recent call last):
  File "/root/deye/deye_docker_entrypoint.py", line 43, in <module>
    main()
  File "/root/deye/deye_docker_entrypoint.py", line 39, in main
    daemon_main()
  File "/root/deye/deye_daemon.py", line 95, in main
    daemon = DeyeDaemon(config)
  File "/root/deye/deye_daemon.py", line 40, in __init__
    mqtt_client = DeyeMqttClient(config)
  File "/root/deye/deye_mqtt.py", line 37, in __init__
    self.__mqtt_client.connect(config.mqtt.host, config.mqtt.port, keepalive=60)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
  File "/usr/local/lib/python3.10/socket.py", line 845, in create_connection
    raise err
  File "/usr/local/lib/python3.10/socket.py", line 833, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```

**To Reproduce**
Reboot and try to start while the mqtt service is not fully started

**Expected behavior**
Retry after a few seconds without exit the service

**Hardware (please complete the following information):**
 - Inverter model:deye sun600g3-eu-230
 - Hardware: Raspberry Pi (3b)

**Software (please complete the following information):**
 - Logger firmware version: MW3_16U_5406_1.57
 - deye-inverter-mqtt version: V1.1.00.10 (?)
 - config.env (remove passwords before pasting):
```
DEYE_LOGGER_IP_ADDRESS=192.168.1.139
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=41xxxxxxxx

MQTT_HOST=192.168.1.3
MQTT_PORT=1883
MQTT_USERNAME=user
MQTT_PASSWORD=password
MQTT_TOPIC_PREFIX=deye

LOG_LEVEL=INFO
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=micro
```

**Additional context**

Many thanks for your great Work!
Regards

kbialek, 2023-05-04:
@bismosa Could you share your init.d script so I can better understand what you are doing?

bismosa, 2023-05-04:
It is enough if I start the following script immediately after starting the Raspberry pi:
```
#!/bin/sh
cd /root/deye
bash -c "$(cat config.env | xargs) python3.10 deye_docker_entrypoint.py" bash -c "$(cat config.env | xargs) python3.10 deye_docker_entrypoint.py"
```
I can also share my init.d script. But actually that doesn't matter.
If I wait a bit before starting the script, the service runs as expected.

Regards

kbialek, 2023-05-04:
Running this tool outside of the docker is not supported and not documented :slightly_smiling_face: 
Anyway, how do you access your raspi? Over ssh, or do you have display and keyboard connected to it?

bismosa, 2023-05-05:
oops 😄  Feel free to close this issue.

I have the command to start the service from here:
https://github.com/kbialek/deye-inverter-mqtt/issues/22#issuecomment-1412820076

I have several services running on the Raspberry. I have different reasons why I don't want to run this in the docker container.
Once started, the service runs without any problems. Then I'll add a "start delay" myself. No problem.
But the "error" only seems to occur while the connection cannot yet be established at startup.

I now have 4 Raspberrys running. All for different tasks (1x Kodi, 2x surveillance camera, 1x home automation). I use all of them without a graphical interface and only access them via ssh. If necessary, I also connect a monitor and a keyboard. But rather rarely.

Regards

kbialek, 2023-05-05:
I'm asking about display and keyboard, because the error indicates that the networking is not working properly after startup. But if you connect to raspi over ssh, then the problem is elsewhere. Maybe you have some additional scripts that configure network (e.g. iptables), or do you use eth and wifi at your raspi? wifi surely needs some time to connect.

I strongly doubt if the problem is in the bridge service itself. You can use `telnet` to check if network connectivity is okay.

bismosa, 2023-05-05:
The network is already up and running if i try to start this service.
I only use ethernet. Wifi is disabled.

The MQTT service needs some time to start. Presumably this service also starts before the Mosquitto is started.
I have a lot of services running. So I'm not surprised that there are delays at the start.

I'll just add a delay in my startup script. Then it works without any problems.

Regards

kbialek, 2023-05-05:
So your MQTT broker runs on this same raspi as deye-inverter-mqtt service, correct?

bismosa, 2023-05-06:
Yes this is correct. 
I have now found that I currently even have 2 MQTT servers running.
Mosquito: Port 1888
FHEM: port 1883
Fhem (a home automation software) also has its own MQTT server.
I used this for different services. I don't think I have anything running on port 1888 anymore. I think I need to investigate this again. I guess I won't need the Mosquitto anymore.

So it wasn't the start time of Mosquitto but of FHEM.

I have now solved the problem by adapting my startscript a bit:
```
#!/bin/sh
while ! nc -z localhost 1883; do   
  sleep 0.1 # wait for 1/10 of the second before check again
  echo "Port 1883 noch nicht verfügbar!"
done

cd /root/deye
bash -c "$(cat config.env | xargs) python3.10 deye_docker_entrypoint.py" bash -c "$(cat config.env | xargs) python3.10 deye_docker_entrypoint.py"
```
Now the service starts without any problems even after a reboot.

Thanks for your help 😃 

Regards

kbialek, 2023-05-08:
btw, if you use systemd, here is the proper solution for your problem https://www.baeldung.com/linux/systemd-service-start-order

closed 2023-05-10 by kbialek