kbialek/deye-inverter-mqtt #74: SUN-5K-SG03LP1-EU support

Jim5975, 2023-05-12:
Hi Hope you can help, I'm getting some data from the current version, but I'm looking to also pull the Battery Charge, Discharge and current capacity. I'm open to trying to read the Modbus address but can't figure out where to start. 

**Hardware (please complete the following information):**
 - Inverter model: SUN-5K-SG03LP1-EU

**Software (please complete the following information):**
 - Logger firmware version:LSW3_15_FFFF_1.0.92R
 - deye-inverter-mqtt version:2023.5.2
 - config.env (remove passwords before pasting):

DEYE_LOGGER_IP_ADDRESS=10.22.20.231
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=2730158931

MQTT_HOST=10.22.1.2
MQTT_PORT=1883
MQTT_USERNAME=
MQTT_PASSWORD=
MQTT_TOPIC_PREFIX=deye

LOG_LEVEL=DEBUG
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=string

edvandreas, 2023-05-13:
You need to change DEYE_METRIC_GROUPS=string to DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery to read the correct values including the battery

kbialek, 2023-05-14:
@Jim5975 Please follow the suggestion from @edvandreas 
Please let me know if it works this way, so I can update the documentation

kbialek, 2023-05-24:
@Jim5975 Meanwhile I've added a new `deye_hybrid` metrics group. It might work with your inverter.

closed 2023-07-09 by github-actions[bot]