kbialek/deye-inverter-mqtt #75: Hello,  is it possible to track more than one inverter? I tried to run docker with deye-mqtt2 but in samrthome both inverter send to same variable

Jagu1967, 2023-05-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kbialek, 2023-05-14:
Yes, it is possible. Read this comment https://github.com/kbialek/deye-inverter-mqtt/issues/46#issuecomment-1479541330

Jagu1967, 2023-05-14:
thanks, perfect :-)