kbialek/deye-inverter-mqtt #76: SUN-5K-SG01LP1-EU compatibility

gwest7, 2023-05-18:
MQTT messages are published, but besides the mostly 0.0 values the rest are extremes. For example `dc.pv1.power = 59395.0`, `ac.l1.current = -61.4` and `ac.temperature = -100.0`.

I realise the binary (modbus) packet sizes are probably very different from what this library expects. I hope I can help in some way?

DEYE_METRIC_GROUPS: tried both `deye_sg04lp3_battery` and then `deye_sg04lp3`.

kbialek, 2023-05-18:
hi @gwest7 

I think your inverter model is not supported by this tool so far. 

Here, I've added the hybrid inverter support in an experimental build for you. Use this image `ghcr.io/kbialek/deye-inverter-mqtt:2023.05.3-beta-1`. Set metric groups to `deye_hybrid,deye_hybrid_battery`

Let me know if it works

gwest7, 2023-05-19:
I did as you suggested and it all looks good. It's hard to compare with the Deye Android app as the reading fluctuate between the MQTT messages and the Deye app refresh.

Here is the last output:
```
 paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-2748121764'
paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'tele/deye/status'', ... (6 bytes)
DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.0.5:1883
DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
DeyeDaemon - INFO - Reading start
DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0003-0070]
paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
DeyeConnector - DEBUG - Request frame: a5170010450000a402cda302000000000000000000000000000001030003006e34265315
DeyeConnector - DEBUG - Response frame: a5ef0010150004a402cda302014bad290083110000ba5e3d640103dc32333032313630303630000081000000151500003382c35d0000c35000000201000000ff000a170513072729000113880514000000000000000027100bb81f4007d003520000000003e800000000000000010000000100000000000000000000000000001964000000000000000100010002fffa00fa0000fd83ffff00000638ff2800000000000e000a0422000002430000001c00000e411388000000000000000000160bc200000a8e00000000054305120000000107d0000000000000fe60ffff0a0100000000000000000000000000c80000000800000014000145846615
DeyeDaemon - INFO - Reading registers [metrics group: deye_hybrid_battery, range: 0096-00f8]
DeyeConnector - DEBUG - Request frame: a5170010450000a402cda3020000000000000000000000000000010300960063e5cf3515
DeyeConnector - DEBUG - Response frame: a5d90010150005a402cda302014bad290083110000ba5e3d640103c6092f0000092f0000092f0000092f0959000000000136000001680000005a00000000029200000292000000000000ffc30000ffc30255000002550255000000950481160f006402550000000000000000ffd2ffad138b138b000100100000000000000000000116801680159000c811940014005a00000000003c0050000000010001000803de00140032002311f81450128e00f000001324001e00281324001e00280000000100000000000013ec005f151800640000000000030001000113880000000000ff1f9bd215
DeyeDaemon - DEBUG - PV1 Power: 0.0
DeyeDaemon - DEBUG - PV2 Power: 0.0
DeyeDaemon - DEBUG - PV1 Voltage: 0.8
DeyeDaemon - DEBUG - PV2 Voltage: 2.0
DeyeDaemon - DEBUG - PV1 Current: 0.0
DeyeDaemon - DEBUG - PV2 Current: 0.1
DeyeDaemon - DEBUG - Daily Production: 0.0
DeyeDaemon - DEBUG - Total Production: 0.0
DeyeDaemon - DEBUG - Micro-inverter Power: 0.0
DeyeDaemon - DEBUG - Total Battery Charge: 105.8
DeyeDaemon - DEBUG - Total Battery Discharge: 57.9
DeyeDaemon - DEBUG - Daily Battery Charge: 1.4
DeyeDaemon - DEBUG - Daily Battery Discharge: 1.0
DeyeDaemon - DEBUG - Battery Status: 0.0
DeyeDaemon - DEBUG - Battery Power: -46.0
DeyeDaemon - DEBUG - Battery Voltage: 56.5
DeyeDaemon - DEBUG - Battery SOC: 100.0
DeyeDaemon - DEBUG - Battery Current: -0.8
DeyeDaemon - DEBUG - Battery Temperature: 15.3
DeyeDaemon - DEBUG - Total Grid Power: 658.0
DeyeDaemon - DEBUG - Grid Voltage L1: 235.1
DeyeDaemon - DEBUG - Grid Voltage L2: 0.0
DeyeDaemon - DEBUG - Internal CT L1 Power: 658.0
DeyeDaemon - DEBUG - Internal CT L2 Power: 0.0
DeyeDaemon - DEBUG - External CT L1 Power: 0.0
DeyeDaemon - DEBUG - External CT L2 Power: 0.0
DeyeDaemon - DEBUG - Daily Energy Bought: 2.8
DeyeDaemon - DEBUG - Total Energy Bought: 32768364.9
DeyeDaemon - DEBUG - Daily Energy Sold: 0.0
DeyeDaemon - DEBUG - Total Energy Sold: 0.0
DeyeDaemon - DEBUG - Total Power: -61.0
DeyeDaemon - DEBUG - Current L1: 0.9
DeyeDaemon - DEBUG - Current L2: 0.0
DeyeDaemon - DEBUG - Inverter L1 Power: -61.0
DeyeDaemon - DEBUG - Inverter L2 Power: 0.0
DeyeDaemon - DEBUG - Load Frequency: 50.0
DeyeDaemon - DEBUG - DC Temperature: 34.7
DeyeDaemon - DEBUG - AC Temperature: 29.8
```

Thank you for this! Let me know if there is anything else you'd like me to do.

kbialek, 2023-05-19:
Looks good. Watch it please for a day or two. Check if the values make sense, also when there is more sunlight outside. Let me know the results.

closed 2023-05-22 by kbialek