kbialek/deye-inverter-mqtt #81: Adding UPS Load values

edvandreas, 2023-05-25:
The DEYE sg04lp3 Inverters have a UPS function with a separate load connection.

Can we please add these Modbus Registers to the metric group?

|Metric|Modbus address|MQTT topic suffix|Unit|
|UPS load-side phase power A|640|`ac/ups1/power`|W|
|UPS load-side phase power B|641|`ac/ups2/power`|W|
|UPS load-side phase power C|642|`ac/ups3/power`|W|
|UPS load-sidetotal power|643|`ac/ups_total/power`|W|

Thanks,
Andy

gwest7, 2023-05-25:
Are these header locations? Could we get the locations for the grid values too?

[image]

kbialek, 2023-05-27:
@gwest7 The service reports grid voltages at these MQTT topics. You may see differences because the service and the app are sampling the values at different points in time.
```
ac/l1/voltage
ac/l2/voltage
ac/l3/voltage
```

kbialek, 2023-05-27:
@edvandreas Are you referring to this part of the HA mapping?
```
    - name: "Load L1 Power"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x028A]
      icon: 'mdi:lightning-bolt-outline'

    - name: "Load L2 Power"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x028B]
      icon: 'mdi:lightning-bolt-outline'
      
    - name: "Load L3 Power"
      class: "power"
      state_class: "measurement"      
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x028C]
      icon: 'mdi:lightning-bolt-outline'
```

These are not currently mapped to MQTT, and I'm asking because these are regs 650,651,652 and not 640,641,642 as you suggested.

edvandreas, 2023-05-27:
> @edvandreas Are you referring to this part of the HA mapping?
>
> ```
>     - name: "Load L1 Power"
>       class: "power"
>       state_class: "measurement"
>       uom: "W"
>       scale: 1
>       rule: [...]

I'm not sure which regs are correct. But the Inverter makes a difference between classic home load and UPS load. The regs I recommended are from the modbus documentation and refers to the UPS Load.

Is there an easy way to read regs? I could identify them by comparing to the app or directly at the inverter.

kbialek, 2023-05-27:
> Is there an easy way to read regs? I could identify them by comparing to the app or directly at the inverter.

Yes, see https://github.com/kbialek/deye-inverter-mqtt#reading-and-writing-raw-register-values

edvandreas, 2023-05-27:
> > Is there an easy way to read regs? I could identify them by comparing to the app or directly at the inverter.
> 
> Yes, see https://github.com/kbialek/deye-inverter-mqtt#reading-and-writing-raw-register-values

I will try but since I'm using Portainer (20+ Containers to manage) it's not that easy.

Can I add additional registers in the env file?

kbialek, 2023-05-27:
> Can I add additional registers in the env file?

There is no such a feature

> I will try but as I'm using Portainer (20+ Containers to manage) it's not that easy.

No need to run it on the server. You can run it on your laptop directly.

edvandreas, 2023-05-27:
> > Can I add additional registers in the env file?
> 
> There is no such a feature
> 
> > I will try but as I'm using Portainer (20+ Containers to manage) it's not that easy.
> 
> No need to run it on the server. You can run it on your laptop directly.

Did it. I read the registers 640, 641 and 642. The values looking good.  643 should be the summary of the values but gives an error but I can live with that. How I can get these registers to the mqtt broker?

kbialek, 2023-05-27:
> How I can get these registers to the mqtt broker?

We will have to define them and add to the metric group

It's quite surprising, as HomeAssistant mapping uses regs 650,651,652. Could you also check them?

edvandreas, 2023-05-28:
I compared 653 (summary of 650/651/652) with 643 (summary of 640/641/642) and they look almost the same.

Can we add 65x to the metric group please?

kbialek, 2023-05-28:
@edvandreas Test this beta image `ghcr.io/kbialek/deye-inverter-mqtt:2023.05.4-beta-3` You have to enable additional metric group `deye_sg04lp3_ups`

edvandreas, 2023-05-28:
@kbialek Nice work! Looking good so far.
[image]

closed 2023-05-28 by kbialek

gwest7, 2023-05-29:
I currently use the metric groups `deye_hybrid,deye_hybrid_battery ` for a SUN-5/6K-SG01LP1-EU inverter. I went ahead and read the registers above, but might be missing something as I get zeros all round.
```
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 640
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 641
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 642
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 643
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 650
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 651
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 652
int: 0, l: 0, h: 0
% docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 653
int: 0, l: 0, h: 0
```

kbialek, 2023-05-29:
@gwest7 No surprise, hybrid inverters are using different set of modbus registers. Does your inverter also have UPS feature?