kbialek/deye-inverter-mqtt #83: Remove duplicate log messages in case of connectivity issues

palto42, 2023-05-28:
The current error handling for IP connectivity issues and empty data responses from `DeyeConnector.send_request()` creates multiple warning and error messages. 

I think it would be better to only submit the root cause error and suppress related follow up issues.

In below example the 2 errors are just a consequence of the first socket connection issue:

```text
2023-05-28 13:42:09,002 - DeyeConnector - WARNING - Could not open socket on IP 192.168.0.123
2023-05-28 13:42:09,003 - DeyeModbus - ERROR - No response frame
2023-05-28 13:42:09,003 - DeyeModbus - ERROR - Modbus frame is too short or empty
```

Similar an empty data frame received would create a warning log message `No data received` in `send_request`  and then again the same error logs in `DeyeModBus`.

**Proposed solution:**

1. Increase log severity in DeyeConnector to ERROR if socket connection fails or no data was received. 
2. If `send_request` in `DeyeModBus` received an empty `bytearray` back, then directly return an empty dict  in case of a read or `False` in case of a write attempt.
3. Remove "empty data" check in `__extract...` functions (obsolete)

One more related code section which is unclear to me is the socket receive exception handling:

```python
try:
    data = client_socket.recv(1024)
    try:
        data
    except:
        self.__log.error("No data received")
    self.__log.debug("Response frame: %s", data.hex())
    return data
```

What is the reason for the second `try`? It doesn't really check for empty data but in fact for data variable undefined, which should never happen if the previous ` client_socket.recv` hasn't raised an exception (in which case the second `try` would not be reached). For a real check on empty data it would need to be changed to `if data:`. What I'm also missing is a `break` after the except log command, currently it just continues with the success path, returning the (undefined) data. Should it retry in case of empty data or return an empty bytearray?

If you agree, I can work on a PR to implement the changes/improvements.

kbialek, 2023-05-28:
An improvement is always welcome.

Regarding the exception handling and the validation of`data`. Yes, I think you're right. I took this code over from some other project at the time when I was starting coding this tool as a pet project for myself. At that time I did not put much attention to these details.

> Should it retry in case of empty data or return an empty bytearray?

Imho it should retry, similarly like in the case of timeout

palto42, 2023-05-29:
@kbialek I'm thinking about some more improvements in the connect module.
One improvement could be to suppress subsequent connect errors and instead log a "re-connect" on next successful connect. This would be useful to avoid lots of connect errors during the downtime at night.
As I was looking into this, I was wondering why the socket connection is re-created each time a `send_request`is called. It should be more efficient to keep the socket connection in the `DeyeConnect` class and re-use it. This approach would require an automatic re-connect in case the connection is broken.

kbialek, 2023-05-29:
> One improvement could be to suppress subsequent connect errors and instead log a "re-connect" on next successful connect. This would be useful to avoid lots of connect errors during the downtime at night.

Makes sense

> It should be more efficient to keep the socket connection in the `DeyeConnect` class and re-use it. This approach would require an automatic re-connect in case the connection is broken.

Probably, but what real life problem would it solve? The existing implementation is already proven to work fine. On the other hand, re-connecting logic means more code, maybe more bugs and corner cases to cover. This project does not need it.

palto42, 2023-05-31:
>> It should be more efficient to keep the socket connection ...
>
> Probably, but what real life problem would it solve? 

Tried the idea with maintaining the socket connection, but it was not improving anything. So agree that it doesn't really solve any problem. It looked useful if the interval would be short, but the required handling of broken connections made it rather complex.

>> One improvement could be to suppress subsequent connect errors ...
>
> Makes sense

One thing I learned during this exercise is that there is a more efficient new higher level  `socket.create_connection()` function which could significantly simplify the related code part. I changed the code to use it because it made the implementation of the suppression during disconnect (night) easier/cleaner.

I have another question regarding the retry implementation. Currently the `send_request` function only retries the data receive part in case of issues like timeout or empty data. Shouldn't it better re-connect and re-send the request data command? I don't have much experience with the behavior of the Deye inverters, but to me it looks unlikely that the current retries are successful. If there could be a response after the 10 seconds timeout, then we could just increase the timeout instead of retrying in a loop.

kbialek, 2023-05-31:
> I have another question regarding the retry implementation. Currently the `send_request` function only retries the data receive part in case of issues like timeout or empty data. Shouldn't it better [...]

Good point. I've borrowed this method from another project at the very beginning, and thus I cannot answer your question. We could start logging, how many retries is needed for a successful read, and this way check if the retry loop works as intended.

kbialek, 2023-06-01:
> I added the number of attempts in the logs and found that in ~7% of the requests it took 2 attempts, all other only 1

So this means, that the retry logic actually works

palto42, 2023-06-02:
Retry logic works, but since it's only used for timeout (never saw empty response and doubt if this will ever happen), it is easier to increase the timeout instead.

I have tried without retry-loop and increased the timeout to 15 s (configurable) and this works as good as with the loop.

One wired thing I saw in the logs is that there is actually a pattern for the timeout: there is a timeout warning every ~30.5 minutes (my request interval is 30 s):

```text
2023-06-02 10:26:31,736 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 10:57:01,745 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 11:27:31,743 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 11:58:01,745 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 12:28:31,743 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 12:59:01,743 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 13:29:31,732 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 14:00:01,742 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
2023-06-02 14:30:01,742 - DeyeConnector - WARNING - Connection response timeout after 15 seconds
``` 

Maybe the inverter is busy with something else every 1/2 h ? My inverter is not connected to the internet, so maybe it tries in some intervals to connect to the cloud service (just guessing).

Since the logger usually responds in a second, I will try with reduced timeout. For me a longer one doesn't make sense since the interval is only 30 s and the total timeout should be less.

kbialek, 2023-06-02:
I'm super busy last days. Haven't even tested your other branch yet. 

You are doing a great job here with this research :+1:

> Maybe the inverter is busy with something else every 1/2 h ? My inverter is not connected to the internet, so maybe it tries in some intervals to connect to the cloud service (just guessing).

This is very likely. You can try to spot this by intercepting the network traffic (DNS queries and HTTP(S) requests sent by the logger)

palto42, 2023-06-10:
Delivered in [2023.06.1](https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.06.1)

closed 2023-06-10 by palto42