kbialek/deye-inverter-mqtt #85: docker buildx builder not removed after build

palto42, 2023-05-29:
**Describe the bug**
After running `make docker-build-local` the buildx builder container is not removed as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Check with `docker ps --filter "name=buildx.*"` that there are no active con
2. Run  `make docker-build-local` 
3. Check with `docker ps --filter "name=buildx.*"` that the build instance was not removed
```console
docker ps --filter "name=buildx.*"
CONTAINER ID   IMAGE                           COMMAND       CREATED         STATUS         PORTS     NAMES
be4f1c72e7eb   moby/buildkit:buildx-stable-1   "buildkitd"   8 minutes ago   Up 8 minutes             buildx_buildkit_deye-docker-build0
```

I observed that each `make build` created a new buildx instance. They are not removed because they don't seem to go into `inactive` state after the build.

Using docker version 24.0.1, build 6802122

**Expected behavior**
As per `Makefile` docker build section last command `@docker buildx rm --all-inactive --force`, it was expected that the build instance should be removed after the build.

**Proposed fix**
In order to correctly remove the buildx instance after the build has been completed, I suggest to create the instance with a name and then remove the instance by name:

```text
@docker buildx create --use --name deye-docker-build
@docker buildx build \
	--platform $* \
	--output type=docker \
	-t deye-inverter-mqtt:$(VERSION) \
	-t deye-inverter-mqtt:latest \
	.
@docker buildx rm deye-docker-build
```

kbialek, 2023-05-29:
Will you open a PR, or should I change it right away?

palto42, 2023-05-29:
Either way is fine for me.

palto42, 2023-06-01:
PR merged

closed 2023-06-01 by palto42