kbialek/deye-inverter-mqtt #88: 2023.05.4 breaks with Python < 3.10

aresius, 2023-06-01:
This is not necessarily a bug, but something that imho should have been mentioned in the release notes at least.

I do not know if you are aware of this, but using the new New Type Union Operator (PEP 604) destroys compatibility to python <3.10

Perhaps this does not matter with the docker solution since python is always new in there, but in my case on a Debian based Raspberry Pi I am stuck with python 3.9 until next Debian release in this summer (probably in august).

If you are willing to keep Python 3.9 compatibility for a little longer, here is how this can be achieved:

```diff
diff --git a/deye_config.py b/deye_config.py
index dce3b41..6588b81 100644
--- a/deye_config.py
+++ b/deye_config.py
@@ -18,13 +18,15 @@
 import os
 import ssl
 
+from typing import Union
+
 
 class DeyeMqttTlsConfig():
     def __init__(self,
                  enabled: bool = False,
-                 ca_cert_path: str | None = None,
-                 client_cert_path: str | None = None,
-                 client_key_path: str | None = None,
+                 ca_cert_path: Union[str, None] = None,
+                 client_cert_path: Union[str, None] = None,
+                 client_key_path: Union[str, None] = None,
                  tls_version = ssl.PROTOCOL_TLSv1_2,
                  insecure = False,
                  ):
@@ -46,7 +48,7 @@ class DeyeMqttTlsConfig():
         )
 
 class DeyeMqttConfig():
-    def __init__(self, host: str, port: int, username: str|None, password: str|None, topic_prefix: str,
+    def __init__(self, host: str, port: int, username: Union[str, None], password: Union[str, None], topic_prefix: str,
                  availability_topic: str = 'status',
                  logger_status_topic: str = 'logger_status',
                  tls=DeyeMqttTlsConfig()):
```

kbialek, 2023-06-01:
This project is distributed as a docker image from day one. Exactly to avoid this kind of problems. If you want to run it differently, then it is your responsibility to setup the environment as needed. If you don't want to use docker, you may consider using `podman`, or eventually use `pyenv` to install python 3.10 along python 3.9 in a non-intrusive way.

aresius, 2023-06-01:
Alright, just wanted to mention that and perhaps someone comes along and wants to use the patch in his installation as I will in mine until summer.

closed 2023-06-01 by aresius