kbialek/deye-inverter-mqtt #89: Alternative/additional processor: Output data in a well-known, machine-readable format

hoegaarden, 2023-06-02:
**Is your feature request related to a problem? Please describe.**

It's not really a problem, more of an enhancement, I guess.
My personal main goal is, to get the inverter data into a local prometheus, without the need to e.g. parse the debug output or run my own message broker.
However, I could see this proposal open up way more possibilities for other users, and not just my specific usecase.

**Describe the solution you'd like**

    As a user  
    I want to get inverter data in a well-known, machine-readable format  
    So that I am more flexible on how / where I push, process, and store it

I think it might be useful, to just output the collected data on stdout, in a well-known format alongside or as an alternative to pushing them to mqtt. If that were the case, users could use complementary tools (e.g. [telegraf] with the [execd], [file], or [tail] plugin) to pick up the data and process / forward it in whatever way they want.

[telegraf]: https://github.com/influxdata/telegraf
[execd]: https://github.com/influxdata/telegraf/tree/master/plugins/inputs/execd
[file]: https://github.com/influxdata/telegraf/tree/master/plugins/inputs/file
[tail]: https://github.com/influxdata/telegraf/tree/master/plugins/inputs/tail

I just glanced the code, and I think it should be possible to do that by implementing another processor, similar to the `DeyeMqttPublisher`. This processor would be dis/enabled by a new config setting (e.g. `DEYE_FEATURE_STDOUT` or similar). When that processor is enabled, it would be called [here](https://github.com/kbialek/deye-inverter-mqtt/blob/d15f98a6848ccac189c034f5396a9f17e71c9a61/deye_daemon.py#L74) to process the events. It would then "just" put those events in a well-known format on stdout. Regarding the format I'd probably lean on the [list that telegraf supports][formats], with a slight preference for json, because that's probably the easiest to consume by other consumers than [telegraf].

[formats]: https://github.com/influxdata/telegraf/blob/master/docs/DATA_FORMATS_INPUT.md

Example usecases:
- `telegraf` runs `deye-inverter-mqqt` with the [execd] plugin, thus it is in control of the `deye-inverter-mqtt` child process and directly can consume data from its stdout
- `telegraf` and `deye-inverter-mqtt` run as separate containers in a kubernetes pod; they can use a fifo on a shared volume, kubernetes would be in control of the processes, `deye-inverter-mqtt`'s stdout can be redirected into the fifo, and `telegraf` can use the [tail] plugin to consume the data from the fifo
- In case (and I can really think of a good one, but still, might be useful for some folks) `deye-inverter-mqtt` runs on a different system than the consumer (e.g. telegraf), users should be able to use [`socat`][socat] or similar tools to push the data over the network.

[socat]: http://www.dest-unreach.org/socat/doc/socat.html

**Describe alternatives you've considered**

I thought about making `deye-inverter-mqtt` expose a metrics API, to be directly scraped by prometheus. This, however, would pull in a lot of dependencies to this project which are not strictly necessary and might be better handled by external tooling (e.g. telegraf).

**Additional context**

If something like that has a chance to be accepted, I'd be willing to PR that in.

kbialek, 2023-06-02:
hi @hoegaarden Interesting proposal. Let's discuss it a bit more. I have a couple of questions
1. stdout is already used to print service logs. How do you plan to combine it with proposed metrics logging?
2. Adding a feature is easy, maintaining it is harder. This enhancement goes beyond the scope of this project. How do you imagine the maintenance of the new code in the future?

hoegaarden, 2023-06-03:
> stdout is already used to print service logs. How do you plan to combine it with proposed metrics logging?

My preference is for tools that only put "data" / machine-readable stuff on stdout and messaging (logs, errors, ...) to stderr. This should make the tool in question play nice(r) in combination with other tools, e.g. in command pipelines.

I didn't check if this is or isn't the case for `deye-inverter-mqtt`, but it if not this could be a useful change anyway.

I understand that this could be a touchy subject and what is deemed "useful data" might be nuanced and subjective, thus in case that is not desirable, I can think of two other options:
- the output of my proposed processor has another config knob which let's the user configure the FD data needs to be sent to.
- `deye-inverter-mqtt` could check if stdout is a tty. Based on that it could make the assumption if stdout is consumed by a human or by another program and can behave accordingly (whatever that means would need to be hashed out).

> Adding a feature is easy, maintaining it is harder. This enhancement goes beyond the scope of this project. How do you imagine the maintenance of the new code in the future?

If this is out of scope or not is solely your call, definitely.

From the quick glance I had at the code, and how I imagine it, the proposed additional processor would be quite slim. It might just end up effectively being a wrapper around `json.dumps()`, dumping out all [`event`](https://github.com/kbialek/deye-inverter-mqtt/blob/d15f98a6848ccac189c034f5396a9f17e71c9a61/deye_daemon.py#L74)s, maybe with some slight restructuring of the input data.

Furthermore, it should be implemented in separation, have as little dependency on any internals of `deye-inverter-mqtt` as possible, and be tested in separation and at integration level. It could (and should) probably implemented in a way, so that it can be ripped out easily if needed. I am very happy with keeping the mqtt thing the main star of the show and my proposal just being a side-actor. 

Having said all that, this will sure incur maintenance cost. For the benefit of making this project quite more flexible, IMHO. If that's worth it sure is your call.

What I can do is to spike this out, and see if we can come to an agreement about if and how such a feature could be implemented and integrated. WDYT?

hoegaarden, 2023-06-03:
See https://github.com/kbialek/deye-inverter-mqtt/issues/89 for a quick & dirty implementation of my porposal.

kbialek, 2023-06-04:
> My preference is for tools that only put "data" / machine-readable stuff on stdout and messaging (logs, errors, ...) to stderr. This should make the tool in question play nice(r) in combination with other tools, e.g. in command pipelines.

This is useful for command line tools, like `grep` or `sed`, that can work in processing pipelines in the CLI. This project does not belong to this family. It's a long running daemon which communicates over the network rather than stdin/stdout. It is also supposed to be run on a server, as a docker container. In this world it's a common good practice to print the logs into stdout directly. This will stay, and reusing stdout for anything else is not an option.

As an alternative, your processor may print the output into an arbitrary file. Then you can `tail` this file in your `execd` for influx. This approach will preserve appropriate separation of concerns between the components.

hoegaarden, 2023-06-04:
In #90 I made the destination of the logs configurable. The default behavior stays as is: log to stdout. But users can set the config `LOG_STREAM` to `STDERR` to have the logs land on, well, stderr.
Would you be open to accept that?

kbialek, 2023-06-04:
> In #90 I made the destination of the logs configurable. The default behavior stays as is: log to stdout. But users can set the config `LOG_STREAM` to `STDERR` to have the logs land on, well, stderr.

Yes, this is acceptable. Please move this change to a separate PR so it will be clearly visible in the release notes.

hoegaarden, 2023-06-11:
We didn't reach an agreement re. that, thus will close that now.

However, a [new plugin mechanism](https://github.com/kbialek/deye-inverter-mqtt/tree/b6f75d50a2e23daf5034d93c12c81c3f95d08051#custom-plugins) is on the horizon / being implemented.

For people who are looking to getting the data a json, there is https://github.com/hoegaarden/deye-inverter-mqtt-plugins/tree/main/plugins/stdout-publisher.

closed 2023-06-11 by hoegaarden