kbialek/deye-inverter-mqtt #98: Connect to Deye Sun 12K SG04LP3 EU  via LAN

develth, 2023-06-09:
**Hardware:**
 - Inverter model: `Deye SUN-12K-SG04LP3`

**Software:**
 - Logger firmware version: LAN-Logger `where to find?`
 - deye-inverter-mqtt version: `2023.05.4`
 - config.env:

```
LOG_LEVEL=DEBUG
DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_ups,deye_sg04lp3_battery
DEYE_LOGGER_SERIAL_NUMBER=2107051094
DEYE_LOGGER_IP_ADDRESS=192.168.16.243
DEYE_LOGGER_PORT=8899
MQTT_HOST=192.168.16.20
MQTT_USERNAME=deye-mqtt
MQTT_PASSWORD=****
```

Hi, 

getting following Errors on the Debugging Log in the LAN-Logger:

```
2023-06-09 15:51:32,752 - DeyeDaemon - INFO - Reading start
2023-06-09 15:51:32,753 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_ups, range: 0202-022e]
2023-06-09 15:51:32,769 - DeyeConnector - DEBUG - Request frame: a5170010450000560c977d02000000000000000000000000000001030202002d25afed15
2023-06-09 15:51:34,669 - paho.mqtt.client - DEBUG - Sending PINGREQ
2023-06-09 15:51:34,685 - paho.mqtt.client - DEBUG - Received PINGRESP
2023-06-09 15:51:42,782 - DeyeConnector - DEBUG - Connection timeout
2023-06-09 15:51:52,787 - DeyeConnector - DEBUG - Connection timeout
2023-06-09 15:51:56,530 - DeyeConnector - DEBUG - Response frame: 000800000003018302
2023-06-09 15:51:56,531 - DeyeModbus - ERROR - Response frame is too short
2023-06-09 15:51:56,531 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-06-09 15:51:56,531 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_battery, range: 024a-024f]
2023-06-09 15:51:56,540 - DeyeConnector - DEBUG - Request frame: a5170010450000560c977d0200000000000000000000000000000103024a0006e5a6c515
2023-06-09 15:51:56,577 - DeyeConnector - DEBUG - Response frame: 000800000003018302
2023-06-09 15:51:56,578 - DeyeModbus - ERROR - Response frame is too short
2023-06-09 15:51:56,578 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-06-09 15:51:56,578 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3, range: 0256-027c]
2023-06-09 15:51:56,585 - DeyeConnector - DEBUG - Request frame: a5170010450000560c977d020000000000000000000000000000010302560027e478c315
2023-06-09 15:51:57,811 - DeyeConnector - DEBUG - Response frame: a5170010450300d602
2023-06-09 15:51:57,812 - DeyeModbus - ERROR - Response frame is too short
2023-06-09 15:51:57,812 - DeyeModbus - ERROR - Modbus frame is too short or empty
2023-06-09 15:51:57,812 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3_ups, range: 0284-028d]
2023-06-09 15:51:57,819 - DeyeConnector - DEBUG - Request frame: a5170010450000560c977d02000000000000000000000000000001030284000a845c5815
```

Any idea how to fix?

Thanks!

kbialek, 2023-06-09:
For the start, please double check if you took the S/N of the LOGGER and **not** the S/N of the INVERTER

develth, 2023-06-09:
Yes, I did selected the S/N of the Logger

kbialek, 2023-06-09:
Interesting. What logger is it? Can you post a link?

develth, 2023-06-10:
Do not have an link, got it directly from the merchant. Looks like WIFI just with Ethernet

[image]

kbialek, 2023-06-10:
I found this project .https://github.com/MichaluxPL/Sofar_LSW3 The author claims that it also works with LSE3. You can give it a try. However essentially it is based on this same "parent" project and borrowed this same communication code as I did in this project. 

In the log that you have sent, the modbus responses are truncated to 10 bytes. Possible reasons I can think of are:
1. Bug in this project
2. Incompatible (newer?) protocol version used by the logger

You can and should try with the most recent version released yesterday because it contains improvements in the modbus communication part.

develth, 2023-06-10:
Thanks! Will test it later and provide feedback

develth, 2023-06-10:
> You can and should try with the most recent version released yesterday because it contains improvements in the modbus communication part.

Did docker stop && docker rm && docker pull. seems like same output:

```
2023-06-10 07:55:22,114 - DeyeModbus - ERROR - Response frame is too short
2023-06-10 07:55:22,115 - DeyeDaemon - INFO - Reading registers [metrics group: deye_sg04lp3, range: 02a0-02a7]
2023-06-10 07:55:22,126 - DeyeConnector - DEBUG - Request frame: a5170010450000560c977d020000000000000000000000000000010302a0000845966d15
2023-06-10 07:55:27,529 - DeyeConnector - DEBUG - Received response frame in 1. attempt: 004800000003018302
2023-06-10 07:55:27,530 - DeyeModbus - ERROR - Response frame is too short
2023-06-10 07:55:27,531 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2023-06-10 07:55:27,531 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m3), 'b'deye/logger_status'', ... (7 bytes)
2023-06-10 07:55:27,539 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 3)
2023-06-10 07:55:27,539 - DeyeMqttClient - INFO - Logger is offline
2023-06-10 07:55:27,539 - DeyeDaemon - INFO - Reading completed
```

develth, 2023-06-10:
Found also that:

https://github.com/pfostenberg/DeyeHI_LSW3_2Mqtt

But seems like an issue with register stuff. 

```
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00V\x0c\x97}\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00F\x00~$?\x00\x15')
Received data:  b'\x00:\x00\x00\x00\x03\x01\x83\x02'
No value in response for register 0x0046
Check register start/end values in config.cfg
```

What i also recognized is that if i go via Solarman App into Remote Control, everything is empty. Maybe a general issue

kbialek, 2023-06-11:
This response
```
DEBUG - Received response frame in 1. attempt: 004800000003018302
```
and this response
```
Received data:  b'\x00:\x00\x00\x00\x03\x01\x83\x02'
```

look very similar to each other, although they are logged by two from different clients.

Perhaps your logger uses different protocol version, or different modbus device address. Just guessing.
What is your logger firmware version? Maybe you can seek for help here https://github.com/dasrecht/deye-firmware

edvandreas, 2023-06-11:
I had the same problem with my LSW3. Due lack of time I used my wifi dongle.

Otherwise the LAN dongle worked fine with solarman and deye app.

@develth The empty Remote Control is a known problem. Write a mail to the deye support they can fix it for you.

develth, 2023-07-12:
as @edvandreas suggested, with WIFI Dongle no issues. So there is maybe a bug on the LAN Dongle.

closed 2023-08-27 by github-actions[bot]