kbialek/deye-inverter-mqtt #100: Huge delay between actual and reported measurements

palto42, 2023-06-10:
This is not an issue with this software but I just want to share an observation from my Deye SUN600G3-EU-230 firmware MW3_16U_5406_1.57

The measurements reported by the logger are significantly delayed if I compare them with the measurements from the official energy meter of my house (reports in 1 s intervals.

I did a test where I covered my 2 solar panels for 1 minute and I can quite exactly see this drop in the measurements of the house energy meter. But the Deye logger reports the drop 5 minutes later and the "downtime" is reported as 10 minutes instead of only 1 min and the power drop is much less (so the total energy could be correct).

The top graph is the Deye logger reporting and the bottom one shows the energy meter (power injected).

[image]

Don't think there is anything we can do from the software side, but I think it's important to be aware of this major inaccuracy and maybe you want to add a note/warning in the Readme.

Maybe other user can comment if this is typical on Deye or just for this model or maybe the firmware version.

kbialek, 2023-06-11:
Look here https://www.deyeinverter.com/product/accessory-monitoring-1/stick-logger.html. The default data acquisition interval is 5 minutes. I'm not sure how to interpret this. Is it the interval at which the logger reads from the inverter? Or is it the interval at which the logger reports to the cloud? In case of cloud connection, both options might be true.

palto42, 2023-06-11:
Good point, looking at the data again I can confirm that the reported values only seem to change every 5 min.
Since the reported data is not just delayed but the power change is also much less and much longer, it seems to report the average power of the past period. 
In my test the actual power change was ~500 W for 1 minute, the reported power change was ~90 W for two 5 min intervals.

palto42, 2023-06-11:
Seems that the reporting to cloud and modbus are separate, see https://github.com/Hypfer/deye-microinverter-cloud-free#increasing-the-data-logging-interval

palto42, 2023-07-08:
Closing here since this is an issue of the Deye logger.

closed 2023-07-08 by palto42