kbialek/deye-inverter-mqtt #99: Configurable connection timeout and number of attempts

palto42, 2023-06-10:
The current `connection.send_request()` function uses a hard coded timeout and fixed number of retries for the response receive part and none for the connect itself or the message send.

I suggest to make the timeout and number of retries configurable, so that users can tune it based on their needs (e.g. WiFi quality, logger model, read interval).

The connection attempt and send message should also be included in the retry loop which would allow to simplify the exception handling and make this code part more consistent. My proposal is to use the socket connection context handler in the retry loop so that in case of some issue (connect, send or receive) the whole connection will be reset and retried.

If such change/improvement is accepted for inclusion, I can raise a PR with an implementation which I already tested.

For my use case with 30 seconds read interval and some WiFi stability issues I found that for example a timeout of 5 seconds and 4 retries works good.

kbialek, 2023-06-11:
Nobody complained about that so far, so probably the existing connection retry logic works pretty well. I park this proposal until the problem materializes.

palto42, 2023-06-11:
In case anyone is interested, this is my branch for  the configurable timeout and retry.
https://github.com/palto42/deye-inverter-mqtt/tree/conf_connect_attempts

palto42, 2023-07-08:
Closing since there seems no real demand.

closed 2023-07-08 by palto42