kbialek/deye-inverter-mqtt #104: Deye SUN-20K-SG01HP3-AM2 Support

Verkalets, 2023-07-06:
- Inverter model: Deye SUN-20K-SG01HP3-AM2

Hi, I'd like to check if Deye SUN-20K-SG01HP3-AM2 will be supported? Is the any chance to do it before installation?

kbialek, 2023-07-06:
Hi! So far no one reported this inverter as compatible, so I can only make an educated guess here. Imho, most likely it will not work out of the box. This is a 3 phase hybrid inverter from SG01 product line. As for now, this tool supports SG01LP1 1 phase hybrid inverters (deye_hybrid) and SG04LP3 3-phase inverters (deye_sg04lp3).

Verkalets, 2023-07-07:
@kbialek Thank you for explanation!

closed 2023-07-07 by Verkalets

kbialek, 2023-07-15:
@Verkalets Perhaps it's something interesting for you https://github.com/kbialek/deye-inverter-mqtt/issues/41#issuecomment-1636741106

kbialek, 2023-07-17:
@Verkalets Please read this thread #110