kbialek/deye-inverter-mqtt #106: Alternative modbus CRC calculation

m-kress, 2023-07-11:
A word first: This is just a suggestion to make the whole installation of the program more slim. You can immediatelly close this again.

The idea is to use another method to calculate the CRC for Modbus. Currently you use libscrc, which has a lot of other CRCs, which are actually not necessary for this project. It also contains C code, which always is trouble for cross compiling to other architectures. I guess, the higher efficiency of natively compiled code is not needed, when twice a minute a CRC must be calculated.

Have a look at modbus-crc: https://pypi.org/project/modbus-crc
Actually it consists of only two tables and single function. 

I took the liberty to copy/paste the necessary parts into deye_modbus.py. Here is a compressed archive of this file, because gitlab would not allow a python file attached here:
[deye_modbus.tar.gz](https://github.com/kbialek/deye-inverter-mqtt/files/12014777/deye_modbus.tar.gz)

If you consider this, please also respect the license BSD:
https://github.com/webtoucher/modbus-crc/blob/master/LICENSE.md

kbialek, 2023-07-11:
Thanks for the proposal. Perhaps it's worth considering. Could you come up with a real life problem that this change would solve? You've mentioned cross-compilation might not work. This project already supports four architectures. Is there another architecture that you'd like to add? Or maybe the process will use less memory?

> Here is a compressed archive of this file, because gitlab would not allow a python file attached here:
> [deye_modbus.tar.gz](https://github.com/kbialek/deye-inverter-mqtt/files/12014777/deye_modbus.tar.gz)

Please open a PR, so we can discuss it.

> If you consider this, please also respect the license BSD

Afaik, BSD license is very liberal and is also compatible with Apache 2.0, so it shouldn't be a problem. Project's source code is publicly available.

kbialek, 2023-07-11:
> I'm running this in an LXC container on an
embedded industrial router,

Cool setup. I like it! Kudos!

Here it's explained how to open a PR: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

m-kress, 2023-07-12:
Ah - Pull Request! Will do.

closed 2023-08-26 by github-actions[bot]