kbialek/deye-inverter-mqtt #107: Installation for a newbie :-)

svenv0, 2023-07-11:
Hallo,

how and where do i have to save the file config.env to start the installation ?
Now is unpossible to start the docker container:

[image]

thx for helping,greeting Sven

closed 2023-07-11 by kbialek

JeanDechant, 2024-07-29:
It actually is very muchn not answered there. Specifically for "newbies". 
How to install the MQTT Broker? No instructions at all.
How to use the commands? Not really explained.
How to set up anything? Not very beginner friendly to say the least. 

Look, im not complaining about it, i am struggling with this myself but thats because i am using something that isnt meant for noobs. But telling someone who very much would love to use this feature to maybe set their 600W profile to a 800W profile (german regulations changed and this device got sold there a lot), "look on the main page, its explained there" is not cool. It would have been better to say "Look, this isnt really meant for beginners and explaing all of it to someone who has almost no idea about these things is too time consuming".

kbialek, 2024-08-01:
@JeanDechant I fully agree, that the entry level can be pretty high for some people, and that the documentation might be hard to understand. On the other hand, I'm not gonna teach people how to install MQTT broker. This is not my goal and is beyond the scope of this project.

This project is open sourced for reasons. One of them is to let people contribute to the project, so if you are willing to write an installation tutorial, then let me know and I will happily link it in the README.