kbialek/deye-inverter-mqtt #110: Deye 30kW SUN-30K-SG01HP3-EU-BM3

fakehec, 2023-07-15:
Tested with Deye 30kW SUN-30K-SG01HP3-EU-BM3 hybrid with battery
Works ok everything except data from PV3

Please implement PV3 line data in deye_sg04lp3

kbialek, 2023-07-15:
@fakehec For that, I'd need to know respective modbus register addresses. Basing on the existing mapping I suspect that this mapping could work:
* 674 - PV3 Power
* 680 - PV3 Voltage
* 681 - PV3 Current

Please confirm if they return correct values for you using this command. Voltage and Current might be shown as multiplied by 10.

`docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r <reg_address>`

fakehec, 2023-07-15:
Thanks @kbialek 

docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 674
int: 310, l: 54, h: 1

docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 680
int: 5231, l: 111, h: 20

docker run --rm --env-file config.env ghcr.io/kbialek/deye-inverter-mqtt r 681
int: 59, l: 59, h: 0

At a moment later, these are the values reported by MQTT

dc/pv1/power 535.0
dc/pv2/power 1.0
dc/pv1/voltage 419.8
dc/pv2/voltage 418.4
dc/pv1/current 12.7
dc/pv2/current 0.0

Let me know any other test i can perform.
Thanks in advance

fakehec, 2023-07-15:
The values observed on the LCD screen of the inverter are similar with the values reported by the addresses 674, 680, 681, (V and I multiplied by 10) as you guessed :-)

fakehec, 2023-07-15:
Also as a side note, the values reported by 672,673 and 674 for powers PV1, PV2, PV3, are in unit W, but divided by 10. This has been checked with the data reported by the inverter LCD.

kbialek, 2023-07-15:
> the values reported by 672,673 and 674 for powers PV1, PV2, PV3, are in unit W, but divided by 10.

So either:
1. There is a bug in the register mapping for sg04lp3
2. sg01 requires different mapping than sg04

@edvandreas You own sg04lp3. Are PV1 and PV2 power metrics reported correctly for you?

edvandreas, 2023-07-15:
> @edvandreas You own sg04lp3. Are PV1 and PV2 power metrics reported correctly for you?

Yes they are correct.

kbialek, 2023-07-16:
> > @edvandreas You own sg04lp3. Are PV1 and PV2 power metrics reported correctly for you?
> 
> Yes they are correct.

So it looks like sg01hp3 is not fully compatible with sg04lp3 mapping.

@fakehec Could you please take the effort and compare all reported metrics with the values shown in the mobile app, and note down all the differences? This information will help to create a mapping for sg01hp3. btw, the values in the app might be slightly different. This is normal, because they are sampled at different point in time.

fakehec, 2023-07-16:
These are the differences comparing the app with the data reported by MQTT
dc/pv1/power: 348 (app reported 3485W) therefore /10
dc/pv2/power: 289 (app reported 2873W) therefore /10

battery/power -743.0 (app reported -7.53kW) therefore /10
battery/voltage 21.6 (app reported 215.7V) therefore /10

I am using DEYE_METRIC_GROUPS=deye_sg04lp3,deye_sg04lp3_battery,deye_sg04lp3_ups,settings
The rest of the values are reported ok

My setup is Deye 30kW SUN-30K-SG01HP3-EU-BM3, with 20kW solar panels balanced between 3 of the 4 PV inputs (the inverter has PV1, PV2,PV3,PV4), and 20kWh battery in one the battery inputs (the inverter has 2 battery inputs). Generator  line is also not used right now. 

I checked registers 675, 682 and 683 and they report 0 value, so i suspect they are the ones for PV4, as follows

675 - PV4 Power (/10)
682 - PV4 Voltage (*10)
683 - PV4 Current (*10)

kbialek, 2023-07-16:
@fakehec Thanks. I've added new metrics groups for sg01hp3. 

You can test them using this beta docker image `ghcr.io/kbialek/deye-inverter-mqtt:2023.07.3-beta-1`. 
You have to update your config and start using these metric groups now `deye_sg01hp3,deye_sg01hp3_battery,deye_sg041h3_ups`. 

Updated documentation is available here https://github.com/kbialek/deye-inverter-mqtt/blob/feature/support-sg01hp3/README.md

fakehec, 2023-07-16:
Done, it looks good so far :-)

https://snapshots.raintank.io/dashboard/snapshot/IG6Im0YUFmklrK7vDABtBl5J6vlOycaP
(exp 1 day)
Thanks a lot!