kbialek/deye-inverter-mqtt #115: No zeros when offline

tompsg-git, 2023-07-22:
Is it possible not to send zeros on day & total energy, when inverter is offline.

kbialek, 2023-07-24:
@tompsg-git The service does not send any metrics when the inverter is offline.

tompsg-git, 2023-07-24:
I thought the service just kept sending. Do you have an idea how to suppress the effect (see log) here.

2023-07-24_21:07:55 MQTT2_deye_inverter_4135644 voltage: 236.0
2023-07-24_21:07:55 MQTT2_deye_inverter_4135644 voltage1: 18.1
2023-07-24_21:07:55 MQTT2_deye_inverter_4135644 active_power: 1.0
2023-07-24_21:07:55 MQTT2_deye_inverter_4135644 radiator_temp: 19.6
2023-07-24_21:10:03 MQTT2_deye_inverter_4135644 logger_status: offline
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 logger_status: online
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 day_energy: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 total_energy: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 voltage: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 freq: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 voltage1: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 day_energy1: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 total_energy1: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 active_power: 0.0
2023-07-24_21:10:53 MQTT2_deye_inverter_4135644 radiator_temp: -10.0
2023-07-24_21:14:03 MQTT2_deye_inverter_4135644 logger_status: offline

kbialek, 2023-07-25:
Looks like your inverter is swinging between online and offline status. Online in this context means, reachable over network. When the inverter is online, and the service successfully reads values from it, then they are sent over mqtt. That's it.
Status swinging can happen late in the evening or early in the morning, when the inverter is switching on and off depending on the amount of sunlight. No idea, why day_energy and total_energy were zeroed for you.

The logs that you've posted does not seem to be produced by the service, btw.

closed 2023-09-08 by github-actions[bot]