kbialek/deye-inverter-mqtt #117: Extend README.md with a short note about manual HA integration

CarstenGrohmann, 2023-07-30:
Hi,

please add a short note to the [README.md](https://github.com/kbialek/deye-inverter-mqtt/blob/main/README.md) file that the integration into [Home Assistant](https://www.home-assistant.io/) has to be done manually.

This helps to compare the project to others.

Thank you,
Carsten

kbialek, 2023-07-30:
hi @CarstenGrohmann 

I don't use HA and have not enough knowledge to document the integration. Feel free to fork this repo, extend the README as you wish, and open a PR. I will be happy to accept it.

marcelloceschia, 2023-08-06:
@CarstenGrohmann We can do this together, coz I'm planning to integrate this within the next days

marcelloceschia, 2023-08-06:
I think a better idea is to write a plugin that writes the configuration to auto discovery https://www.home-assistant.io/integrations/mqtt/#discovery-examples

CarstenGrohmann, 2023-08-06:
@marcelloceschia I'm sorry, but unfortunately I don't have the time to contribute.

I was just thinking of a new section in the readme to make it clearer that HA currently can't automatically utilize the data. 
The thought solution with a plugin sounds good.

Proposal for the readme:

--------
## Home Assistant integration

This project currently has no integration with [Home Assistant](https://www.home-assistant.io/). You can manually configure HA to use this data,
or you can write a plugin as described earlier in this document.

For help with integration, see
* [MQTT Client Component](https://esphome.io/components/mqtt.html)
* [MQTT Discovery examples](https://www.home-assistant.io/integrations/mqtt/#discovery-examples)

--------

But we can close this ticket as you'll write a HA integration plugin.

marcelloceschia, 2023-08-06:
@CarstenGrohmann Your point is correct, we should add this notice. I wrote the plugin that can configure the sensors automatically - would be nice if you can check this too.

kbialek, 2023-08-06:
@CarstenGrohmann @marcelloceschia Out of my curiosity. Why do you prefer to use this project over https://github.com/StephanJoubert/home_assistant_solarman ?

marcelloceschia, 2023-08-06:
Because the inverter and the home assistant are not in the same network. And I prefer this abstraction, separate collection and analyze/report

CarstenGrohmann, 2023-08-06:
> @CarstenGrohmann Your point is correct, we should add this notice. I wrote the plugin that can configure the sensors automatically - would be nice if you can check this too.

I have created PR #120. You can update the section when the plugin is ready. I can test the plugin and do a small code review if you want.

CarstenGrohmann, 2023-08-06:
> @CarstenGrohmann @marcelloceschia Out of my curiosity. Why do you prefer to use this project over https://github.com/StephanJoubert/home_assistant_solarman ?

Later in my project, I want to use the inverter data to control the electric power that charges or is fed from a battery. Therefore, I think MQTT as data broker would be very helpful.

UndIch, 2023-08-20:
After SOLARMAN is not working in my setup anymore (and I can’t figure out why…) I looked into integration of this one into my Homeassitant - but like mentioned there is ok documentation at all and I have no clue where to start. I have a mqtt broker (Moskito) running in HA but I can not figure out where to add all this config files mentioned in the readme… to bad … maybe someone will make the efforts and write some how-to for us low level users ;).

PR #120 (2023-08-20): Add a small section about integration with the Home Assistant
Closes #117
closed 2023-08-20 by kbialek