kbialek/deye-inverter-mqtt #118: Logger Serial Number does not match (Deye SUN-M80G3-EU-Q0)

BasGo, 2023-07-30:
**Describe the bug**
Last week I got my PV with a Deye SUN-M80-G3 and tried to get all relevant ModBus values to MQTT. In the config I entered the serial number of the inverter (2305xxxxxx) and also tried the serial of the device itself (3944xxxxxx). Both serial numbers don't work and I get an error code 6 ("Logger Serial Number does not match. Check your configuration file.").

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Expected that this type of inverter works (as posted by someone else in #41.

**Hardware (please complete the following information):**
 - Inverter model: Deye SUN-M80G3-EU-Q0

**Software (please complete the following information):**
 - Logger firmware version: V1.57
 - deye-inverter-mqtt version: latest (also tried out with local debugging)
 - config.env (remove passwords before pasting):
 - 
```
DEYE_LOGGER_IP_ADDRESS=192.168.1.248
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=2305xxxxxx

MQTT_HOST=192.168.1.200
MQTT_PORT=1883
MQTT_USERNAME=xxxx
MQTT_PASSWORD=xxxx
MQTT_TOPIC_PREFIX=deye
MQTT_TLS_ENABLED=false

LOG_LEVEL=DEBUG
LOG_STREAM=STDOUT
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=micro,settings
```

kbialek, 2023-07-30:
hi @BasGo 

What kind of logger do you have? LAN or wireless? It's been reported that Modbus/TCP does not work for LAN loggers.

BasGo, 2023-07-30:
I use the built in logger which is accessible wireless. The inverter/logger uses a wireless connection to my home network and has a fixed IP address.

BasGo, 2023-07-31:
Seemed to be a random issue: after two days of unsuccessful testing, it worked like a charm this morning. Without changing anything. Thanks for your assistance.

closed 2023-07-31 by BasGo