kbialek/deye-inverter-mqtt #123: New firmware not work

Joerglp, 2023-09-07:
**Hardware (please complete the following information):**
 - Inverter model:SUN-600G3-EU. 

**Software (please complete the following information):**
 - Logger firmware version: MW3_16U_5406_2.06
 - deye-inverter-mqtt version:latest
 - config.env (remove passwords before pasting):
LOG_LEVEL=INFO
LOG_STREAM=STDOUT
DEYE_DATA_READ_INTERVAL=60
DEYE_METRIC_GROUPS=micro
DEYE_LOGGER_SERIAL_NUMBER=4196253664
DEYE_LOGGER_IP_ADDRESS=192.168.0.71
DEYE_LOGGER_PORT=8899
DEYE_FEATURE_MQTT_PUBLISHER=true
DEYE_FEATURE_SET_TIME=true
MQTT_HOST=192.168.0.101
MQTT_PORT=1883
MQTT_USERNAME=mqttuser
MQTT_PASSWORD=#######ä
MQTT_TOPIC_PREFIX=deye
PATH=/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
LANG=C.UTF-8
GPG_KEY=A035C8C19219BA821ECEA86B64E628F8D684696D
PYTHON_VERSION=3.10.10
PYTHON_PIP_VERSION=22.3.1
PYTHON_SETUPTOOLS_VERSION=65.5.1
PYTHON_GET_PIP_URL=https://github.com/pypa/get-pip/raw/d5cb0afaf23b8520f1bbcfed521017b4a95f5c01/public/get-pip.py
PYTHON_GET_PIP_SHA256=394be00f13fa1b9aaa47e911bdb59a09c3b2986472130f30aa0bfaf7f3980637

Error:
2023-09-07 15:33:49,329 - DeyeDaemon - INFO - Reading start
2023-09-07 15:33:49,329 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-07 15:33:49,469 - DeyeModbus - ERROR - Response frame has invalid starting byte
2023-09-07 15:33:49,470 - DeyeMqttClient - INFO - Logger is offline

With Firmware 1.57 ist o.k.
Now with the new Firmware are not o.k.

kbialek, 2023-09-07:
Thanks for reporting the problem @Joerglp . It seems like something has changed in version 2 of the firmware. Could you enable DEBUG logs, and try again, so we can see what is returned by the logger?

Joerglp, 2023-09-08:
2023-09-08 04:56:49,420 - DeyeDaemon - INFO - Reading start2023-09-08 04:56:49,421 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]2023-09-08 04:56:51,237 - DeyeConnector - DEBUG - Request frame: a5170010450000e0bf1dfa0200000000000000000000000000000103003c003945d4b6152023-09-08 04:56:51,748 - paho.mqtt.client - DEBUG - Sending PINGREQ2023-09-08 04:56:51,749 - paho.mqtt.client - DEBUG - Received PINGRESP2023-09-08 04:56:52,033 - DeyeConnector - DEBUG - Received response frame in 1. attempt: 41542b595a434d505645523d4d57335f3136555f353430365f322e30360d0a0d0a2023-09-08 04:56:52,033 - DeyeModbus - ERROR - Response frame has invalid starting byte2023-09-08 04:56:52,034 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'2023-09-08 04:56:52,034 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m19), 'b'deye/logger_status'', ... (7 bytes)2023-09-08 04:56:52,035 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 19)2023-09-08 04:56:52,036 - DeyeMqttClient - INFO - Logger is offline2023-09-08 04:56:52,037 - DeyeDaemon - INFO - Reading completedVon meinem iPad gesendet

kbialek, 2023-09-08:
The response "41542b595a434d505645523d4d57335f3136555f353430365f322e30360d0a0d0a" is a hex encoded ascii string. After conversion it looks like this "AT+YZCMPVER=MW3_16U_5406_2.06", so it's a response from the AT command interface. 

Possible options:
1. Someone figures out how to do Modbus/TCP communication with loggers having fw v2.x
2. Someone implements AT command support as a replacement of Modbus/TCP. Something similar to https://github.com/jedie/inverter-connect @jedie Do you know if it's feasible and could solve the problem?

SMCerina, 2023-09-08:
I cannot see any option to enable Modbus/TCP even in the config_hide.html.

kbialek, 2023-09-08:
Okay, perhaps Modbus/TCP interface has been moved to another TCP port. Can you scan your inverter IP with `nmap`?

SMCerina, 2023-09-08:
No other open port on the inverter.

```
80/tcp   open  http       LimitlessLED smart lightbulb bridge httpd
|_http-server-header: HTTPD
| http-methods: 
|_  Supported Methods: GET POST
|_http-title: 401 Unauthorized
| http-auth: 
| HTTP/1.0 401 Unauthorized\x0D
|_  Basic realm=USER LOGIN
8899/tcp open  ospf-lite?
| fingerprint-strings: 
|   GenericLines: 
|     AT+YZCMPVER=MW3_16U_5406_2.27
|     2211101306
|     2211101306
|     2211101306
|   GetRequest, HTTPOptions, NULL: 
|     AT+YZCMPVER=MW3_16U_5406_2.27
|     2211101306
|   RTSPRequest: 
|     AT+YZCMPVER=MW3_16U_5406_2.27
|     2211101306
|_    2211101306
```

Joerglp, 2023-09-08:
I can see the Entry Server and Port 8899 in the config_hide.html.
I have the Firmware 2.06

Joerglp, 2023-09-08:
Per Telnet:
telnet 192.168.0.71 8899
Connected to 192.168.0.71
AT+YZCMPVER=MW3_16U_5406_2.06

��B��X#�&2211236711
�                  p
����B��Y��&2211236711
�                    p
����B�&2211236711
�                p
����B��X#^C

carstenfroehlich, 2023-09-09:
The big problem is, that the new SUN-MI-RELAY-01 is automatically updating the inverter to firmware version 2.27. So most of the users will have this problem in the near future.
Like me since 5 days...

carstenfroehlich, 2023-09-09:
Maybe these settings are helping to analyse? 
[image]
[image]

kbialek, 2023-09-09:
@Joerglp @SMCerina Could you please check if the AT interface works with the new firmware as before? If yes, then there is a chance to use for Modbus-over-AT communication.
For that you can use this project https://github.com/s10l/deye-logger-at-cmd

Here are the results I got from my logger running firmware v1
1. Connectivity test:
```
❯ ./build/main -t 192.168.2.90:48899
2023/09/09 08:46:00 * Connecting :0 -> 192.168.2.90:48899...
2023/09/09 08:46:08 AP settings
2023/09/09 08:46:08 	Mode, SSID and Chanel:  11BGN,AP_XXXXXXXXX,AUTO
2023/09/09 08:46:08 	Encryption:             WPA2PSK,AES,712003b0
2023/09/09 08:46:08 Station settings
2023/09/09 08:46:08 	SSID:                   XXXXXXXXX
2023/09/09 08:46:08 	Key:                    WPA2PSK,AES,XXXXXXXXXX
2023/09/09 08:46:08 	IP:                     DHCP,192.168.2.90,255.255.255.0,192.168.2.1
2023/09/09 08:46:08 Web settings
2023/09/09 08:46:08 	Login:                  admin,admin
2023/09/09 08:46:09 
```
2. Modbus read test:
```
❯ ./build/main -t 192.168.2.90:48899 -xmb 00280001
2023/09/09 09:02:21 * Connecting :0 -> 192.168.2.90:48899...
2023/09/09 09:02:24 +ok=01030204423AB50000
2023/09/09 09:02:25 
```

kbialek, 2023-09-09:
@carstenfroehlich What are the options available here?
[image]

carstenfroehlich, 2023-09-09:
TCP-Server
TCP-Client
UDP

kbialek, 2023-09-09:
> TCP-Server TCP-Client UDP

Exactly the same options as for my logger running on firmware v1

carstenfroehlich, 2023-09-09:
maybe the options of the select box have changed?
`
<select name="net_setting_pro_sel" id="div_5_3_tcp" style="width:250px" onclick="internetSet()">
                    <option value="TCPSERVER">TCP-Server</option>
                    <option value="TCPCLIENT">TCP-Client</option>
                    <option value="UDP">UDP</option></select>
`

Joerglp, 2023-09-09:
root@server01:~/deye-logger-at-cmd/build# ./main -t 192.168.0.71:48899
2023/09/09 11:22:20 * Connecting :0 -> 192.168.0.71:48899...
2023/09/09 11:22:28 AP settings
2023/09/09 11:22:28     Mode, SSID and Chanel:  11BGN,AP_4196253664,AUTO
2023/09/09 11:22:28     Encryption:             WPA2PSK,AES,847a4d1c
2023/09/09 11:22:28 Station settings
2023/09/09 11:22:28     SSID:                   Home
2023/09/09 11:22:28     Key:                    WPA2PSK,AES,9026905867692283
2023/09/09 11:22:28     IP:                     static,192.168.0.71,255.255.255.0,192.168.0.254
2023/09/09 11:22:28 Web settings
2023/09/09 11:22:28     Login:                  admin,admin
2023/09/09 11:22:29 
root@server01:~/deye-logger-at-cmd/build# ./main -t 192.168.0.71:48899  -xmb 00280001
2023/09/09 11:23:01 * Connecting :0 -> 192.168.0.71:48899...
2023/09/09 11:23:04 +ok=0103020064B9AF
2023/09/09 11:23:05

kbialek, 2023-09-09:
@Joerglp Thanks. This looks very promising.

carstenfroehlich, 2023-09-09:
this is from my server:

go run main.go -t 192.168.178.108:48899
2023/09/09 12:43:09 * Connecting :0 -> 192.168.178.108:48899...
2023/09/09 12:43:17 AP settings
2023/09/09 12:43:17     Mode, SSID and Chanel:  11BGN,AP_XXXXXXX,AUTO
2023/09/09 12:43:17     Encryption:             WPA2PSK,AES,XXXXXX
2023/09/09 12:43:17 Station settings
2023/09/09 12:43:17     SSID:                   XXXXXXXXXXX
2023/09/09 12:43:17     Key:                    WPA2PSK,AES,XXXXXXXX
2023/09/09 12:43:17     IP:                     DHCP,192.168.178.108,255.255.255.0,192.168.178.1
2023/09/09 12:43:17 Web settings
2023/09/09 12:43:17     Login:                  XXXXXXX,XXXXXXXX
2023/09/09 12:43:18

go run main.go -t 192.168.178.108:48899 -xmb 00280001
2023/09/09 12:46:08 * Connecting :0 -> 192.168.178.108:48899...
2023/09/09 12:46:11 +ok=0103020064B9AF
2023/09/09 12:46:12

carstenfroehlich, 2023-09-09:
does this help?
`
2023-09-09 12:31:28,362 - DeyeDaemon - INFO - Reading start
2023-09-09 12:31:28,363 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-09 12:31:28,371 - DeyeModbus - ERROR - Response frame has invalid starting byte
2023-09-09 12:31:28,375 - DeyeMqttClient - INFO - Logger is offline
2023-09-09 12:31:28,375 - DeyeDaemon - INFO - Reading completed
`

kbialek, 2023-09-09:
Your tests proven, that it's possible to use Modbus-over-AT to read the data. I've started working on that feature. Stay tuned.

maehtricks, 2023-09-09:
It actually works with https://github.com/jedie/inverter-connect via Modbus over AT

`./cli.py print-values --ip 192.168.71.178 --port 48899 -c`

[image]

kbialek, 2023-09-09:
Here is a beta image that uses AT commands for communication with the inverter exclusively. `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-1` Try it out.

The implementation is very hacky now. Primarily error handling is very poor. However it's ready to start collecting feedback. 

No extra config changes are necessary.

Joerglp, 2023-09-09:
Now I became that:
2023-09-09 15:43:08,215 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2023-09-09 15:43:08,215 - DeyeAtConnector - INFO - Hello
2023-09-09 15:43:08,259 - paho.mqtt.client - DEBUG - Sending CONNECT (u1, p1, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-4196253664'
2023-09-09 15:43:08,261 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-09-09 15:43:08,262 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2023-09-09 15:43:08,262 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.0.101:1883
2023-09-09 15:43:08,263 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
2023-09-09 15:43:08,263 - DeyeDaemon - INFO - Feature "Set inverter time once online": enabled
2023-09-09 15:43:08,263 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 60 seconds
2023-09-09 15:43:08,264 - DeyeDaemon - INFO - Reading start
2023-09-09 15:43:08,264 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-09 15:43:08,265 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2023-09-09 15:43:08,265 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2023-09-09 15:43:08,766 - DeyeAtConnector - DEBUG - Received response frame in 1. attempt: b'192.168.0.71,E8FDF88E20F6,4196253664'
2023-09-09 15:43:08,766 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2023-09-09 15:43:09,267 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'
2023-09-09 15:43:09,768 - DeyeAtConnector - DEBUG - Received response frame in 1. attempt: b'+ok=01\x1003\x1072\x1000\x101D\x1000\x1000\x1000\x1000\x100F\x10F2\x1000\x1000\x1000\x100E\x1000\x100E\x1000\x1000\x1000\x1000\x1007\x10BD\x1000\x1000\x1007\x10D1\x1000\x1000\x1009\x1038\x1000\x1000\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1000\x1013\x1088\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1001\x1068\x1000\x1000\x1000\x1000\x1000\x1000\x1012\x10E8\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10EA\x1000\x1007\x1000\x10FA\x1000\x1007\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1058\x10AB\x10\r\n\r\n'
2023-09-09 15:43:09,768 - DeyeAtConnector - DEBUG - Extracted Modbus response 010372001D000000000FF20000000E000E0000000007BD000007D100000938000000000001000000001388000000000000000000000000016800000000000012E800000000000000000000000000000000000000000000000000000000000000000000000000EA000700FA00070000000000000000
2023-09-09 15:43:09,768 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2023-09-09 15:43:10,269 - DeyeModbus - ERROR - Modbus frame is too short
2023-09-09 15:43:10,270 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2023-09-09 15:43:10,270 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'deye/logger_status'', ... (7 bytes)
2023-09-09 15:43:10,271 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
2023-09-09 15:43:10,272 - DeyeMqttClient - INFO - Logger is offline
2023-09-09 15:43:10,272 - DeyeDaemon - INFO - Reading completed
2023-09-09 15:44:08,344 - paho.mqtt.client - DEBUG - Sending PINGREQ
2023-09-09 15:44:08,375 - paho.mqtt.client - DEBUG - Received PINGRESP
2023-09-09 15:44:10,274 - DeyeDaemon - INFO - Reading start
2023-09-09 15:44:10,278 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-09 15:44:10,279 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2023-09-09 15:44:10,781 - DeyeAtConnector - DEBUG - Received response frame in 1. attempt: b'192.168.0.71,E8FDF88E20F6,4196253664'
2023-09-09 15:44:10,781 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2023-09-09 15:44:11,285 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'
2023-09-09 15:44:11,786 - DeyeAtConnector - DEBUG - Received response frame in 1. attempt: b'+ok=01\x1003\x1072\x1000\x101D\x1000\x1000\x1000\x1000\x100F\x10F2\x1000\x1000\x1000\x100E\x1000\x100E\x1000\x1000\x1000\x1000\x1007\x10BD\x1000\x1000\x1007\x10D1\x1000\x1000\x1009\x1038\x1000\x1000\x1000\x1000\x1000\x1001\x1000\x1000\x1000\x1000\x1013\x1088\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1001\x1068\x1000\x1000\x1000\x1000\x1000\x1000\x1012\x10E8\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x10EA\x1000\x1007\x1000\x10FA\x1000\x1007\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1000\x1058\x10AB\x10\r\n\r\n'
2023-09-09 15:44:11,787 - DeyeAtConnector - DEBUG - Extracted Modbus response 010372001D000000000FF20000000E000E0000000007BD000007D100000938000000000001000000001388000000000000000000000000016800000000000012E800000000000000000000000000000000000000000000000000000000000000000000000000EA000700FA00070000000000000000
2023-09-09 15:44:11,787 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+Q\n'
2023-09-09 15:44:12,290 - DeyeModbus - ERROR - Modbus frame is too short
2023-09-09 15:44:12,292 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
[... cut ...]
2023-09-09 15:46:12,440 - DeyeModbus - ERROR - Modbus frame is too short
2023-09-09 15:46:12,443 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2023-09-09 15:46:12,443 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m4), 'b'deye/logger_status'', ... (7 bytes)
2023-09-09 15:46:12,445 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 4)
2023-09-09 15:46:12,445 - DeyeMqttClient - INFO - Logger is offline
2023-09-09 15:46:12,445 - DeyeDaemon - INFO - Reading completed

Joerglp, 2023-09-09:
With the tool from @maehtricks  works
inverter v0.13.0
                                                                                                            (Set log level 0: ERROR)

InverterInfo(ip='192.168.0.71', mac='E8FDF88E20F6', serial=4196253664)

Fetch...PV1 Voltage,PV2 Voltage,PV1 Current,PV2 Current,Daily Production,Daily Production 1,Daily Production 2,Total Production,Total Production 1,Total Production 2,Active Power Regulations,PV1 Power,PV2 Power,Total Power,
Signing off with "AT+Q"...Goodbye ;)

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
                                     Inverter Values                                      
┏━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━━━━┓
┃ Counter ┃                     Name ┃ Value            ┃ Register ┃ Length ┃   Raw data ┃
┡━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━╇━━━━━━━━━━━━┩
│       1 │              PV1 Voltage │         23.4 V   │   006D   │   1    │       00ea │
│       2 │              PV2 Voltage │         24.6 V   │   006F   │   1    │       00f6 │
│       3 │              PV1 Current │          0.4 A   │   006E   │   1    │       0004 │
│       4 │              PV2 Current │          0.4 A   │   0070   │   1    │       0004 │
│       5 │         Daily Production │          2.9 kWh │   003C   │   1    │       001d │
│       6 │       Daily Production 1 │          1.4 kWh │   0041   │   1    │       000e │
│       7 │       Daily Production 2 │          1.4 kWh │   0042   │   1    │       000e │
│       8 │         Total Production │        408.2 kWh │   003F   │   2    │   0ff20000 │
│       9 │       Total Production 1 │        198.1 kWh │   0045   │   1    │       07bd │
│      10 │       Total Production 2 │        200.1 kWh │   0047   │   1    │       07d1 │
│      11 │ Active Power Regulations │          100 %   │   0028   │   1    │       0064 │
│      12 │                PV1 Power │         9.36 W   │    -     │   -    │ (Computed) │
│      13 │                PV2 Power │         9.84 W   │    -     │   -    │ (Computed) │
│      14 │              Total Power │         19.2 W   │    -     │   -    │ (Computed) │
└─────────┴──────────────────────────┴──────────────────┴──────────┴────────┴────────────┘
───────────────────────────────────────────────────── Sa 09 Sep 2023 18:06:33  ─────────────────────────────────────────────────────

carstenfroehlich, 2023-09-09:
`2023-09-09 16:27:13,632 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2023-09-09 16:27:13,666 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3939493096'
2023-09-09 16:27:13,668 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-09-09 16:27:14,669 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2023-09-09 16:27:14,670 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.178.113:1883
2023-09-09 16:27:14,671 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
2023-09-09 16:27:14,671 - DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
2023-09-09 16:27:14,672 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 30 seconds
2023-09-09 16:27:14,674 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2023-09-09 16:27:14,675 - DeyeDaemon - INFO - Reading start
2023-09-09 16:27:14,677 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-09 16:27:14,681 - DeyeConnector - DEBUG - Request frame: a5170010450000e8e4cfea0200000000000000000000000000000103003c003945d48515
2023-09-09 16:27:14,686 - DeyeConnector - DEBUG - Received response frame in 1. attempt: 41542b595a434d505645523d4d57335f3136555f353430365f322e32370d0a0d0a
2023-09-09 16:27:14,687 - DeyeModbus - ERROR - Response frame has invalid starting byte
2023-09-09 16:27:14,688 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2023-09-09 16:27:14,690 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'deye/logger_status'', ... (26 bytes)
2023-09-09 16:27:14,692 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
2023-09-09 16:27:14,694 - DeyeMqttClient - INFO - Logger is offline
2023-09-09 16:27:14,694 - DeyeDaemon - INFO - Reading completed`

kbialek, 2023-09-09:
Thanks for the logs. They are very helpful. I see where the problem is. My logger appends trailing zeros to the Modbus response. You can see it here https://github.com/kbialek/deye-inverter-mqtt/issues/123#issuecomment-1712438301 I'm talking about this "+ok=01030204423AB5**0000**"
In my initial implementation I've decided to blindly drop them. However your loggers work correctly and do not append those trailing zeros, so the "fix" I've added actually broke it.
Now it's also clear for me why https://github.com/jedie/inverter-connect does not work for me. Exactly because of these trailing zeros.

Here is a new, fixed docker image `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-2`

carstenfroehlich, 2023-09-09:
Thanks a lot.
I have deployed it but I can test it tomorrow, because the inverter is currently down without sun.
I will report then.

Joerglp, 2023-09-09:
I would also like to say a big thank you for your quick work. Update has been installed, but we won't see the result until tomorrow morning. It's hard for me to stand on the carport roof with a headlight and illuminate the solar panel...
what should the neighbors think....
so let's hope for the best!

carstenfroehlich, 2023-09-10:
Many thanks from me too, but it seems still not working.
`
2023-09-10 05:58:19,374 - DeyeDaemon - INFO - Please help me
 build the list of compatible inverters. https://github.com/
kbialek/deye-inverter-mqtt/issues/41
2023-09-10 05:58:19,408 - paho.mqtt.client - DEBUG - Sending
 CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-i
nverter-3939493096'
2023-09-10 05:58:19,410 - paho.mqtt.client - DEBUG - Receive
d CONNACK (0, 0)
2023-09-10 05:58:20,412 - paho.mqtt.client - DEBUG - Sending
 PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2023-09-10 05:58:20,413 - DeyeMqttClient - INFO - Successful
ly connected to MQTT Broker located at 192.168.178.113:1883
2023-09-10 05:58:20,413 - DeyeDaemon - INFO - Feature "Repor
t metrics over MQTT": enabled
2023-09-10 05:58:20,414 - DeyeDaemon - INFO - Feature "Set i
nverter time once online": disabled
2023-09-10 05:58:20,414 - DeyeDaemon - DEBUG - Start to exec
ute the daemon at intervals of 30 seconds
2023-09-10 05:58:20,416 - paho.mqtt.client - DEBUG - Receive
d PUBACK (Mid: 1)
2023-09-10 05:58:20,417 - DeyeDaemon - INFO - Reading start
2023-09-10 05:58:20,419 - DeyeDaemon - INFO - Reading regist
ers [metrics group: micro, range: 003c-0074]
2023-09-10 05:58:20,424 - DeyeConnector - DEBUG - Request fr
ame: a5170010450000e8e4cfea020000000000000000000000000000010
3003c003945d48515
2023-09-10 05:58:20,429 - DeyeConnector - DEBUG - Received r
esponse frame in 1. attempt: 41542b595a434d505645523d4d57335
f3136555f353430365f322e32370d0a0d0a
2023-09-10 05:58:20,430 - DeyeModbus - ERROR - Response fram
e has invalid starting byte
2023-09-10 05:58:20,431 - DeyeMqttClient - DEBUG - Publishin
g message. topic: 'deye/logger_status', value: 'offline'
2023-09-10 05:58:20,431 - paho.mqtt.client - DEBUG - Sending
 PUBLISH (d0, q1, r0, m2), 'b'deye/logger_status'', ... (26
bytes)
2023-09-10 05:58:20,434 - paho.mqtt.client - DEBUG - Receive
d PUBACK (Mid: 2)
2023-09-10 05:58:20,436 - DeyeMqttClient - INFO - Logger is
offline
2023-09-10 05:58:20,436 - DeyeDaemon - INFO - Reading comple
ted
2023-09-10 05:58:50,437 - DeyeDaemon - INFO - Reading start
2023-09-10 05:58:50,437 - DeyeDaemon - INFO - Reading regist
ers [metrics group: micro, range: 003c-0074]
2023-09-10 05:58:50,442 - DeyeConnector - DEBUG - Request fr
ame: a5170010450000e8e4cfea020000000000000000000000000000010
3003c003945d48515
2023-09-10 05:58:50,444 - DeyeConnector - DEBUG - Received r
esponse frame in 1. attempt: 41542b595a434d505645523d4d57335
f3136555f353430365f322e32370d0a0d0a
2023-09-10 05:58:50,446 - DeyeModbus - ERROR - Response fram
e has invalid starting byte
2023-09-10 05:58:50,447 - DeyeMqttClient - DEBUG - Publishin
g message. topic: 'deye/logger_status', value: 'offline'
2023-09-10 05:58:50,448 - paho.mqtt.client - DEBUG - Sending
 PUBLISH (d0, q1, r0, m3), 'b'deye/logger_status'', ... (26
bytes)
2023-09-10 05:58:50,450 - paho.mqtt.client - DEBUG - Receive
d PUBACK (Mid: 3)
2023-09-10 05:58:50,452 - DeyeMqttClient - INFO - Logger is
offline
2023-09-10 05:58:50,452 - DeyeDaemon - INFO - Reading comple
ted
`

I have tested if I really deployed the patch but it looks good.
`
docker pull g
hcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-2
2023.09.1-beta-2: Pulling from kbialek/deye-inverter-mqtt
Digest: sha256:71b42b946f63212d71684cbf86831ab285155dfbc37b7
b143c57f6a313bddc4b
Status: Image is up to date for ghcr.io/kbialek/deye-inverte
r-mqtt:2023.09.1-beta-2
ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-2
`

kbialek, 2023-09-10:
@carstenfroehlich I'm affraid that you are running wrong image. There should be no log messages from `DeyeConnector` class, because it's not in use by the AT command integration.
```
2023-09-10 05:58:20,424 - DeyeConnector - DEBUG - Request fr ame: a5170010450000e8e4cfea020000000000000000000000000000010
```

kbialek, 2023-09-10:
@Joerglp @carstenfroehlich Here is a new docker image with another bugfix `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-3`

carstenfroehlich, 2023-09-10:
Maybe I am doing something absolutely wrong, but I have pulled the third patch and restarted my docker container.
Result is still the same and I have still the debug message from DeyeConnector Class...

kbialek, 2023-09-10:
@carstenfroehlich Pulling is not enough. This image is not tagged as `latest`, thus you also have to explicitly reference the image version in your docker run configuration.

carstenfroehlich, 2023-09-10:
> @carstenfroehlich Pulling is not enough. This image is not tagged as `latest`, thus you also have to explicitly reference the image version in your docker run configuration.

OK, thanks, I have to read about this a bit. Until now I always checked out a branch, restart my container and it works. I have currently no idea where to find this docker run configuration. Do you mean the docker compose file?

kbialek, 2023-09-10:
@carstenfroehlich If you use docker-compose - then yes, you should reference the image like this
```
services:
  deye-mqtt:
    image: ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-3
    env_file:
      - config.env
    restart: unless-stopped
```

Joerglp, 2023-09-10:
Hello, thank.
It‘s work now.

carstenfroehlich, 2023-09-10:
Hi,
the patch is working, but it seems, that the output format for telegraf is not correct.
My influxdb is not receiving other data than the total_energy.
`
2023-09-10T08:46:53Z D! [outputs.influxdb_v2] Wrote batch of 10 metrics in 8.814369ms
2023-09-10T08:46:53Z D! [outputs.influxdb_v2] Buffer fullness: 0 / 10000 metrics
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335603678723268
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=268 1694335604685585777
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335605686476802
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=266 1694335606709000591
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335607642289198
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335608697057763
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=266 1694335609649502380
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335610654414048
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=267 1694335611656056068
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=266 1694335612657921937
2023-09-10T08:46:53Z D! [outputs.file] Wrote batch of 10 metrics in 1.887356ms
2023-09-10T08:46:53Z D! [outputs.file] Buffer fullness: 0 / 10000 metrics
2023-09-10T08:46:53Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:46:54Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:46:55Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:46:56Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:46:57Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: invalid character 'o' looking for beginning of value
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
2023-09-10T08:46:58Z E! [inputs.mqtt_consumer] Error in plugin: must be an object or an array of objects
[... cut ...]
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=282 1694335621677448298
mqtt_consumer,host=dfd91121bec7,topic=tele/tasmota_E27D60/SENSOR LK13BE_Power_curr=282 1694335622687034480
2023-09-10T08:47:03Z D! [outputs.file] Wrote batch of 10 metrics in 1.948691ms
2023-09-10T08:47:03Z D! [outputs.file] Buffer fullness: 0 / 10000 metrics
2023-09-10T08:47:03Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
2023-09-10T08:47:04Z E! [inputs.mqtt_consumer] Error in plugin: invalid character ':' after top-level value
`

kbialek, 2023-09-10:
@carstenfroehlich 

> the patch is working, but it seems, that the output format for telegraf is not correct.
> My influxdb is not receiving other data than the total_energy.

This is strange. My changes have nothing to do with metrics processing. Perhaps there was an unhandled communication error with the logger, which caused that. Please collect more logs.

@Joerglp 

Nice. It's working for me too. Let's run it for one more day to see if it's stable enough.

kbialek, 2023-09-10:
Here is another docker image `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-4`

Before using it, you have to update your configuration
1. Remove `DEYE_LOGGER_PORT` env variable 
2. Add `DEYE_LOGGER_PROTOCOL=at` env variable

For more info read this https://github.com/kbialek/deye-inverter-mqtt/tree/feature/at-interface#installation

@carstenfroehlich Please check if it also resolves your problem with influx integration.

carstenfroehlich, 2023-09-10:
sorry, but there seems to be an error in logging
`docker logs deye-inverter-mqtt_deye-mqtt_1
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'`

carstenfroehlich, 2023-09-10:
The problems in influx integration are not solved.
With the beta-4 image i do not get any data anymore...
When I reset to the master, logging is working again, but not receiving any data like before

SMCerina, 2023-09-10:
The beta 4 image is working for me. Thank you for your quick support.

kbialek, 2023-09-10:
@carstenfroehlich Share debug logs and your config.env

carstenfroehlich, 2023-09-10:
@kbialek this is my config.env:
`
DEYE_LOGGER_IP_ADDRESS=192.168.178.108
#DEYE_LOGGER_PORT=8899
DEYE_LOGGER_PROTOCOL=at
DEYE_LOGGER_SERIAL_NUMBER=3939493096

MQTT_HOST=192.168.178.113
MQTT_PORT=1883
MQTT_USERNAME=
MQTT_PASSWORD=
MQTT_TOPIC_PREFIX=deye
MQTT_TLS_ENABLED=false
#MQTT_TLS_INSECURE=true
#MQTT_TLS_CA_CERT_PATH=certs/ca.crt
#MQTT_TLS_CLIENT_CERT_PATH=certs/client.crt
#MQTT_TLS_CLIENT_KEY_PATH=certs/client.key

LOG_LEVEL=DEBUG
# LOG_STREAM=STDOUT
# PLUGINS_DIR=plugins
DEYE_DATA_READ_INTERVAL=30
DEYE_METRIC_GROUPS=micro
`

this is my whole debug log from this container:
`
docker logs deye-inverter-mqtt_deye-mqtt_1
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[... 75 lines cut ...]
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[... cut ...]
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Error setting up logging: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
`
The error is the only entry in the log, once per minute

kbialek, 2023-09-10:
@carstenfroehlich I'm pretty much confident that you are still running the `latest` or `2023.08.1` image version, which requires `DEYE_LOGGER_PORT` env variable. When this variable is not present the error you posted is printed. You see it once per minute, because docker restarts the container.

Joerglp, 2023-09-10:
@kbialek 
 Beta3 is running flawlessly for me so far. Telegraf fetches the data from Mqtt and delivers it to the influxDB.
Thanks again!
I'll get back to you in a few days and let you know whether the beta is running smoothly.

kbialek, 2023-09-10:
@Joerglp Thanks. Could you please test beta-4 too? Instructions are here https://github.com/kbialek/deye-inverter-mqtt/issues/123#issuecomment-1712760875

carstenfroehlich, 2023-09-10:
> @carstenfroehlich I'm pretty much confident that you are still running the `latest` or `2023.08.1` image version, which requires `DEYE_LOGGER_PORT` env variable. When this variable is not present the error you posted is printed. You see it once per minute, because docker restarts the container.
> 
> EDIT: Or one of the former beta version. Please simply ensure that you are running `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-4`

you are totally right, sorry for the confusion.
I had additionally to the image entry an build entry in my docker compose file, so it was always overwriting your beta image!
Thanks a lot for your support!
I will test it tomorrow, but currently this is the log:

2023-09-10 18:46:17,870 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2023-09-10 18:46:17,870 - DeyeConnectorFactory - INFO - Creating Modbus/AT Logger connector
2023-09-10 18:46:17,871 - DeyeAtConnector - INFO - Hello
2023-09-10 18:46:17,903 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3939493096'
2023-09-10 18:46:17,906 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-09-10 18:46:18,908 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2023-09-10 18:46:18,909 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.178.113:1883
2023-09-10 18:46:18,909 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
2023-09-10 18:46:18,910 - DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
2023-09-10 18:46:18,910 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 30 seconds
2023-09-10 18:46:18,911 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2023-09-10 18:46:18,912 - DeyeDaemon - INFO - Reading start
2023-09-10 18:46:18,913 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
2023-09-10 18:46:18,914 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2023-09-10 18:46:20,015 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-10 18:46:21,017 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-10 18:46:22,018 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-10 18:46:23,020 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-10 18:46:24,022 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-10 18:46:24,022 - DeyeAtConnector - WARNING - Too many connection timeouts
2023-09-10 18:46:24,023 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2023-09-10 18:46:24,124 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'

Looks good!

Joerglp, 2023-09-11:
@kbialek  beta4 ist deployed.  I'll see if it's working during the day.

carstenfroehlich, 2023-09-11:
Now the inverter seems to be damaged... The external relay has worked for one day. Now the inverter shows a permanent red light. So poorly I'm out for testing...

Joerglp, 2023-09-11:
Everything is still going great for me.

Joerglp, 2023-09-11:
It's been working for me without crashes so far

kbialek, 2023-09-11:
Same for me. It seems like the implementation is good enough.

kbialek, 2023-09-12:
The latest beta before the release `ghcr.io/kbialek/deye-inverter-mqtt:2023.09.1-beta-5`

Joerglp, 2023-09-12:
Ok! Is deployed. What was changed? Ich give answer tomorrow.

kbialek, 2023-09-12:
I've merged this https://github.com/kbialek/deye-inverter-mqtt/pull/125

closed 2023-09-13 by kbialek

Joerglp, 2025-04-25:
Dear Sir or Madam,I have been trying to reach your German hotline for some time now, but without success. As this is urgent, I hereby request an immediate exchange.Please confirm receipt of this message and let me know how we can proceed further.Best regards,Jörg KuphalIn der Wormau 4159556 LippstadtVon meinem iPad gesendet

Closed #123 as completed via #124.