kbialek/deye-inverter-mqtt #127: Deye SUN-M80G3-EU-Q0 invalid starting byte

sysrqio, 2023-09-16:
**Describe the bug**
Trying to connect to Deye SUN-M80G3-EU-Q0 in Logger firmware MW3_16U_5406_2.32-D1 gives the error:

> DeyeModbusTcp - ERROR - Response frame has invalid starting byte

**To Reproduce**
Deye Logger Version: MW3_16U_5406_2.32-D1

Output:

`
Recreating deye-inverter-mqtt_deye-mqtt_1 ... done
Attaching to deye-inverter-mqtt_deye-mqtt_1
deye-mqtt_1  | 2023-09-16 08:12:16,569 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
deye-mqtt_1  | 2023-09-16 08:12:16,570 - DeyeConnectorFactory - INFO - Creating Modbus/TCP Logger connector
deye-mqtt_1  | 2023-09-16 08:12:16,645 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3995515266'
deye-mqtt_1  | 2023-09-16 08:12:16,651 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
deye-mqtt_1  | 2023-09-16 08:12:17,651 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
deye-mqtt_1  | 2023-09-16 08:12:17,652 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at 192.168.0.152:1883
deye-mqtt_1  | 2023-09-16 08:12:17,653 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
deye-mqtt_1  | 2023-09-16 08:12:17,654 - DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
deye-mqtt_1  | 2023-09-16 08:12:17,655 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 60 seconds
deye-mqtt_1  | 2023-09-16 08:12:17,663 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
deye-mqtt_1  | 2023-09-16 08:12:17,664 - DeyeDaemon - INFO - Reading start
deye-mqtt_1  | 2023-09-16 08:12:17,667 - DeyeDaemon - INFO - Reading registers [metrics group: micro, range: 003c-0074]
deye-mqtt_1  | 2023-09-16 08:12:17,676 - DeyeTcpConnector - DEBUG - Request frame: a517001045000082b926ee0200000000000000000000000000000103003c003945d44f15
deye-mqtt_1  | 2023-09-16 08:12:17,782 - DeyeTcpConnector - DEBUG - Received response frame in 1. attempt: 41542b595a434d505645523d4d57335f3136555f353430365f322e33322d44310d0a0d0a
deye-mqtt_1  | 2023-09-16 08:12:17,783 - DeyeModbusTcp - ERROR - Response frame has invalid starting byte
deye-mqtt_1  | 2023-09-16 08:12:17,792 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
deye-mqtt_1  | 2023-09-16 08:12:17,793 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'deye/logger_status'', ... (7 bytes)
deye-mqtt_1  | 2023-09-16 08:12:17,799 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
deye-mqtt_1  | 2023-09-16 08:12:17,804 - DeyeMqttClient - INFO - Logger is offline
deye-mqtt_1  | 2023-09-16 08:12:17,804 - DeyeDaemon - INFO - Reading completed
^CGracefully stopping... (press Ctrl+C again to force)
`

**Hardware (please complete the following information):**
 - Inverter model: SUN-M80G3-EU-Q0

**Software (please complete the following information):**
 - Logger firmware version: MW3_16U_5406_2.32-D1
 - deye-inverter-mqtt version: 2023.09.1
 - config.env (remove passwords before pasting):

`
DEYE_LOGGER_IP_ADDRESS=192.168.0.15
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=39955xxxxx
DEYE_LOGGER_PROTOCOL=tcp

MQTT_HOST=192.168.0.152
MQTT_PORT=1883
# MQTT_USERNAME=user
# MQTT_PASSWORD=password
MQTT_TOPIC_PREFIX=deye
# MQTT_TLS_ENABLED=true
# MQTT_TLS_INSECURE=true
# MQTT_TLS_CA_CERT_PATH=certs/ca.crt
# MQTT_TLS_CLIENT_CERT_PATH=certs/client.crt
# MQTT_TLS_CLIENT_KEY_PATH=certs/client.key

LOG_LEVEL=DEBUG
# LOG_STREAM=STDOUT
# PLUGINS_DIR=plugins
# DEYE_DATA_READ_INTERVAL=60
# DEYE_PUBLISH_ON_CHANGE=false
DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=60
DEYE_METRIC_GROUPS=micro
`

MarcErl, 2023-09-16:
Same for me after I installed the Deye relay box and the inverter/logger has made a SW update to the version mentioned by sysrqio.

kbialek, 2023-09-16:
Very likely a duplicate of https://github.com/kbialek/deye-inverter-mqtt/issues/123

tldr: you have to switch from TCP to AT protocol

Fixed in https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.09.1
Also read the installation instructions here https://github.com/kbialek/deye-inverter-mqtt#installation

closed 2023-09-19 by kbialek