kbialek/deye-inverter-mqtt #128: Errors wit Version 2023.09.1 after connect Relais and Firmwareupdate

HNeumeier, 2023-09-17:
**Hardware (please complete the following information):**
 - Inverter model: DEYE SUN 600

**Software (please complete the following information):**
 - Logger firmware version:MW3_16U_5406_2.32-D1
 - deye-inverter-mqtt version: 2023.09.1
 - config.env 
 DEYE_LOGGER_IP_ADDRESS=192.168.2.161
DEYE_LOGGER_PORT=8899
DEYE_LOGGER_SERIAL_NUMBER=41769xxxxxx
DEYE_LOGGER_PROTOCOL=at

MQTT_HOST=192.168.2.160
MQTT_PORT=1883
MQTT_USERNAME=root
MQTT_PASSWORD=xxxx
MQTT_TOPIC_PREFIX=deye

LOG_LEVEL=INFO
DEYE_DATA_READ_INTERVAL=300
DEYE_PUBLISH_ON_CHANGE=true
DEYE_PUBLISH_ON_CHANGE_MAX_INTERVAL=360
DEYE_METRIC_GROUPS=micro

here is how i call it
bash -c "$(cat config.env | xargs) python3 deye_docker_entrypoint.py"

Traceback (most recent call last):
  File "/opt/fhem/bin/deye-inverter-mqtt-main-neu/deye_docker_entrypoint.py", line 21, in <module>
    from deye_cli import main as cli_main
  File "/opt/fhem/bin/deye-inverter-mqtt-main-neu/deye_cli.py", line 20, in <module>
    from deye_config import DeyeConfig
  File "/opt/fhem/bin/deye-inverter-mqtt-main-neu/deye_config.py", line 26, in <module>
    class DeyeEnv:
  File "/opt/fhem/bin/deye-inverter-mqtt-main-neu/deye_config.py", line 57, in DeyeEnv
    def string(env_var_name: str, default_value: str = None) -> str | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

kbialek, 2023-09-17:
Make sure you run it using python 3.10

HNeumeier, 2023-09-17:
Thanks for your help with python 3.10 it is running very good Job.

closed 2023-09-17 by HNeumeier