kbialek/deye-inverter-mqtt #129: Deye SUN600-G3/EU230 FW:2.32 - Too many connection

videejay, 2023-09-17:
**Describe the bug**
I attached the mandatory relay box to my inverter and had to update my firmware to 2.32. I manually updated to version using the inverter UI. After also updating the docker container and switching to AT mode I am facing too many connection errors. Before with older FW 1.57 and no relay it worked flawlessly. Please help identifying the issue. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect using 1.57 FW 
2. Observe that it works
3. Attach Deye-SUN-MI-RELAY-01
4. Update FW to 2.32
5. See too many connection warning in the log

**Expected behavior**
Should be able to read inverter data also with attached relay.

**Hardware (please complete the following information):**
 - Inverter model: Deye SUN600G3/EU230

**Software (please complete the following information):**
 - Logger firmware version: 2.57
 - deye-inverter-mqtt version: latest
 - config.env (remove passwords before pasting):
 - Not using config.env but docker-compose envs:
 -   - DEYE_LOGGER_IP_ADDRESS={IP of inverter}
      - DEYE_LOGGER_PORT=8899
      - DEYE_LOGGER_SERIAL_NUMBER={serial of inverter}
      - MQTT_HOST={BROKER_IP}
      - MQTT_PORT=1883
      #- MQTT_USERNAME=user
      #- MQTT_PASSWORD=password
      - MQTT_TOPIC_PREFIX=deye
      - LOG_LEVEL=DEBUG
      - DEYE_DATA_READ_INTERVAL=60
      - DEYE_METRIC_GROUPS=string
      - DEYE_LOGGER_PROTOCOL=at

**Additional context**
2023-09-17 13:30:42,922 - DeyeDaemon - INFO - Please help me build the list of compatible inverters. https://github.com/kbialek/deye-inverter-mqtt/issues/41
2023-09-17 13:30:42,922 - DeyeConnectorFactory - INFO - Creating Modbus/AT Logger connector
2023-09-17 13:30:43,007 - paho.mqtt.client - DEBUG - Sending CONNECT (u0, p0, wr1, wq1, wf1, c1, k60) client_id=b'deye-inverter-3902349006'
2023-09-17 13:30:43,010 - paho.mqtt.client - DEBUG - Received CONNACK (0, 0)
2023-09-17 13:30:43,021 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r1, m1), 'b'deye/status'', ... (6 bytes)
2023-09-17 13:30:43,021 - DeyeMqttClient - INFO - Successfully connected to MQTT Broker located at {BROKER_IP}:1883
2023-09-17 13:30:43,021 - DeyeDaemon - INFO - Feature "Report metrics over MQTT": enabled
2023-09-17 13:30:43,021 - DeyeDaemon - INFO - Feature "Set inverter time once online": disabled
2023-09-17 13:30:43,022 - DeyeDaemon - DEBUG - Start to execute the daemon at intervals of 60 seconds
2023-09-17 13:30:43,023 - DeyeDaemon - INFO - Reading start
2023-09-17 13:30:43,024 - DeyeDaemon - INFO - Reading registers [metrics group: string, range: 003c-0074]
2023-09-17 13:30:43,024 - DeyeAtConnector - DEBUG - Sending AT command: b'WIFIKIT-214028-READ'
2023-09-17 13:30:43,029 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 1)
2023-09-17 13:30:44,130 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:45,131 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:46,134 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:47,143 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:48,144 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:48,144 - DeyeAtConnector - WARNING - Too many connection timeouts
2023-09-17 13:30:48,144 - DeyeAtConnector - DEBUG - Sending AT command: b'+ok'
2023-09-17 13:30:48,245 - DeyeAtConnector - DEBUG - Sending AT command: b'AT+INVDATA=8,0103003c003945d4\n'
2023-09-17 13:30:50,347 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:51,348 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:52,349 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:53,350 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:54,352 - DeyeAtConnector - DEBUG - Connection response timeout
2023-09-17 13:30:54,352 - DeyeAtConnector - WARNING - Too many connection timeouts
2023-09-17 13:30:54,352 - DeyeMqttClient - DEBUG - Publishing message. topic: 'deye/logger_status', value: 'offline'
2023-09-17 13:30:54,353 - paho.mqtt.client - DEBUG - Sending PUBLISH (d0, q1, r0, m2), 'b'deye/logger_status'', ... (7 bytes)
2023-09-17 13:30:54,356 - paho.mqtt.client - DEBUG - Received PUBACK (Mid: 2)
2023-09-17 13:30:54,357 - DeyeMqttClient - INFO - Logger is offline
2023-09-17 13:30:54,357 - DeyeDaemon - INFO - Reading completed

kbialek, 2023-09-17:
Remove `DEYE_LOGGER_PORT` from the config. Read the release notes of the latest release https://github.com/kbialek/deye-inverter-mqtt/releases/tag/2023.09.1

videejay, 2023-09-17:
Worked! Thank you so much for helping. This is somehow confusing though, as telnet worked on that port. So I didn’t really assume any issue in that direction. Sorry for not reading the release notes.

closed 2023-09-17 by videejay