kbialek/deye-inverter-mqtt #131: Direct to InfluxDB2

m1k3f15h, 2023-10-13:
**Is your feature request related to a problem? Please describe.**
At the moment a Telegraf instance put the values from the Deye (over MQTT) to influxdb2.

**Describe the solution you'd like**
Would be awesome if this docker-container would write the values direct to influxdb2,

**Describe alternatives you've considered**
Telegraf

**Additional context**
Maybe it is possible to create a plugin?

Somebody else have a usecase for this?

greetings from austria
Mike

kbialek, 2023-10-28:
Direct integration with Influxdb is out of the scope of this project. However, it should be doable by implementing a plugin, as described here https://github.com/kbialek/deye-inverter-mqtt#custom-plugins

closed 2023-10-28 by kbialek

fakehec, 2023-11-14:
Hi @m1k3f15h,

I've got a working telegraf.conf that works properly for influxdb/grafana. The logic is

deye-inverter-mqtt --> mosquitto (MQTT)--> telegraf --> InfluxDB

You should config a telegraf input plugin as follows

[[inputs.mqtt_consumer]]
   name_prefix = "<INFLUXDB MEASUREMENT NAME>"
   servers = ["tcp://<MQTT SERVER IP> :<MQTT SERVER PORT>"]
   username = "<MQTT SERVER USER>"
   password = "<MQTT SERVER PASS>"
   data_format = "value"
   data_type = "float"
   topics = [
        "<MQTT TOPIC PREFIX>/#"
        ]
        [[inputs.mqtt_consumer.topic_parsing]]
                topic = "<MQTT TOPIC PREFIX>/+/+"
                #measurement = "_/_/_/_/measurement/_/_/_"
                tags = "site/_/_/location/inverter/field"
       [[inputs.mqtt_consumer.topic_parsing]]
                topic = "<MQTT TOPIC PREFIX>/+/+/+"
                #measurement = "_/_/_/_/measurement/_/_/_"
                tags = "site/_/_/location/inverter/system/field"
        [[inputs.mqtt_consumer.topic_parsing]]
                topic = "<MQTT TOPIC PREFIX>/+/+/+/+"
                #measurement = "_/_/_/_/measurement/_/_/_"
                tags = "site/_/_/location/inverter/system/line/field"
        [[inputs.mqtt_consumer.topic_parsing]]
                topic = "<MQTT TOPIC PREFIX>/+/+/+/+/+"
                #measurement = "_/_/_/_/measurement/_/_/_"
                tags = "site/_/_/location/inverter/system/line/subsystem/field"

MQTT TOPIC PREFIX should be set on the config.env of deye-inverter-mqtt

Hope this helps.